MarlinFirmware/Marlin #5742: Compiling Issue with DISTINCT_E_FACTORS

vin-iachini, 2017-01-24:
When #define DISTINCT_E_FACTORS is uncommented in Configuration.h (line 149), a compiling issue arises in planner.h (line 353) stating "invalid conversion from 'int' to 'AxisEnum'".

I have modified the DEFAULT_STEPS_PER_UNIT in Configuration.h (line 489) to add the steps for the additional extruder.

I do not currently have a fix other than to comment out DISTINCT_E_FACTORS again.

Bob-the-Kuhn, 2017-01-25:
There's also DEFAULT_MAX_FEEDRATE and DEFAULT_MAX_ACCELERATION that have places for the active extruders.

If you still see this error on the latest RCBugFix then 
post your configuration.h, congiguration_adv.h & pins_xxxx.h files
tell us what version of Adruino IDE you're using

With the above info I can probably track down the problem.

vin-iachini, 2017-01-25:
Hi Bob-the-Kuhn, I appreciate the help!  I did address DEFAULT_MAX_FEEDRATE and DEFAULT_MAX_ACCELERATION and I still have this issue.  I have uploaded the files that you asked for via an edit to my original post.

Bob-the-Kuhn, 2017-01-25:
It compiled fine for me.

I tried RC8, the latest RCBugFix, 1.8.1, 1.6.12 and all combinations.  No compile issues in any of the four cases.

The only thin I can suggest is to download fresh copies of RC8 and 1.81 and try again.

vin-iachini, 2017-01-26:
Hmmm.  I'm still having no luck after reinstalling arduino 1.8.1 and downloading RC8 again.  I have now included a picture of my error message screen in the original post.

Bob-the-Kuhn, 2017-01-26:
I still can't reproduce your problem.  You've made almost no changes to the configuration.h file and the others you supplied are a 100% match to RC8.

What board have you selected in the Arduino IDE?  

What does Boards Manager say is installed?  Mine only has "Arduino AVR Boards Built-In by Arduino version 1.6.17"

Please make a text copy of the Arduino message window and attach it to this issue.  In Windows click in the box and then do a control A followed by a control C.

vin-iachini, 2017-01-27:
The board selection was the issue I guess.  I had "Arduino/Genuino Mega or Mega 2560" but now it works with "Arduino Mega ADK."  I apologize for the wild goose chase, but I appreciate the help!

closed 2017-01-27 by vin-iachini

Bob-the-Kuhn, 2017-01-27:
??? 
"Arduino/Genuino Mega or Mega 2560" is what I thought was the right selection for a 2560 based RAMPS system.  Seems like the more complex the system the more quirks they have.

Oh, Well.  You're up & running so all is well.