MarlinFirmware/Marlin #5743: RC8 Bugfix Branch G38.2/3 not working (does not stop on Z_MIN/PROBE hit)

RedbullVienna, 2017-01-24:
I tried to use the G38.2./3 command to probe (with my Z probe) the edge (X or Y) of my work piece (for milling).
e.g. current X pos is 10 and I send the command G38.2 X100 the machine moves to position X100 but it does not stop when my probe is triggering as my work peace is on the path from X10 to X100. So finial it just moves but does not probe/stop when the probe is triggered at the edge of the work piece.
The probe itself is working, as G28 Z works fine when the probe touches the top surface of the work piece.
Also G30 is working fine.

Any idea what went wrong here? Are my setting just wrong, something missed or is their a bug in marlin?

Is use the following settings: 
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define PROBE_DOUBLE_TOUCH
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_ZMIN_PLUG
#define FIX_MOUNTED_PROBE
#define G38_PROBE_TARGET
#define ENDSTOP_INTERRUPTS_FEATURE

Bob-the-Kuhn, 2017-01-25:
The probe works with G28 so a G38.2 in the negative Z direction should also work.

What does your probe consist of?  

I've used it with end mills and sometimes the sides of the end mill wasn't conductive.  Rubbing some smooth metal on the side of the mill usually did the trick.

RedbullVienna, 2017-01-25:
For milling my probe is simply a wire on the Z_MIN input (with pull up resistor) attached to the end mill and the GND wire connect to the metal work piece. But I'm sure it is not an electrical issue as I did it for testing just by connection the input wire and ground wire manually during the X or Y movement is ongoing after G38.2 Xxxx or G38.2 Yxxx. I tried it right now also with G38.2 Zxxx, which also not works. As I wrote for G28 X Y or Z or G30 it works fine.

Bob-the-Kuhn, 2017-01-25:
Confirmed - G38 is not working.

I'll see what went wrong & get back to you with a fix.

RedbullVienna, 2017-01-25:
I think I found the issue. It is within endstops.cpp.
.) UPDATE_ENDSTOP is not call for directions where you don't have an end stop configured. In my case I do not have XYZ_MAX.
.) if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX)) && stepper.current_block->steps[_AXIS(AXIS)] > 0) here it is check that we have movement on "AXIS" ( = Z for G38) but in case of G38 X0 we move on X and stepper.current_block->steps[X] is 0

I added the following code and now it works fine :-)

  #if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN)  // If G38 command then check Z_MIN for every axis and every direction
    if (G38_move) {
      UPDATE_ENDSTOP_BIT(Z, MIN); 
      if (TEST_ENDSTOP(_ENDSTOP(Z, MIN)) && ((stepper.current_block->steps[_AXIS(X)] > 0) ||  (stepper.current_block->steps[_AXIS(Y)] > 0) || (stepper.current_block->steps[_AXIS(Z)] > 0))) { 
        _ENDSTOP_HIT(Z); 
        stepper.endstop_triggered(_AXIS(Z)); 
        G38_endstop_hit = true;
      }
    }
  #endif

I just miss an output on the serial console with the position where the probe hit occurred. But I need this to do the work piece alignment with G92.

Bob-the-Kuhn, 2017-01-25:
Yes, I just tested an almost identical code change.

Here's the G-code program I use to set the 0,0,0 position to the corner of the workpiece.  It assumes a 3.175mm dia endmill and a touch plate similar to the [Triquetra](http://triquetra-cnc.com/) one.
[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/731026/endstops.zip)

RedbullVienna, 2017-01-25:
Thank you for the G code and checking that issue!
Regarding your fix I think it will not work for all directions if you don't have end stops on all axis for min and max. e.g. I have them only on XYZ min, not on XYZ max. Due to the pre compiler code e.g. #if HAS_X_MAX -> UPDATE_ENDSTOP(X, MAX) is not called on a machine without X_MAX end stop. So you can not probe along X+. Same for Y+ and Z+. Maybe Z+ no one uses but X+ and Y+ I want to have. ;-)

Bob-the-Kuhn, 2017-01-25:
Maybe even remove the requirement for X_MIN & Y_MIN.  Just a little more code is all.

Bob-the-Kuhn, 2017-01-26:
Try this one out.  It doesn't care if any endstops (except X_MIN) are enabled.  I had to modify SanityCheck.h to disable endstop checks before I could test it without endstops.
[endstops (2).zip](https://github.com/MarlinFirmware/Marlin/files/731288/endstops.2.zip)

Bob-the-Kuhn, 2017-01-26:
This bug is fixed by PR #5678

RedbullVienna, 2017-01-26:
If have tested right now your second solution. I works fine in all six direction with my setting! 
THX!

closed 2017-04-02 by thinkyhead