MarlinFirmware/Marlin #5747: Setting the extruder temperature command from inside the Arduino code

irmas, 2017-01-25:
Hello,

This may not be the best place to ask this type of questions, but I didn't find a better place to ask. 

I'd like to execute the Gcode (M109) to set the extruder temperature to a specific value from within the arduino code (instead of setting it through an LCD screen or by a serial command). 

when I execute the gcode_M109() function, no temperature parameter is passed and I can't find a way to set the temperature to the temperature I want. 

Do you have any tips ?

Bob-the-Kuhn, 2017-01-26:
These commands are what my slicer uses to set the temp & wait.
```
M190 S95 ; set bed temperature
M104 S240 ; set temperature
M109 S240 ; wait for temperature to be reached
```

Bob-the-Kuhn, 2017-01-26:
Sorry - I didn't realize you wanted something compiled.

You are correct,  gcode_M109() doesn't take any arguments.

Looks like you'll need to create your own routine modelled after M109.

Bob-the-Kuhn, 2017-01-26:
Curiosity - why do you want to do this?

MagoKimbra, 2017-01-26:
code_seen() reads commands and all sub commands.
After reading the M109 command is called the routine then always check with code_seen if the command is in the S or R sub command.
`
bool no_wait_for_cooling = code_seen('S');
if (no_wait_for_cooling || code_seen('R')) {
`
After that the numerical value is read from the routine code_value_temp_abs() e put it in target hotend.
`
thermalManager.setTargetHotend(code_value_temp_abs(), TARGET_EXTRUDER);
`

irmas, 2017-01-26:
Thank you very much.

I managed to create a copy of the gcode_M109() function that takes a temperature as an argument as @Bob-the-Kuhn suggested.

Setting the temperature is done by passing the argument to thermalManager.setTargetHotend as @MagoKimbra pointed out.

I am doing this because my mechanical platform can't directly be controlled with Gcode. So I set up my platform to execute a trajectory and send orders Marlin to set the temperature, start extruding etc. It isn't ideal but it should work.

Thank you for your help

closed 2017-01-26 by irmas

thinkyhead, 2017-02-12:
There's always this too:
```cpp
enqueue_and_echo_commands_P(PSTR("M109 S200"));
```