MarlinFirmware/Marlin #5749: G92 E0 affects X axis when DISTINCT_E_FACTORS defined

rekeyed, 2017-01-26:
I have two extruders on Prusa i3, and enabling DISTINCT_E_FACTORS leads to such results:

> 19:03:23.548 : N19 G28\*43
> 19:03:23.562 : echo:Active Extruder: 0
> 19:03:47.679 : echo:Active Extruder: 1
> 19:03:47.679 : X:100.00 Y:131.00 Z:10.80 E:0.00 Count X: 8000 Y:10480 Z:34560
> 19:04:44.778 : N47 G92 E0\*116
> 19:04:44.797 : X:100.00 Y:131.00 Z:10.80 E:0.00 Count X: 8000 Y:10480 Z:34560
> 19:05:04.949 : N54 G1 X10 F5000\*3

all seems to be good, but carriage moves right 10mm instead of 90mm left (physically X position is 110, in firmware 10). So, `G92 E0` command sets X to zero, but not shows that.

relative coordinates are broken too:
> 19:13:59.456 : N230 G28\*18
> 19:13:59.462 : echo:Active Extruder: 0
> 19:14:16.346 : echo:Active Extruder: 1
> 19:14:16.347 : X:100.00 Y:131.00 Z:10.80 E:0.00 Count X: 8000 Y:10480 Z:34560
> 19:14:22.972 : N238 G91\*24
> 19:14:43.265 : N246 G92 E0\*71
> 19:14:43.287 : X:100.00 Y:131.00 Z:10.80 E:0.00 Count X: 8000 Y:10480 Z:34560
> 19:14:55.805 : N251 G1 X-50 F5000\*29

and carriage moves 50 mm right instead of left

I repeat this on RC8 and today's RCBugFix branches
when DISTINCT_E_FACTORS not defined, machine works well.

alejandrocragnolini, 2017-02-11:
+1
I'm having the same problem.

thinkyhead, 2017-02-12:
Please post your `Configuration.h` file. (You may need to ZIP it before you can drop it on the textfield.)

rekeyed, 2017-02-12:
here is my Configuration.h
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/769551/Configuration.zip)

alejandrocragnolini, 2017-02-13:
Here is mine.
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/772227/Configuration.h.zip)

thinkyhead, 2017-02-14:
I'll look at these more closely this afternoon (i.e., in about 10 hours).

thinkyhead, 2017-03-20:
I think this one is fixed now in `RCBugFix`. Please confirm.

rekeyed, 2017-03-24:
latest RCBugFix works well with DISTINCT_E_FACTORS enabled on my machine. thanks!

closed 2018-02-06 by thinkyhead