MarlinFirmware/Marlin #5752: Hotend configuration swap menu

studiodyne, 2017-01-28:
Can anyone make a code , to have a simple swap menu to our predefined hotends ' we will define our own setting for each , in firmware '

If we want to swap from chimera to cyclops or diamond we have to ' refirm ' the board 
A lcd menu that set the offsets of the nozzles + trick the software with Heater0/sensor0 only for all 
Or may be , just ' separate extruders/heaters/sensors ' or ' singlenozzle/singleheater/singlesensor' , you will fix the extruder number to 4 , and it will works for anyone
Thks

I'm beginner , and for the moment , i cannot code , my knowledge is too bad , may be in the future i will contribute seriously , but now ... i need help 

Thks

Bob-the-Kuhn, 2017-01-29:
I've been playing with Marlin for 5 months now and am still finding options that are useful.

As I understand it, you have one extruder slot/location on your machine and want to swap different extruders in/out of that slot.

I think you can setup Marlin to do most of what you want but there are probably some G-code commands you’ll still need to issue when switching extruders.  

You’ll probably need to have unique slicer settings for each extruder so you can probably automate the G-code commands by adding them to the slicer’s “start up g-code” script.  If that's the case then all you need to do is:
1. slice for the target extruder
2. put the target extruder into the printer
3. send the G-code file to the printer

////////////////////////////////////////////////////////////////////////////////////////////

DETAILS

I'm running a single extruder so I can't test the following. 

As I read the config files and the code, Marlin allows the following to be set for each extruder (max 4 extruders):
- stepper driver
- heater driver
- thermal sensor input
- fan
- DEFAULT_AXIS_STEPS_PER_UNIT
- DEFAULT_MAX_FEEDRATE
- DEFAULT_MAX_ACCELERATION

The following are shared by all extruders:
- PID values
- X, Y offset from extruder zero (zero in this case)
- Homing offsets
- Z offset
- Jerk

 

By setting up the configuration.h, the configuration_adv.h and pins_YOUR_BOARD.h files properly you can change from one extruder to the other by issuing the following commands:
- T0, T1, T2, T3  (selects the active extruder)
- M851 Zxxx   (sets the Z offset from the bed – M206 may also do this)

You may also need to issue some of these commands:
- M206 Xxxx Yxxx Zxxx (sets homing offsets – I’m assuming  that the nozzles will be slightly different from each other but I doubt that the X & Y changes are enough to matter)
- M205 Exxx (max jerk for extruder)
- M207 (retract length)
- M208 (set recover length)
- M301 (set PID parameters)

///////////////////////////////////////////////////////////////

In your configuration.h file you’ll need to set TEMP_SENSOR_0 … TEMP_SENSOR_3 all the same.

In your pins_YOUR_BOARD.h file, you’ll need to set the stepper driver, heater driver, thermal sensor input & fan the same for all extruders.  The following should do that:
```
#define  E0_STEP_PIN      xx               
#define  E0_DIR_PIN       xx               
#define  E0_ENABLE_PIN    xx               
                                           
#define  E1_STEP_PIN      E0_STEP_PIN      
#define  E1_DIR_PIN       E0_DIR_PIN       
#define  E1_ENABLE_PIN    E0_ENABLE_PIN    
                                           
#define  E2_STEP_PIN      E0_STEP_PIN      
#define  E2_DIR_PIN       E0_DIR_PIN       
#define  E2_ENABLE_PIN    E0_ENABLE_PIN    
                                           
#define  E3_STEP_PIN      E0_STEP_PIN      
#define  E3_DIR_PIN       E0_DIR_PIN       
#define  E3_ENABLE_PIN    E0_ENABLE_PIN    
                                           
#define  TEMP_0_PIN       xx               
#define  TEMP_1_PIN       TEMP_0_PIN       
#define  TEMP_2_PIN       TEMP_0_PIN       
#define  TEMP_3_PIN       TEMP_0_PIN       
                                           
#define  HEATER_0_PIN     xx               
#define  HEATER_1_PIN     HEATER_0_PIN     
#define  HEATER_2_PIN     HEATER_0_PIN     
#define  HEATER_3_PIN     HEATER_0_PIN     
                                           
#define  E0_AUTO_FAN_PIN  xx               
#define  E1_AUTO_FAN_PIN  E0_AUTO_FAN_PIN  
#define  E2_AUTO_FAN_PIN  E0_AUTO_FAN_PIN  
#define  E3_AUTO_FAN_PIN  E0_AUTO_FAN_PIN  

// strange - Arduino also wanted these defined
#define  E4_STEP_PIN      E0_STEP_PIN      
#define  E4_DIR_PIN       E0_DIR_PIN       
#define  E4_ENABLE_PIN    E0_ENABLE_PIN 

#define  E5_STEP_PIN      E0_STEP_PIN      
#define  E5_DIR_PIN       E0_DIR_PIN       
#define  E5_ENABLE_PIN    E0_ENABLE_PIN 
                                           
 // you may also need some of the following if using a board that pulls in the pins_RAMPS.h file                                          
                                           
#undef   E0_STEP_PIN                       
#undef   E0_DIR_PIN                        
#undef   E0_ENABLE_PIN                     
                                           
#undef   E1_STEP_PIN                       
#undef   E1_DIR_PIN                        
#undef   E1_ENABLE_PIN                     
                                           
#undef   E2_STEP_PIN                       
#undef   E2_DIR_PIN                        
#undef   E2_ENABLE_PIN                     
                                           
#undef   E3_STEP_PIN                       
#undef   E3_DIR_PIN                        
#undef   E3_ENABLE_PIN                     
                                           
#undef   TEMP_0_PIN                        
#undef   TEMP_1_PIN                        
#undef   TEMP_2_PIN                        
#undef   TEMP_3_PIN                        
                                           
#undef   HEATER_0_PIN                      
#undef   HEATER_1_PIN                      
#undef   HEATER_2_PIN                      
#undef   HEATER_3_PIN                      
                                           
#undef   E0_AUTO_FAN_PIN                   
#undef   E1_AUTO_FAN_PIN                   
#undef   E2_AUTO_FAN_PIN                   
#undef   E3_AUTO_FAN_PIN
```

studiodyne, 2017-01-29:
Thks for your help , but it's not exactly what i need , i will explain you

I want Numbers of extruders directly set on the lcd , why  ??
I know how to make each configuration , for one , dual , cyclops or diamonds , but there are some problems

Case 1 : When you have CYCLOPS config ,  all heaters/sensors the same pins , and you change the carrier , with dual extrusion , you can not heat the second extruder to get out the filament , you have to refirm the board , just for ejecting a filament

Case  2: When you are in dual config ,  and you change the carrier with  cyclops , it's impossible to load the left filament because , you don't have  temp on the left heater/sensor 

Case 3 : same for diamonds

So my idea , is to change ' THE DEFINED HEATERS/sensors' with VARIABLES , and in this case , i can code a value change in an item menu 
Using gcode , is for the users that have a computer always connected , but i have standalone SDCARD printer , i don't want to put a computer each time i want to change the carrier

Using the slicer doesn't serve me if i just want to test a hotend or load/unload filament

For the moment , i just use ' CYCLOPS/CHIMERA' , so , i trick ....   I use the two sensors , and the 2 resistors to always have a functionnal firmware

On cyclops , i put the 2 heaters and the 2 sensors

I will try to code it , i just need , to have variable datas that replace , some pins ,  add some item menusto change value , and it will be perfect

I'm afraid to be not enough experimented , to store the datas on the eeprom

THKS

Bob-the-Kuhn, 2017-01-30:
You're running 3 distinct extruder systems - chimera, cyclops & diamond 

By "change the carrier" do you mean you're changing the extruder system?

I hope your extruder steppers are all identical.  The DISTINCT_E_FACTORS option doesn't appear to be functioning properly.

Sounds like you've solved the two heaters vs. one heater problem by 1) enabling two heaters in the configuration and 2) tricking the hardware into thinking it has two active thermal sensors on Cyclops by wiring in an unused thermal sensor (or a fixed resistor) in addition to the one on the Cyclops.

Would the following work for you? Basically set it up as four independent extruders but do some tricks to get the sensors, fans & heaters defined so that Marlin doesn't complain.  I'm thinking that T0 would be used on all, T1 only for the second chimera hotend and T2 & T3 for diamond only.  This assumes you can setup your slicer for these T0 - T3 definitions.
- only use one type of sensor on the hotends
- setup cyclops sensors as you have it now (one real sensor, one dummy sensor)
- setup chimera sensors as you have it now (two real sensors)
- setup diamond sensors the same as cyclops
- set #extruders to 4
- do NOT enable single nozzle, switching nozzle or mixing nozzle
- set all four thermal sensors to the same sensor type
- in pins_YOUR_BOARD.h set TEMP_0, TEMP_1,HEATER_0, HEATER_1, E0 fan & E1 fan for chimera
- in pins_YOUR_BOARD.h set other temp, heater & fan pins the same as TEMP_0, HEATER_0 & E0 fan
- in pins_YOUR_BOARD.h set all the E2 stepper pins to the E1 pins
- #define HOTEND_OFFSET_X {0.0, 0.0, 0.0, 0.0}  // use M218 command to set chimera's value
- #define HOTEND_OFFSET_Y {0.0, 0.0, 0.0, 0.0}  // use M218 command to set chimera's value
- #define HOTEND_OFFSET_Z {0.0, 0.0, 0.0, 0.0}  // code change needed to M218 if chimera's value isn't 0

To switch between extruder systems you'd have to issue the appropriate M218 command.  Can you do that via your LCD menu?  

You'd also need to include M104 T1 Syyy commands in the slicer's start code when not slicing for the chimera and M104 T2 S75 & M104 T3 S75 commands in all case.  This is so that Marlin always has target temperatures for the non-existent hotends.  S75 is any temperature below the real target to T0.  Syyy is something lower than what Marlin reports for the dummy sensor.  If Syyy is too low then the "allow cold extrusion" option will have to be enabled.

///////////////////////////////////////

Do you want/need the mixing extruder function where all the extruder steppers could be pushing filament at the same time?  The above assumes you're only pushing one filament at a time.

studiodyne, 2017-01-30:
I've made your suggestion ,  use the M218 gcode by lcd , is easy , but not solve my problem of using the machine
If we print it's ok , but , now , i'm creating a new printer and i never print , i just try the motors , the heaters and  the command is not possible for all extrusion system

Changing offset by lcd is easy , a simple menu item with gcode instruction

But changing pin need to modify the source

' The chimera have two resistors , but half size , it's easy to place these two resistors in the cyclops , that fit perfect ,  for the 2 sensors i fix the second with kapton and some insulation 

It's perfect

I will apply this

2 resistors/ 2 sensors every time , permanent mounting
and i will make my lcd menu item with the gcode m218 , it's perfect

Yes the ' carrier' is ' the extrusion system'

Thks for your help

Bob-the-Kuhn, 2017-01-30:
I just saw your request for custom code (issue #5759 ).

If the above works, you could put a g-code command file on the SD for each extruder system.  In the file would be the M218 command needed for that particular extruder system.  They're accessible from LCD menu system.  Not as nice as having a custom system but workable with what we have available today.

studiodyne, 2017-01-30:
More simple

If the two sensors are active , and the two heaters too 
All works everytime without anything 
For chimera and cyclops it's perfect
The LCD menu ' offset change ' not exist on marlin RC8 , i will make it by myself
Two lines , one for chimera , the second for cyclop

Tks

Bob-the-Kuhn, 2017-01-30:
I don't understand "But changing pin need to modify the source".  What pin are you changing?

Are you putting two heaters into the cyclops?  If yes, I hope they are tied to one driver.  If tied to two drivers there's a small possibility you might end up with only one doing all the work (which means slower heating times).  On the other hand you may need two drivers just to keep from over powering the controller board.

If you want to get rid of the kapton tape you could replace the dummy sensor with a fixed resistor.  If  you tell me what sensor you're using and the temperature you want it to read then I can calculate (or look up) a value for the resistor. Hmmmm .... might be better to set it up to read room temperature and enable cold extrudes via the PREVENT_COLD_EXTRUSION flag and/or the M302 command.  But this enables or disables the cold extrude for all steppers.    Maybe this resistor idea isn't as nice as it first looked.

studiodyne, 2017-01-30:
Yes , the two heater of the ' CHIMERA' have the half size of the ' CYCLOPS' , so , i can place the two heaters in the cyclops easily , and ,of course the two sensors together on the screw

Now , with this config , all works , there is two heating procedure at the same time , on the same hotend , it's not a problem , the security works in case of overheat 

Changing pin , mean that i need , to modify the firm , to give the same pins , for all heater and sensor , it's the most simple way for a beginner to trick the firmware for only one HOTEND and MANY EXTRUDERS

My systems works fine ,  because the hardware is not modified , two heaters , two sensors , in cyplops mounting or chimera 
I use two separate extrusion system that i can change quickly , i swap between chimera and cyclops when i need to print something , it's the reason that i need to have the immediate switching possibility
To keep The heaters and the sensors , is SIMPLE AND PERFECT
I will add a ' OFFSET ' m218 gcode ITEM MENU for each config and   I RULE THE WORLD

Thks for your help

Bob-the-Kuhn, 2017-01-30:
You don't hear "SIMPLE AND PERFECT" very often.

Glad to see you're running the way you want to.

Bob-the-Kuhn, 2017-01-31:
If you're done then please close your issue(s).

closed 2017-04-02 by thinkyhead