MarlinFirmware/Marlin #5759: is there someone to code something for me please  ?

studiodyne, 2017-01-29:
EASY ! Promise

It's a configuration swap menu item .... I explain

It's just to have CYCLOPS/CHIMERA/DIAMOND all in one firmware ... Why ?

Case 1 : If you are in Cyclops Firm , impossible to change the carrier with CHIMERA , the second heater is not active and impossible to load/unload or use the left without changing THE FIRMWARE
Case 2 : If you are in Chimera Firm , impossible to change the carrier without refirm the board , the second motors wait the temps that will never up
Case 3 : Diamonds hotends , same problem

The solution

Add VARIABLES to replace the ' DEFINED heaters and sensors , not the FIRST because always the same '
Add VARIABLES to replace the ' DEFINED extruders hotends OFFSET'
add MENU ITEM for each personnal configuration, defined on the firmware

Ex :
Prepare menu -> Carrier_Config ---> CHIMERA
--->CYCLOPS
----> DIAMONDS
----->custom other config

In the firmware i will enter my parameters , ex
for chimera , i will set the Number of extruders 2, pins for heater 1 + 2 , sensors 1 + 2 , and the extruders offsets
For cyclops , i trick , i will set Number of extruder 2 , pins heaters and sensors the same , offset to zero
For diamonds i trick , 3 extruders , but same heaters , same pins

To program this , i have to put variable to replace the ' #define Heaters .... ' , but , i don't know where to make declaration , i always have a compilation message , and i'm not enough experimented , to put the last config variable datas in the EEPROM storing/ reading process

Is there someone , to help me to code it , where to put the declaration or the way to give an issue to this

For the moment , i only use chimera/cyclops , and i put the 2 resistors ' heaters ' inside the cyclops ' 2 x40 watts' , very power full , i use the 2 sensors too in the same hotend , it's perfect , but , i need a software solution
I don't want computer close to the machines to refirm , i don't want a computer to use gcode instructions , i don't want to trick the slicer with instruction everywhere
I JUST ONE TO SWITCH FOR EACH CARRIER and i just want my printer and my sdcard , it's all

Thk you very much , for all that have read all

Steeve
i know c++ language , but i don't know the rules and the way of programming , i can read and modify , but for the moment , my knowledge too poor , to create , to make a program

Roxy-3D, 2017-02-02:
At a higher level, what are you trying to accomplish?

studiodyne, 2017-02-02:
Simple , i want the possibility to change the extrusion system on the carriage , without refirm
For that , the firm must have many configuration inside selectable by lcd menu
If i use the CYCLOPS extrusion system , or a DIAMONDS or a DUAL V6HOTEND ,  some settings don't exist on LCD

For each configurations, the only setting i need  to change
Number of hotends and number of extruder ' direct in firmware , we set our max number of extruders '
Offset between each ' already possible with M218 i will place on a ITEM MENU '
PINS of heaters/sensors ' for cyclops and diamonds only one sensor/heater same for all extruder ' , for dual hotends ' two  heaters and two sensors '
BUT IMPOSSIBLE to  change the pin attribution or , anything that permit to set a ' pin configuration ' by LCD

For this , we have to declare the heater 1 + 2 + 3 , and sensor 1+2+3 as ' variable' ,the first hotend/heater/sensor is always the same ' no need to modify it '  for the moment they are ' constant ' ,  when this done ,  we can change the hotends configuration as we want and without load the right firmware everytime

EASY TO CODE

For EXAMPLE 

lcd -- tune --- ' EXTRUSION SYSTEM CFG ' --- Config ' Cyclops ' , and in one click the software , declare the same pin for all heaters and sensors , declare 2 extruders and one hotend , declare the new gear step ,

lcd -- tune --- ' EXTRUSION SYSTEM CFG ' --- Config ' Chimera '  , the software ,  set 2 hotends ,2  extruders , 2 different heater and sensor , and all i have written in the firmware for this config

Etcc

Simple !

Roxy-3D, 2017-02-04:
> EASY TO CODE
> For EXAMPLE
> lcd -- tune --- ' EXTRUSION SYSTEM CFG ' --- Config ' Cyclops ' , and in one click the software , declare the same pin for all heaters and sensors , declare 2 extruders and one hotend , declare the new gear step 
> lcd -- tune --- ' EXTRUSION SYSTEM CFG ' --- Config ' Chimera ' , the software , set 2 hotends ,2 extruders , 2 different heater and sensor , and all i have written in the firmware for this config
> Etc
> Simple !

Please show us how simple it is.   Preferably, fork the RCBugFix branch because that will make it easier to merge your valuable contributions to the code base.

MoonshineSG, 2017-02-09:
i tried to do something similar, having a "special" M code to switch between Cyclops and Chimera, but even though the idea is simple, the implementations is not.  

It used to work before the "SWITCHING_EXTRUDER" and "MIXING_EXTRUDER" was introduced to Marlin, but after that, things got more complicated and I gave up  (see https://github.com/MarlinFirmware/Marlin/issues/3980) . In theory it should have been easier. Most of the pieces are there, just need someone smart(er) to put them together. 

_Now I have a "simple" firmware switch. I always compile 2 versions of the firmware (`-prefs=build.extra_flags="-DSINGLENOZZLE"`) and I have "smart" hotends that "tell" the RaspberryPi to switch the firmware to the correct one. It even switches to GRBL (sorry Marlin!) and replaces OctoPrint with cnc.js if it's Spindle or Laser :D

thinkyhead, 2017-02-12:
I concur that a major overhaul of the code would be required to support such flexibility. It will also tend to add some performance overhead to be checking these extra conditions during printing.

rjacobs1969, 2017-06-07:
Came here thanks to a Google search.. the OP describes exactly my problem.
I made my hotends easy swappable.
Currently I have single and dual nozzle setup, and it works great... but I´ve bought a diamond (mixing) hotend as well. As far I understand the documentation I cannot switch between dual hotends and the diamond without flashing marlin with a different configuration. 
Ideally I would like to create a dual nozzle setup with one of the nozzles a diamond. (color mixing + a separate support material nozzle) 

I THINK everything I want could be possible if Mixing Extruder would support multiple nozzles

a config would then be something like 

#define EXTRUDERS 4 // total extruders 4 ( 3 mixing, 1 additional) 
#define HOTEND_OFFSET_X {0.0, 0.0, 0.0, 30.0} // first three nozzles (mixing nozzle) share the same position, nozzle 4 is on a physically different location 
#define MIXING_STEPPERS 3  // defining Extruder 0,1,2 as mixing

I hope one of the devs likes the challenge to make this happen

Roxy-3D, 2017-06-07:
The big problem is very few people have the required hardware to get this setup and tested.  Some Mixing Extruder vendor should start posting here and decide where they want to send some hardware.

Because without hardware to test against...   This code probably won't get developed.  It is too complicated to try to get right without something to test against.

closed 2018-02-18 by Studiodyne