MarlinFirmware/Marlin #5764: Velleman k8200

reblair, 2017-02-01:
The Marlin/pins_K8200.h must be wrong.  It seems the y direction is reversed causing the printer to crash into the high y end and ignore the stop switch when sent home.

mperdue, 2017-02-02:
Have you tried reversing the Y direction?

#define INVERT_Y_DIR true

thinkyhead, 2017-02-12:
Or, the example configuration for K8200 has `INVERT_Y_DIR` set incorrectly. Or, the stepper is plugged in the wrong way.

This is a question for @CONSULitAS

closed 2017-02-12 by thinkyhead

CONSULitAS, 2017-02-12:
@reblair @thinkyhead 

The example configuration for K8200 works right now on my printer.

The printer worked without changes on the Stepper cables with stock firmware. So I would say mine is OK. And so the example is ok, too.

@reblair Please try the stock firmware to check this. If it shows similar behavior, please fix your stepper cables. They are soldered wrong. Check the build instructions to be sure: http://www.k8200.eu/manual/building/?c=016