MarlinFirmware/Marlin #5765: Velleman k8200

reblair, 2017-02-01:
The y axis must be reversed in pins_K8200.h .  The printer crashes into the high y limit and ignores the stop switch.  If you push the high y stop switch steps are stopped in the decreasing y direction.

Bob-the-Kuhn, 2017-02-02:
You can fix this in either of two ways:
- Software - in Configuration.h, change **INVERT_Y_DIR** to the opposite of it's current setting
- Hardware - switch the stepper wires on **ONE** of the coils (not both)

reblair, 2017-02-02:
For the sake of others I would hope that it might be resolved in a way that
would prevent other K8200 owners from having to hack the code.  Was this
issue there in earlier versions or has it crept in recently?

Bob-the-Kuhn, 2017-02-02:
This is normal.  It's not specific to your board.  The expectation is you'll determine if the motors are turning as desired and then apply one of the above fixes to the one(s) that don't.

Usually when you wire up the stepper motors to the driver you really don't know which way they'll turn.  This is because the usual wiring instructions are similar to:
1. Identify coils 1 and 2 by twisting two wires together until you find a pair that makes the stepper harder to turn by hand.  That's one coil.  The other two wires go to the other coil .
2. Wire one coil to the A1 and A2 driver pins and the other to the B1 and B2 driver pins.  It doesn't matter which color of wire goes to A1 or A2 or which color of wire goes B1 or B2.

Since the wire colors aren't specified then you don't know which way the steppers will rotate.  

I don't remember if wire colors change from vendor to vendor or model to model.  I'd be surprised if there was an industry standard.

Bob-the-Kuhn, 2017-02-02:
I remember what I missed ...

"hack the code" is NOT what you're doing.  The files **Configuration.h** and **Configuration_adv.h** are **EXPECTED** to be modified by the user to suit his machine.

**pins_YOUR_BOARD.h** is another file the user can modify to suit his machine.  You usually don't need to touch it unless you have an LCD on your machine.  There's so many different LCD controllers out there that it would be a huge task to make each pins_XXXX.h file compatible with every one of them.

Please use the file [Configuring Marlin]( http://marlinfw.org/docs/configuration/configuration.html) as the starting point in your Marlin learning journey.  Ignore the certificate warning that pops up when you click the link.

thinkyhead, 2017-02-12:
As for what the correct configuration settings should be, that is a question for @CONSULitAS.

CONSULitAS, 2017-02-12:
@reblair @thinkyhead 

The example configuration for K8200 works right now on my printer.

The printer worked without changes on the Stepper cables with stock firmware. So I would say mine is OK. And so the example is ok, too.

@reblair Please try the stock firmware to check this. If it shows similar behavior, please fix your stepper cables. They are soldered wrong. Check the build instructions to be sure: http://www.k8200.eu/manual/building/?c=016

see #5764

thinkyhead, 2017-02-14:
Thanks @CONSULitAS!

@reblair No guarantee that every K8200 is made the same, but we hope they get the wiring right!

reblair, 2017-02-14:
Hmmm - mine works fine with the firmware provided by Velleman but not what
was in the git repository as of my post.  The y is upside down in steps
between the two.  Ir's clearly not a wiring issue.

CONSULitAS, 2017-02-14:
@reblair 

Strange!

Have you tried this fork&branch in my repository?

https://github.com/CONSULitAS/Marlin-K8200/tree/K8200_stable_2016-11-18

This should be ready to compile and run.

Please give feedback, because i am a bit busy now and have no time to test. 😞

thinkyhead, 2017-02-15:
We'll need to test with one or two more K8200 machines to determine what the correct `INVERT_Y_DIR` setting should be.

reblair, 2017-02-15:
This firmware appears to be fine.  I didn't do extensive testing, but the y
axis issue is not present with this version.  Thanks.

CONSULitAS, 2017-02-15:
@reblair 
Thanks for the feedback. I will test the current RCBugfix as soon as possible. Perhaps something changed or is wrong. 😄

closed 2017-10-31 by thinkyhead