MarlinFirmware/Marlin #5767: Feature Request

balbotizy, 2017-02-02:
Request of the laser section with the possibility of associating a pin (44) via the display where I have the possibility to turn off/on  the laser and adjust the intensity in pwm (0-255) and display the value on the display

FHeilmann, 2017-02-02:
Sounds like the functionality that is present for the Part-cooling fan. You can turn it on/off using the menu, and its current speed/intensity is displayed on the display.

Does the PWM output have to power the laser or just control it? If it is the latter I bet you can already accomplish most of what you're trying to do without any SW modifications.

balbotizy, 2017-02-02:
I controll the pwm pin 44 to 0-255 (0-5v) for adjust a laser power in function of the material that engroved.
Is very nice a similar control fan pin but at 44 pwm pin

FHeilmann, 2017-02-02:
You can use the `pins_YOURBORDNAME.h` to remap the FAN_PIN to pin 44 instead of whatever is there.

balbotizy, 2017-02-02:
But in this case the fan_pin is out? i need two controll, fan pin and 44 pwm pin

FHeilmann, 2017-02-02:
So the only thing that is really missing is the display of the current state?

balbotizy, 2017-02-02:
No there isn t this section. i want is possible to replicate a controll fan pwm in pin 44 and into the display i read a value of pwm (0-255)

Bob-the-Kuhn, 2017-02-02:
See PR #5145 .  It implements M3, M4 & M5 spindle control commands.

These should be able to do all that you're asking. 

They use up to 3 pins:
- on/off
- hardware PWM for speed (0-255)
- direction

If you want I can attach the code to this issue.

balbotizy, 2017-02-02:
Yes, i modify a pin in 44 , direction is not request: on 0v off 5v and pwm setup 0-255

Bob-the-Kuhn, 2017-02-03:
Here's the code for the M3, M4 & M5 commands.

I've added it to the latest RCBugFix.  Only Configuration_adv.h & Marlin_main.cpp are changed.

I took the liberty of changing the equation in Configuration_adv.h so that S100 gives you 100% duty cycle & S0 gives you zero.  I've also turned off the direction feature (also in Configuration_adv.h).

[spindle control.zip](https://github.com/MarlinFirmware/Marlin/files/749475/spindle.control.zip)

balbotizy, 2017-02-03:
Thanks a lot , but i controll the 44 pin pwm ?

Bob-the-Kuhn, 2017-02-03:
Yes, just put the following in your pins_XXXX.h file
`#define SPINDLE_SPEED_PIN  44  // hardware PWM`

The code assumes there is an on/off signal.  You'll also need to add one of the following to your pins_XXXX.h file
```
#define SPINDLE_ENABLE_PIN     xx   // any digital I/O (if you need an on/off signal)
#define SPINDLE_ENABLE_PIN     yy   // an unused pin (if you DON'T need an on/off signal)
```

Bob-the-Kuhn, 2017-02-03:
The main reason for using the M3, M4, M5 commands is they map the command input to the 0-255 range of the PWM.  This way you can setup the configuration so that M3 S100 could mean 100% or 100 Watts.

thinkyhead, 2017-02-15:
See [this branch](https://github.com/thinkyhead/Marlin/blob/MakerArmV1/Marlin/Marlin_main.cpp#L5041) for an example of a laser implementation.

svogl, 2017-03-11:
@Bob-the-Kuhn , @thinkyhead 
I am working to make the MakeBlock LaserBot ( http://www.makeblock.com/laserBot-robot-kit ) workable - the stock software is based on Marlin 1.0.2 with some quick hacks that enable laser control via M3..M5 as well.

I have a pin_* file and Configuration file ready so far but would like to base the laser stuff on something that goes into mainline instead of rolling  my own.

The controller board is based on an ATmega2560, the laser PWM is on pin D10. The original code simply uses analogWrite() so I assume a ~1kHz modulation frequency, haven't checked with my scope yet though.

If I can test anything, drop me a line
Simon

Bob-the-Kuhn, 2017-03-11:
Take a look further up this thread & you'll find some code from a PR that implements M3, M4 & M5.

The PWM frequency varies with duty cycle.  If I remember correctly it varies for about 20kHz down to 500Hz.

svogl, 2017-03-11:
great, will give it a try!

closed 2017-10-31 by thinkyhead

mjscosta, 2018-12-27:
Hi, I'm setting up a laser into an mpcnc using marlin, and a couple questions popped.

I'm using an arduino due and a RADDS board.

The laser has a PWM signal for power control at 20kHz max.

I'm missing some detail on the PWM, mainly if I understood properly PWMs are driven by timers, and different PWM pins share the same timer. I cannot change a specific pin PWM frequency without affecting other pins, and I'm concerned that trying to do it, it would affect the steppers PWM and timing.

Does this make sense ?

How do I know which pins I can change using setPwmFrequency(pin, val), without impacting the steppers PWM?

Thanks

TwatBurglar, 2019-03-06:
@mjscosta  Did you get your Due board outputting a 20kHz or 25kHz  PWM signal ?

mjscosta, 2019-03-07:
@TwatBurglar still haven't tried to address the issue. The point where I am in this, is that I still need to check the pulse frequency of the PWM, to check if indeed something else is required.

 If it is below 20kHz I wont need to change anything, if not, I'm considering using a "frequency divider circuit" that decreases the PWM frequency, so that I don't need to tweak the the code.

Considering all that, I knew of this issue with Marlin I'd rather use a laser with a voltage power control and connect it to a DAC, avoiding all these PWM issues.

TwatBurglar, 2019-03-07:
@mjscosta Did you try any code changes to get a 20khz pwm from the DUE?    It is my understanding that on Due, Marlin implements only the analogWrite 1khz PWMs.   Due has the hardware capability (**ie. Timer5 on pin 44**), but Marlin doesn't implement yet.

fyi - my K40 laser psu needs TTL input (except for the analog pot pin, which I want to leave as an analog pot), so a DAC won't work.