MarlinFirmware/Marlin #5768: How works the Marlin Firmware ?

clemsept, 2017-02-02:
Hello everybody,

I use Marlin for 6 months, without touching the source code (except the conf files).
But now, I would like to begin with the source code and understand how this firmware works.

However, I don't find documentation or diagrams about that, and my analysis has not resulted in much.
More, the Wiki is very light and there is no code documentation (Class, Methods, Dependencies, etc).

It can be a good idea to develop this type of documentation.
A good documentation can improve the quality of the development with a clear understanding of the environment and how it works, but also motivate new developers to contribute in this project.

Thanks everybody for your job in this project.

Tannoo, 2017-02-02:
Trial and error has been my teacher. Lol

clemsept, 2017-02-02:
@Tannoo _Your comment is needless. Documentation would save time. Analyzing the code without a starting point and without guiding thread is wasting time, and some of us don't have much._

A good documentation can help beginners like me, and structure the project to ensure a good cohesion in the developments.
Greatest OpenSource projects have API references, Building procedures, Code documentation, etc.
It's easy to start, even if the project is stated several years ago. Why don't we start ?

Finally, what the target of a project like Marlin ?
 - To be restricted to a closed community of depressive warrior geeks ?
 - Or to be opened to a large community, with various skills and different levels ?

Roxy-3D, 2017-02-02:
Have you looked at:   https://github.com/MarlinFirmware/MarlinDocumentation
What topics have failed to help you?

By definition, Marlin is not restricted to a closed community of depressive geeks.   Everybody is welcome and we accept good (meaning useful, well thought out and carefully coded) contributions from where ever they appear.

Tannoo, 2017-02-03:
@clemsept, if you had looked around, you would have found the documentation project.

I can't just simply give you my opinion?

I have been using Marlin for about a year and have tried to contribute things I have done to my firmware for features that I use.

BTW, I don't believe Marlin is the code to start with for a beginner in Arduino.

clemsept, 2017-02-03:
Ok thanks for your help :)

closed 2017-02-03 by clemsept

thinkyhead, 2017-02-12:
Please contribute to the documentation project. We are very short on help.