MarlinFirmware/Marlin #5769: Adjustable Maximum Fan speed?

gsart84, 2017-02-02:
Hello 

sorry for my bad english skills.

i trying to get a fan speed maximum limit like this:

Step 1:

In configuration.h, after “#define EXTRUDE_MAXLENGTH (X_MAX_LENGTH+Y_MAX_LENGTH) //prevent extrusion of very large distances.” add this lines:
// Fan speed limit
#define FAN_SPEED_MAX 135

Step 2:

In planner.cpp, search and substitute “analogWrite fanSpeed” with:
analogWrite(FAN_PIN, map(fanSpeed, 0, 255, 0, FAN_SPEED_MAX));

Step 3:

In ultralcd.cpp, search and substitute “analogWrite fanSpeed” with:
analogWrite(FAN_PIN, map(fanSpeed, 0, 255, 0, FAN_SPEED_MAX));

Step 4:

Compile and upload the new firmware

the problem is, this works in older firmware, not in new.

can anyone please help?

Bob-the-Kuhn, 2017-02-03:
The latest Marlin (RCBugFix) definitely doesn't have  "fanSpeed".  I do see "fanSpeed**s**[x]" where x is the index.

Take a look at "#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed" in Configuration_adv.h.  This may do what you want.  

I don't know when this was introduced.  I do know it's in RC8 and later.

gsart84, 2017-02-03:
hello,

Yes, i know the option with Extruder_Auto_Fan_Pin
and this works perfect with my 2 extruder fans.
Now i need to set a maximum speed for my 2 filament cooling fans. I try to get a maximum output voltage of 1V-5V
Example: i set fan speed in marlin to a maximum value of 106. The option for auto fan pin is not good for filament cooling fan.
Any idea?

Bob-the-Kuhn, 2017-02-03:
How about using the CONTROLLERFAN?  If you are going to run them at the same speed then you can hook both to the same output pin.

If they need to be different speeds then create a CONTROLLERFAN2 and add parallel code in Marlin_main.cpp to the existing code:
```
      uint8_t speed = (!lastMotorOn || ELAPSED(ms, lastMotorOn + (CONTROLLERFAN_SECS) * 1000UL)) ? 0 : CONTROLLERFAN_SPEED;

      // allows digital or PWM fan output to be used (see M42 handling)
      digitalWrite(CONTROLLERFAN_PIN, speed);
      analogWrite(CONTROLLERFAN_PIN, speed);
```

Or create your own fan names and add some code to Extruder fan speed routine **checkExtruderAutoFans()** in temperature.cpp

gsart84, 2018-10-07:
Yes. Now There is an Option in configurstion_adv.
define FAN_MAX_PWM 106

closed 2018-10-26 by thinkyhead