MarlinFirmware/Marlin #5770: Feature Request: Custom M code to dock/undock toolhead

wookie666, 2017-02-02:
I have a custom unit that docks and undocks toolheads depending which tool to use. I want to create a custom Mcode that will then run through a series of gcode to dock/undock t0/t1.
Something along the lines of M870 T0 ;dock t0, M870 T1 ;dock t1 , M871 T0 ;undock T0, M871 T1 ;undock T1

But I want to have it so I can use the X max and Z max endstop inputs to determine whether the tools are already docked/undocked and if i try to dock T0 with x enstop triggered then it just doesn't do it.

Anyone able to help, or point me in the right direction?

Roxy-3D, 2017-02-03:
We need a lot more information about what you are trying to accomplish.   For starters, tell us about the different tools and how and why they get 'docked'.     If you are trying to make changes for your Marlin fork, that will help us advise you.   But it seems you want us to add capability to address your needs.   

We are not opposed to doing that.  But we would want to solve the general case so anybody with a similar setup can benefit.   

We need a lot more information before we understand your situation.

wookie666, 2017-02-03:
Thanks for the reply. I'm more than happy to create it myself with some
guidance.

Basically I want to call a gcode. Say G101 T0. I then want this to run a
small group of code. Say about 10 lines worth which will move the head a
specific way to disengage a toolhead. But if for instance X max is
triggered (I will use a endstop in a specific way to see if the tool is
docked already) I don't want the code to run.

So basically I want to create a custom gcode to run a group of G1 move
commands If the endstop is not triggered.

wookie666, 2017-02-03:
This video will show you what I mean 
https://youtu.be/is5lhVlLW2o

Roxy-3D, 2017-02-03:
That doesn't sound too difficult.   One thing to consider...  And it isn't that big of a deal, especially if this is just for you and not going into the main stream Marlin code base:   Typically, 'T' is used for 'T'ool commands.   Not 'G'.   But if you want to use a 'G', it will work just fine.

I would suggest you add a new GCode or TCode to Marlin_Main.cpp to support your movement.   Start small.  Have the new G or T command just move to the center of the bed for starters, and then return to the main command interrupter.   Something like:

```cpp
        case 30: // G30 Single Z probe
          gcode_G30();
          break;
        case 876: // new docking command
          gcode_G876();
          break;
```
and then else where in the file:
```cpp
void gcode_G876() {
          do_blocking_move_to_z(10.00);
          do_blocking_move_to_xy(100.0, 100.0);
        } 
```
If you can get the above code to compile and flash...  When you give it a G876 it should move the nozzle to a height of 10.0mm and then move it to (X,Y)=(100,100).    If you can get that to happen... Then you can add new do_blocking_move_to_xy() commands to get the exact path you need to accomplish your goals.

wookie666, 2017-02-03:
Thanks for that.

How would I put If x_max_endstop = triggered then do nothing else do code
below?

Am i able to use the G876 to Tnnn to dock the tool noted by nnn, and then
G877 Tnnn to undock it. Same thing, using the xmax endstop to either
continue to the void gcode or to just disregard the code.

wookie666, 2017-02-03:
This is what I currently have

void gcode_G876() {

          relative_mode = true;
          do_blocking_move_to_z(10.00);
          relative_mode = false;
          do_blocking_move_to_xy(60.0, 10.0);
          do_blocking_move_to_xy(60.0, -30.0);
          do_blocking_move_to_xy(8.0, -30.0);
          do_blocking_move_to_xy(8.0, 10.0);
          relative_mode = true;
          do_blocking_move_to_z(-10.00);
          relative_mode = false;
        }

How would I add feedrates to that?. And then how would I put a switch in
before hand to check if Xmax is triggered or not triggered?

Tannoo, 2017-02-03:
```
void gcode_G876() {
          bool tool_present = read(X_MAX_PIN);
          relative_mode = true;
          if (tool_present == false) {
            do_blocking_move_to_z(10.00);
            relative_mode = false;
            do_blocking_move_to_xy(60.0, 10.0);
            do_blocking_move_to_xy(60.0, -30.0);
            do_blocking_move_to_xy(8.0, -30.0);
            do_blocking_move_to_xy(8.0, 10.0);
            relative_mode = true;
            do_blocking_move_to_z(-10.00);
            relative_mode = false;
          }
        }
```

Shouldn't this work?

Tannoo, 2017-02-03:
Search for and look at the function of `do_blocking_move_to_xy`. Maybe that will shed some light on a possible feedrate option for that function.p

Roxy-3D, 2017-02-03:
> How would I add feedrates to that?.

The do_blocking_move_to_??() declarations are done in Marlin.h  That tells the compiler what to expect:
```cpp
void do_blocking_move_to(const float &x, const float &y, const float &z, const float &fr_mm_s=0.0);
void do_blocking_move_to_x(const float &x, const float &fr_mm_s=0.0);
void do_blocking_move_to_z(const float &z, const float &fr_mm_s=0.0);
void do_blocking_move_to_xy(const float &x, const float &y, const float &fr_mm_s=0.0);
```
If you look at the syntax, the last parameter of each call is optional.  If a parameter is not specified, the compiler is told to default the value of fr_mm=0.0     So the answer to that question is easy.   You just add one more parameter to specify the desired feed rate in mm / sec.

> And then how would I put a switch in before hand to check if Xmax is triggered or not triggered?

The endstops are checked a lot of places.  But if you look at Endstop.cpp you will find a lot of different activity with regard to endstops.   M119 reports the state of the endstops.  Here is the front end of that function:
```cpp

void Endstops::M119() {
  SERIAL_PROTOCOLLNPGM(MSG_M119_REPORT);
  #if HAS_X_MIN
    SERIAL_PROTOCOLPGM(MSG_X_MIN);
    SERIAL_PROTOCOLLN(((READ(X_MIN_PIN)^X_MIN_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
  #if HAS_X_MAX
    SERIAL_PROTOCOLPGM(MSG_X_MAX);
    SERIAL_PROTOCOLLN(((READ(X_MAX_PIN)^X_MAX_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
```
The expression   ( READ(X_MAX_PIN)^X_MAX_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT ) will evaluate and return a Boolean that can be used in an 'if' statement to decide if the X_MAX endstop is triggered.   You would say something like:
```cpp
  if (READ(X_MAX_PIN)^X_MAX_ENDSTOP_INVERTING) ? MSG_ENDSTOP_HIT ) {
       special logic because of the extruder being pressed against X-Max endstop;
  }
```

wookie666, 2017-02-03:
Cheers for that. Really appreciated

wookie666, 2017-02-09:
Ok. can i get some advice from the guys here 

I have this `
//===========================================================================
//========================= Tool Dock and undock ============================
//===========================================================================
#define TOOL_NUM        2
#define IS_EMPTY(PIN)   (!digitalRead(PIN))
#define DOCKN_PIN       {X_MAX_PIN, Y_MIN_PIN}

#define MOVES_IN_DOCK_SEQUENCE  4 //number of blocking moves in tool docking procedure
      
#define DOCK_X                  {{70, 70, 11, 11},     /*Dock T0 X coords*/    {162, 162, 222, 222}    /*Dock T1 X coords*/    }    //So docking T0 will move to X70 Y10, X70 Y-31, X11 Y-31 X8 Y10, 
#define DOCK_Y                  {{10, -31, -31, 10}, /*Dock T0 Y coords*/    {10, -31, -31, 10}  /*Dock T1 Y coords*/    }    // docking T1 will move to X162 Y10, X162 Y-31, X222 Y-31, X222 Y10
#define DOCK_FR                 {{20000, 20000, 3600, 3600}, /*Dock T0 Feedrates*/    {20000, 20000, 3600, 3600}    /*Dock T1 Feedrates*/    }   
#define Z_LIFT_SPEED            1200 //Z lift/lower speed in mm/min
`

In my Config H

and i have this 
`/**
 * Start dock Command
 */

inline void tool_dock_undock(bool undock) {
  
  int tool = -1;
  int dock = -1;

  int dockPin[] = DOCKN_PIN;
  float dockX[][MOVES_IN_DOCK_SEQUENCE] = DOCK_X;
  float dockY[][MOVES_IN_DOCK_SEQUENCE] = DOCK_Y; 
  float dockFR[][MOVES_IN_DOCK_SEQUENCE] = DOCK_FR; 

  //if (! (axis_known_position[X_AXIS] && axis_known_position[Y_AXIS]) )
  //{
  // SERIAL_ECHO_START;
  //  SERIAL_ECHOLN("Please home before calling");
  //  return; // abort, since we don't know where we are
  //}

  //check that the selected tool is valid (i.e T5 is invalid if we only have two tools)
  if(tool < 0 || tool >= TOOL_NUM) {
    //warn and exit if invalid
    SERIAL_ECHO("Warning! Please pick a tool from 0 to ");
    SERIAL_ECHOLN(TOOL_NUM - 1);
    return;
  }

  if (code_seen('T')) {
     tool = code_value_int();
  } else {
    //if no tool is specified we send a message and exit here
    SERIAL_ECHOLN("Please specify a tool using the T parameter.");
    return;
  }

  //check if the dock is empty (if docking) or full (if undocking)
  if(undock ? !IS_EMPTY(dockPin[tool]) : IS_EMPTY(dockPin[tool])) {
    dock = tool;
  } else {
    //that tool is already docked, error and exit
    if(!undock) {
      SERIAL_ECHOLN("Tool is currently in that dock!");
    } else {
      SERIAL_ECHOLN("No tool detected in that dock!");
    }
    return;
  }

  //if dock is empty, dock there
  if(dock != -1) {

    //"Docking T0..."
    if(undock) {
      SERIAL_ECHO("Undocking ");
    } else {
      SERIAL_ECHO("Docking ");
    }
    SERIAL_ECHO("T");
    SERIAL_ECHO(tool);
    SERIAL_ECHOLN("...");
    
    //do blocking moves to dock
    relative_mode = true;
    do_blocking_move_to_z(10.00, MMM_TO_MMS(Z_LIFT_SPEED));
    relative_mode = false;

    int loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-1) : 0);
    int loopEnd   = (undock ? -1 : MOVES_IN_DOCK_SEQUENCE);
    int loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {                  //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);      
    }                                                                   
                                                                  
    relative_mode = true;
    do_blocking_move_to_z(-10.00, MMM_TO_MMS(Z_LIFT_SPEED));
    relative_mode = false;

    if(undock) {
      SERIAL_ECHO("Undocking done.");
    } else {
      SERIAL_ECHO("Docking done.");
    }
  }
}

/**
 * End Dock Command
 */`

and 
`
      #if ENABLED(BEZIER_CURVE_SUPPORT)
        // G5
        case 5: // G5  - Cubic B_spline
          gcode_G5();
          break;
      #endif // BEZIER_CURVE_SUPPORT
      

case 6: // G6  
        tool_dock_undock(false);
        break;

case 7: // G7
        tool_dock_undock(true);
        break;
        
      #if ENABLED(FWRETRACT)
        case 10: // G10: retract
        case 11: // G11: retract_recover
          gcode_G10_G11(codenum == 10);
          break;
      #endif // FWRETRACT`

in my marlin_main.cpp

For this to work, do i need to enable Use_x_max_plug and Use_y_min_plug for this to work at all? As i assume without those two enabled, then xmaxplug and yminplug are not assigned to a pin.

wookie666, 2017-02-09:
Can I get some advice on this

I have this code in my config.h

`//===========================================================================
//========================= Tool Dock and undock ============================
//===========================================================================
#define TOOL_NUM        2
#define IS_EMPTY(PIN)   (!digitalRead(PIN))
#define DOCKN_PIN       {X_MAX_PIN, Y_MIN_PIN}

#define MOVES_IN_DOCK_SEQUENCE  4 //number of blocking moves in tool docking procedure
      
#define DOCK_X                  {{70, 70, 11, 11},     /*Dock T0 X coords*/    {162, 162, 222, 222}    /*Dock T1 X coords*/    }    //So docking T0 will move to X70 Y10, X70 Y-31, X11 Y-31 X8 Y10, 
#define DOCK_Y                  {{10, -31, -31, 10}, /*Dock T0 Y coords*/    {10, -31, -31, 10}  /*Dock T1 Y coords*/    }    // docking T1 will move to X162 Y10, X162 Y-31, X222 Y-31, X222 Y10
#define DOCK_FR                 {{20000, 20000, 3600, 3600}, /*Dock T0 Feedrates*/    {20000, 20000, 3600, 3600}    /*Dock T1 Feedrates*/    }   
#define Z_LIFT_SPEED            1200 //Z lift/lower speed in mm/min`

wookie666, 2017-02-09:
And this is my marlin main

`case 6: // G6  
        tool_dock_undock(false);
        break;

case 7: // G7
        tool_dock_undock(true);
        break;`

and 
`/**
 * Start dock Command
 */

inline void tool_dock_undock(bool undock) {
  
  int tool = -1;
  int dock = -1;

  int dockPin[] = DOCKN_PIN;
  float dockX[][MOVES_IN_DOCK_SEQUENCE] = DOCK_X;
  float dockY[][MOVES_IN_DOCK_SEQUENCE] = DOCK_Y; 
  float dockFR[][MOVES_IN_DOCK_SEQUENCE] = DOCK_FR; 

  //if (! (axis_known_position[X_AXIS] && axis_known_position[Y_AXIS]) )
  //{
  // SERIAL_ECHO_START;
  //  SERIAL_ECHOLN("Please home before calling");
  //  return; // abort, since we don't know where we are
  //}

  //check that the selected tool is valid (i.e T5 is invalid if we only have two tools)
  if(tool < 0 || tool >= TOOL_NUM) {
    //warn and exit if invalid
    SERIAL_ECHO("Warning! Please pick a tool from 0 to ");
    SERIAL_ECHOLN(TOOL_NUM - 1);
    return;
  }

  if (code_seen('T')) {
     tool = code_value_int();
  } else {
    //if no tool is specified we send a message and exit here
    SERIAL_ECHOLN("Please specify a tool using the T parameter.");
    return;
  }

  //check if the dock is empty (if docking) or full (if undocking)
  if(undock ? !IS_EMPTY(dockPin[tool]) : IS_EMPTY(dockPin[tool])) {
    dock = tool;
  } else {
    //that tool is already docked, error and exit
    if(!undock) {
      SERIAL_ECHOLN("Tool is currently in that dock!");
    } else {
      SERIAL_ECHOLN("No tool detected in that dock!");
    }
    return;
  }

  //if dock is empty, dock there
  if(dock != -1) {

    //"Docking T0..."
    if(undock) {
      SERIAL_ECHO("Undocking ");
    } else {
      SERIAL_ECHO("Docking ");
    }
    SERIAL_ECHO("T");
    SERIAL_ECHO(tool);
    SERIAL_ECHOLN("...");
    
    //do blocking moves to dock
    relative_mode = true;
    do_blocking_move_to_z(10.00, MMM_TO_MMS(Z_LIFT_SPEED));
    relative_mode = false;

    int loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-1) : 0);
    int loopEnd   = (undock ? -1 : MOVES_IN_DOCK_SEQUENCE);
    int loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {                  //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);      
    }                                                                   
                                                                  
    relative_mode = true;
    do_blocking_move_to_z(-10.00, MMM_TO_MMS(Z_LIFT_SPEED));
    relative_mode = false;

    if(undock) {
      SERIAL_ECHO("Undocking done.");
    } else {
      SERIAL_ECHO("Docking done.");
    }
  }
}

/**
 * End Dock Command
 */`

Do i need to enable X_MIN_PLUG and Y_MIN_PLUG for this to work. I assume it doesn't assign a pin if theyre not enabled.

The only thing I see is that this would interfere with the code and think that it should stop when those switches are hit.

The switches will be in the tool dock, so it knows if the tool is docked or not, before running the code.

wookie666, 2017-02-09:
Im sorry about the code. I cant seem to get it to paste it all as code for some reason

wookie666, 2017-02-09:
`#define TOOL_NUM        2
#define IS_EMPTY(PIN)   (!digitalRead(PIN))
#define DOCKN_PIN       {X_MAX_PIN, Y_MIN_PIN}

#define MOVES_IN_DOCK_SEQUENCE  4 //number of blocking moves in tool docking procedure
      
#define DOCK_X                  {{70, 70, 11, 11},     /*Dock T0 X coords*/    {162, 162, 222, 222}    /*Dock T1 X coords*/    }    //So docking T0 will move to X70 Y10, X70 Y-31, X11 Y-31 X8 Y10, 
#define DOCK_Y                  {{10, -31, -31, 10}, /*Dock T0 Y coords*/    {10, -31, -31, 10}  /*Dock T1 Y coords*/    }    // docking T1 will move to X162 Y10, X162 Y-31, X222 Y-31, X222 Y10
#define DOCK_FR                 {{20000, 20000, 3600, 3600}, /*Dock T0 Feedrates*/    {20000, 20000, 3600, 3600}    /*Dock T1 Feedrates*/    }   
#define Z_LIFT_SPEED            1200 //Z lift/lower speed in mm/min`

Roxy-3D, 2017-02-09:
> Do i need to enable X_MIN_PLUG and Y_MIN_PLUG for this to work. I assume it doesn't assign a pin if they're not enabled.

Yes.   The lines
```cpp
  #define DOCKN_PIN       {X_MAX_PIN, Y_MIN_PIN}
  int dockPin[] = DOCKN_PIN;
```
are going to try to define DOCKN_PIN as a set of numbers to pre-initialize the dockPin[] array.

wookie666, 2017-02-09:
ok. So by defining X_max_plug and Y_min_plug in Config.h will fix that, but i also assume it will cause issues with the endstops. Ill have to turn off the endstops stopping the movement until i can figure out how to define a whole new pin name for those two pins

Roxy-3D, 2017-02-09:
I don't think you should do that in Configuration.h    You are going to be messing with all the Sanity_Check.h logic and affecting who knows what.    If (for example) you are using a RAMPS board, the various pins are defined for you:
```cpp
//
// Limit Switches
//
#define X_MIN_PIN           3
#define Y_MIN_PIN          14
#define Y_MAX_PIN          15
#define Z_MIN_PIN          18
#define Z_MAX_PIN          19
//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  32
#endif
```
What ever electronics you are using...   Can't you just use the pins you are looking at?   Like X_MIN_PIN or Y_MAX_PIN ?   That is why they are #define in the file.   They are there to use.

wookie666, 2017-02-09:
im using the rumba board. I will have the switches plugged into the Xmax
and Ymin pin, The code i have just checks whether they are active or not,
and then either runs the move commands or doesn't. So if in configuration.h
i define xmax and ymin endstops, then this will work as per normal?

By activitating those two inputs, will that cause issues thinking that xmax
and ymin are hit, while its printing, as the second tool will be docked and
it will be hitting ymin endstop.

wookie666, 2017-02-09:
So far so good. I have it at least reading codes and erroring out when it
should. I haven't hooked it to the printer but so far it's doing what it
should

Roxy-3D, 2017-02-09:
> So if in configuration.h i define xmax and ymin endstops, then this will work as per normal?

No..  In your case...   In the pins_RUMBA.h file you will find these definitions 
```cpp
//
// Limit Switches
//
#define X_MIN_PIN          37
#define X_MAX_PIN          36
#define Y_MIN_PIN          35
#define Y_MAX_PIN          34
#define Z_MIN_PIN          33
#define Z_MAX_PIN          32
//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  32
#endif
```
so if you have some special sensors plugged into the X_MAX or Y_MIN connectors of your board, you would just say something like
```cpp
  if (READ(X_MAX_PIN)==0 ) {
       special logic because of the extruder being pressed against X-Max endstop;
  }
```
or maybe if it is inverted
```cpp
  if (READ(X_MAX_PIN)!=0 ) {
       special logic because of the extruder being pressed against X-Max endstop;
  }
```
Just because you have stuff plugged into the X_MAX or Y_MIN connectors doesn't mean they are doing that function.   You don't want to tell Marlin that stuff is active.    You just want to use those pins using the name Marlin would have used to implement that functionality.   So go ahead and reference the pins that Marlin defined for you but don't turn on those pin's functionality in Configuration.h !!!

wookie666, 2017-02-09:
Could I comment out the xmax pin line and put my own define x dock pin and
y dock pin the. Change my code to use those

Or where would I put your code exactly

wookie666, 2017-02-11:
Thanks Roxy

one other question.

so far its working great, 
The only thing is, i cant seem to get the z axis to read relative mode before and after moving,

I have 
relative_mode = true;
do_blocking_move_to_z(5, Z_LIFT_SPEED);
relative_mode = false;

'some other code

relative_mode = true;
do_blocking_move_to_z(-5, Z_LIFT_SPEED);
relative_mode = false;

but what its doing is moving to z5 then trying to move to z-5 and jamming up.

What is the best way to have relative mode active before and after each z move.

basically i want

G91
G1 Z5
G90

'some other code

G91
G1 Z-5
G90

wookie666, 2017-02-11:
nevermind, I was able to get it to work with

do_blocking_move_to_z(current_position[Z_AXIS]+Z_LIFT_HEIGHT, Z_LIFT_SPEED);
and
do_blocking_move_to_z(current_position[Z_AXIS]-Z_LIFT_HEIGHT, Z_LIFT_SPEED);

wookie666, 2017-02-11:
sorry, another question

How can i get this code to prime the extruder defined by E_PRIME where it currently says 

destination[E_AXIS] += E_PRIME;

`/**
 * Start dock Command
 */

inline void tool_dock_undock(bool undock) {
  
  int tool = -1;
  int dock = -1;

  int dockPin[] = DOCKN_PIN;
  float dockX[][MOVES_IN_DOCK_SEQUENCE] = DOCK_X;
  float dockY[][MOVES_IN_DOCK_SEQUENCE] = DOCK_Y; 
  float dockFR[][MOVES_IN_DOCK_SEQUENCE] = DOCK_FR; 
  float cleanX[][MOVES_IN_CLEAN_SEQUENCE] = CLEAN_X;
  float cleanY[][MOVES_IN_DOCK_SEQUENCE] = CLEAN_Y; 
  float cleanFR[][MOVES_IN_DOCK_SEQUENCE] = CLEAN_FR; 

  if (! (axis_known_position[X_AXIS] && axis_known_position[Y_AXIS]) )
  {
   SERIAL_ECHO_START;
    SERIAL_ECHOLN("Please home before calling");
    return; // abort, since we don't know where we are
  }
   
   
   if (code_seen('T')) {
     tool = code_value_int();
  } else {
    //if no tool is specified we send a message and exit here
    SERIAL_ECHOLN("Please specify a tool using the T parameter.");
    return;
  }

  
  //check that the selected tool is valid (i.e T5 is invalid if we only have two tools)
  if(tool < 0 || tool >= TOOL_NUM) {
    //warn and exit if invalid
    SERIAL_ECHO("Warning! Please pick a tool from 0 to ");
    SERIAL_ECHOLN(TOOL_NUM - 1);
    return;
  }

  

  //check if the dock is empty (if docking) or full (if undocking)
  if(undock ? !IS_EMPTY(dockPin[tool]) : IS_EMPTY(dockPin[tool])) {
    dock = tool;
  } else {
    //that tool is already docked, error and exit
    if(!undock) {
      SERIAL_ECHOLN("Tool is currently in that dock!");
    } else {
      SERIAL_ECHOLN("No tool detected in that dock!");
    }
    return;
  }

  //if dock is empty, dock there
  if(dock != -1) {

    //"Docking T0..."
    if(undock) {
      SERIAL_ECHO("Undocking ");
    } else {
      SERIAL_ECHO("Docking ");
    }
    SERIAL_ECHO("T");
    SERIAL_ECHO(tool);
    SERIAL_ECHOLN("...");
    
    //do blocking moves to dock
   
    do_blocking_move_to_z(current_position[Z_AXIS]+Z_LIFT_HEIGHT, Z_LIFT_SPEED);
    

    int loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-1) : 0);
    int loopEnd   = (undock ? 0 : MOVES_IN_DOCK_SEQUENCE);
    int loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {           //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);
      
            } 
                                                                              
 if(undock){
    int cleanStart = (0);
    int cleanEnd   = (MOVES_IN_CLEAN_SEQUENCE);
    int cleanInc   = (1); 

    
   destination[E_AXIS] += E_PRIME; 
    for(int i = cleanStart; i != cleanEnd; i+=cleanInc) {           //Perform Movements in X,Y and Feedrates
        do_blocking_move_to_xy(cleanX[dock][i], cleanY[dock][i], cleanFR[dock][i]);
      } 
   }                                                              
    
    do_blocking_move_to_z(current_position[Z_AXIS]-Z_LIFT_HEIGHT, Z_LIFT_SPEED);
    

    if(undock) {
      SERIAL_ECHO("Undocking done.");
      SERIAL_EOL;
    } else {
      SERIAL_ECHO("Docking done.");
      SERIAL_EOL;
    }
  }
}

/**
 * End Dock Command
 */`

Roxy-3D, 2017-02-11:
> sorry, another question
> How can i get this code to prime the extruder defined by E_PRIME where it currently says destination[E_AXIS] += E_PRIME;

The extruder is just one more axis of movement.   The M600 command is messing with the LCD Panel, but here is how it does the filament prime:
```cpp
    #if defined(FILAMENT_CHANGE_EXTRUDE_LENGTH) && FILAMENT_CHANGE_EXTRUDE_LENGTH > 0
      do {
        // Extrude filament to get into hotend
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_EXTRUDE);
        destination[E_AXIS] += FILAMENT_CHANGE_EXTRUDE_LENGTH;
        RUNPLAN(FILAMENT_CHANGE_EXTRUDE_FEEDRATE);
        stepper.synchronize();
        // Ask user if more filament should be extruded
        KEEPALIVE_STATE(PAUSED_FOR_USER);
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_OPTION);
        while (filament_change_menu_response == FILAMENT_CHANGE_RESPONSE_WAIT_FOR) idle(true);
        KEEPALIVE_STATE(IN_HANDLER);
      } while (filament_change_menu_response != FILAMENT_CHANGE_RESPONSE_RESUME_PRINT);
    #endif
```
and the UBL code base does a nozzle prime with this  (it knows it is sitting on top of a Mesh Bed Leveling scheme so it uses mesh_buffer_line().  You can just use planner.buffer_line().
```cpp
		set_destination_to_current();
		destination[E_AXIS] += Prime_Length;
		mesh_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], planner.max_feedrate[E_AXIS]/(15.0), 0 );
		stepper.synchronize();
		set_destination_to_current();
```

wookie666, 2017-02-12:
Hey Roxy,

can you please help me if possible.

I have this in my code
```
 // Define runplan for move axes
   
      #define RUNPLAN(RATE_MM_S) line_to_destination(MMS_TO_MMM(RATE_MM_S));
  

    
    //do blocking moves to dock
   
    do_blocking_move_to_z(current_position[Z_AXIS]+Z_LIFT_HEIGHT, Z_LIFT_SPEED);
    

    int loopStart = (undock ? (MOVES_IN_DOCK_SEQUENCE-1) : 0);
    int loopEnd   = (undock ? 0 : MOVES_IN_DOCK_SEQUENCE);
    int loopInc   = (undock ? -1 : 1);
    
    for(int i = loopStart; i != loopEnd; i+=loopInc) {           //Perform Movements in X,Y and Feedrates
      do_blocking_move_to_xy(dockX[dock][i], dockY[dock][i], dockFR[dock][i]);
      
            } 
                                                                              
 if(undock){
    int cleanStart = (0);
    int cleanEnd   = (MOVES_IN_CLEAN_SEQUENCE);
    int cleanInc   = (1); 

   destination[E_AXIS] += E_PRIME;
    RUNPLAN(E_PRIME_FEEDRATE);
    stepper.synchronize();
```
What this does is slowly move back towards home, and it doesn't extrude, even with the hotends heated. 

Any idea what i need to add, I have tried searching through other parts, and I just cant understand it, could you please show me exactly what i need to add to have it extrude 'E_PRIME' by E_PRIME_FEEDRATE and if i can remove the runplan work from there as im not sure if that is needed, i just copied it from the filament change function.

Thanks.

Roxy-3D, 2017-02-12:
I don't use the RUNPLAN macro.   Once a programmer finds a way to do things that works, they tend to replicate that method.  I would use:
```cpp
            set_destination_to_current();
            destination[E_AXIS] += E_PRIME;
            planner.buffer_line(
                 current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS],
                 current_position[E_AXIS], MMM_TO_MMS(E_PRIME_FEEDRATE), active_extruder );
           stepper.synchronize();
           set_destination_to_current();
```

wookie666, 2017-02-12:
Excellent. I'll try it out. Thanks again

wookie666, 2017-02-14:
Thanks Roxy,

I just wanted to let you know, your code helped me, although I found an
error.

it should say currentx, currenty, current z, but then it should say
destination E instead of current e, and I have put my FR in as MM/S anyway,
so i removed that. everything else worked

Thanks.

On Mon, Feb 13, 2017 at 6:17 AM, Christopher Hemmings <chemmings60@gmail.com

thinkyhead, 2017-02-15:
> `MOVES_IN_CLEAN_SEQUENCE`

We have a `NOZZLE_CLEAN_FEATURE` option if you want to use that.

wookie666, 2017-02-15:
Thanks Thinkyhead

I tried the nozzle clean, removed the Lift and lower z function from it, but the speed was just too slow for what i wanted. I also didn't want it to pass over and then back. The issue is i have each tool on the outsides, then the wiper in the middle. So for T0 i wanted it to run between x90 and x140 like >,<,> and tool one i wanted <,>,<

The G12 function didn't do this, it would always start at X90 and end at X90. Im being pedantic i know.

So far its working great, I need to build one more switch function in and then it will be done.

thinkyhead, 2017-04-02:
How's it going @wookie666 ?

wookie666, 2017-04-02:
Hey a
@thinkyhead. All good. I was able to build the code in an use the endstops. Please close this case

marcossfernandez, 2017-07-06:
@wookie666 Hi Wookie666, hello from Spain.

I'm trying to do the similar approach on my CoreXY 3 independent printheads and 3 docks. It work with toolchange gcode, but i would want to integrate in Marlin firmware as you did.

https://photos.google.com/photo/AF1QipPrkprSMp72lGJGt_kH_XNHlfAwQs7Cx9gEU2iv

Can you send me the Marlin files to analise the changes and try to implement in my development? Any help will be welcome ;-)

marcossoutofernandez (at) gmail (dot) com

Thank you!

closed 2018-02-06 by thinkyhead