MarlinFirmware/Marlin #5771: How to save auto leveled parameter

Jolly2015, 2017-02-03:
Hello friends,I have two questions:
1.I would like to add auto leveling on my kossel,but I don't want to probe everytime before printing,coz if I don't change the level of the bed,the important of auto leveling parameter must be same.So when G29 implemented,what parameter should I store to eeporm for next print without G29?
2.Where can I get the newest FW for delta machine
Thanks very much.

Tannoo, 2017-02-03:
Enable `EEPROM_SETTINGS`, then send M502 (restore factory defaults from the configuration files), then M500 to store the settings..

Tannoo, 2017-02-04:
https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

Jolly2015, 2017-02-04:
@Tannoo  Hi Tannoo, thanks for your reply.I download  marlin 1.1.0-rc version. I opened AUTO_BED_LEVELING_BILINEAR function,after machine probed 9 points ,the LCD shows "Err:ZPROBE",I don't know the reason. 
So  I  delete  the code about this error in main file and test the level, it seem perfect.Then I open EEPROM_SETTINGS, and send M502 and M500,when I restart machine and  re-test the level,it changed, the useful level parameter didn't restored.
Please help me with this two problems,thank you ,thank you

TheCoolDude, 2017-02-04:
Just so you all know,  you might want to read  this "http://marlinfw.org/docs/features/auto_bed_leveling.html".

Tannoo, 2017-02-04:
```
      if (_TRIGGERED_WHEN_STOWED_TEST == deploy) { // State hasn't changed?

        if (IsRunning()) {
          SERIAL_ERROR_START;
          SERIAL_ERRORLNPGM("Z-Probe failed");
          LCD_ALERTMESSAGEPGM("Err: ZPROBE");
        }
        stop();
        return true;
```

This tells me that your probe limit switch is being triggered while it is stowed (parked).
Check to insure you do NOT have that limit triggered when the probe is not in use.
M119 will help with that.

BillStruve, 2017-02-05:
So, using M502 followed by M500 stores the settings, but after powering up, any use of G28 will erase the settings from RAM. How do you get the EPROM settings back into RAM after homing with G28?

hexane360, 2017-02-05:
[M501](http://marlinfw.org/docs/gcode/M501.html) restores EEPROM settings.

However, looking at [Configuration_store.cpp](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/configuration_store.cpp), it appears that bed leveling points are only stored for `MESH_BED_LEVELING`, and not `AUTO_BED_LEVELING_BILINEAR`
```
#if ENABLED(MESH_BED_LEVELING)
      // Compile time test that sizeof(mbl.z_values) is as expected
      typedef char c_assert[(sizeof(mbl.z_values) == (MESH_NUM_X_POINTS) * (MESH_NUM_Y_POINTS) * sizeof(dummy)) ? 1 : -1];
      uint8_t mesh_num_x = MESH_NUM_X_POINTS,
              mesh_num_y = MESH_NUM_Y_POINTS,
              dummy_uint8 = mbl.status & _BV(MBL_STATUS_HAS_MESH_BIT);
      EEPROM_WRITE(dummy_uint8);
      EEPROM_WRITE(mbl.z_offset);
      EEPROM_WRITE(mesh_num_x);
      EEPROM_WRITE(mesh_num_y);
      EEPROM_WRITE(mbl.z_values);
    #else
      // For disabled MBL write a default mesh
      uint8_t mesh_num_x = 3,
              mesh_num_y = 3,
              dummy_uint8 = 0;
      dummy = 0.0f;
      EEPROM_WRITE(dummy_uint8);
      EEPROM_WRITE(dummy);
      EEPROM_WRITE(mesh_num_x);
      EEPROM_WRITE(mesh_num_y);
      for (uint8_t q = 0; q < mesh_num_x * mesh_num_y; q++) EEPROM_WRITE(dummy);
#endif // MESH_BED_LEVELING
```

Bob-the-Kuhn, 2017-02-06:
Just a little bit further you'll find the BILINEAR section.

Looks like they allocate room for all the bed leveling schemes in the EEPROM.  The active one gets real values written for the matrix.  The inactive ones get dummy value written to their matrix.

Jolly2015, 2017-02-06:
My delta printer use micro switch as auto bed leveling and trigger by V5 hit to bed,so I haven't  use ALLEN key and servo, when I choose Z_PROBE_SLED, it said incompatible with DELTA, so  what should I do? canceling this check?

closed 2017-02-07 by Jolly2015

Bob-the-Kuhn, 2017-02-08:
See PR #5781 .  It has code to automatically restore your bed leveling grid/matrix from EEPROM.

I've also attached the two files that were changed in case you're interested in doing some testing.  This is based on last night's RCBugFix.

[ABL auto restore.zip](https://github.com/MarlinFirmware/Marlin/files/759575/ABL.auto.restore.zip)

Jolly2015, 2017-02-11:
@Bob-the-Kuhn  WOW, nice, this is what I really want, I modified to RC8,and test  it a few times,It works fine so far(sorry about I just saw the message this morning ).

Bob-the-Kuhn, 2017-02-11:
Glad to be of help.

Jolly2015, 2017-02-13:
@Bob-the-Kuhn Hi, I tested ABL auto restore those days, I get a problem,when I adjust "z offset" on lcd, It's no effect if I didn't run G29, so the value of  Z offset only take effect after G29, did I missed something?

Bob-the-Kuhn, 2017-02-13:
Did you save to EEPROM after making your "z offset" change?

EEPROM contents are only modified if the user issues a M500 command.  I assume there's an LCD equivalent.

I think the sequence should be
1. modify "z offset"
2. run G29 so you have bed leveling values using the new "z offset"
3. test the new bed leveling 
4. save it to EEPROM

Jolly2015, 2017-02-14:
@Bob-the-Kuhn yeah, of couse,I can see the value was changed and saved, but the sequence you said is no problem,I just feel G29 only need to run one time and Z heigh maybe should adjust many times when do bed leveling what will more convenient,coz once G29 run,the bed already flat,then the effect only relate to Z heigh.Anyway, it's a good function,I like it.(I hope you can understand what I say,though my poor English)

Bob-the-Kuhn, 2017-02-14:
Your English is fine.

See PR#5646 - I believe he wants to do the same.

thinkyhead, 2017-02-14:
> I think the sequence should be
1. modify "z offset"
2. run G29 so you have bed leveling values using the new "z offset"
3. test the new bed leveling 
4. save it to EEPROM

Actually, the bed leveling will work (generally) even if the Z offset is incorrect. Except, I think we added code to `G29` to make a correction if the Z offset can be determined to be inaccurate.

akaJes, 2017-02-14:
hi @Jolly2015 
how do You remove model from bed?
do you have the same temperature/humidity in Your garage/room/classroom  etc?
do you know about metal/wood/plastics deformations after ambient changing?
all this is deforming bed with unpredictable results

webliya, 2017-09-01:
As far as I understand that the feature mentioned below in Marlin features document has not been implemented yet.

"**Bed Leveling and Printing**
Currently, Marlin doesn’t save the bed leveling results to EEPROM, and it throws away the probe results on G28. So the machine must always do a new G29 after any G28. Marlin will be getting a Save Leveling function in the near future, and we’ll make an announcement at that time."

But people talk about send M502 then M500. I believe they mean to send G28 and G29 at between them. Does this mean that Marlin stores the ABL matrix to the EEPROM? If so how we will use it back? That is not clear in any of the messages here or anywhere.

There are some code changes mentioned around we can use but I want to stick to original Marlin code.

If there is a solution other than changing the the code please help.

Roxy-3D, 2017-09-01:
The Mesh Bed Leveling systems (UBL and Bi-Linear) both allow you to save the correction information to EEPROM.

webliya, 2017-09-01:
I find the code which reads the bed levelling but I stil do not understand how to make 3d printer to read it. I try;

M502
G29
M500

But it does not work. After I send G28 and begin to print the Z axis do not move as it moves when I send G29 after G28.

I am totally lost.

Roxy-3D, 2017-09-01:
Have you turned the EEPROM on in Configuration.h ?     There are step by step directions for bringing up UBL here:   https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

webliya, 2017-09-01:
Thank you for the fast response.

I am using ABL and was tired to wait for G29 for each print. Now I found out what to do.

M502
G29
M500

And I added M420 S1 just after G28 in my slicer start script. Now all works fine.