MarlinFirmware/Marlin #5773: PR#2504 broke MAKRPANEL support

unorthodox-engineers, 2017-02-05:
I've been hammering my head on the desk, trying to get RCBugFix to work on my Malyan M150 (A Melzi clone) and finally worked out what the issue is.

Basically, this commit: https://github.com/MarlinFirmware/Marlin/commit/f7e7862588da66b97f2035940f14583342c469aa#diff-8337bbb240d933174bc8593e22eaf681

Broke MAKERPANEL support by changing the structure of the #IF statement from:

#ifdef ULTRA_LCD		
#endif //ULTRA_LCD
#ifdef MAKRPANEL
#endif // MAKRPANEL

to:

#if defined(ULTRA_LCD) && defined(NEWPANEL)
#elif ENABLED(MAKRPANEL)
#endif // MAKRPANEL

So, you changed two seperate #IF statements to an #IF, #ELIF chain. (In the name of "cleaning up", and I assume never tested it) Problem is, the MARKPANEL banch will _never be called_ because in Conditionals_LCD.h you have this:

  #if ENABLED(MAKRPANEL) || ENABLED(MINIPANEL)
    #define DOGLCD
    #define ULTIPANEL
    #define NEWPANEL
    #define DEFAULT_LCD_CONTRAST 17
  #endif

 and:

#if ENABLED(ULTIPANEL)
    #define NEWPANEL  //enable this if you have a click-encoder panel
    #define ULTRA_LCD

So, once MAKRPANEL is switched on, ULTRA_LCD and NEWPANEL  get switched on by the conditionals, so it never gets to the #ELIF MAKRPANEL 

So, basically, the MakrPanel support has been broken for about three years now in the RC branch. Am I seriously the first one to notice this? Sorry if this sounds grumpy, but I've lost days to this now, and when I track down why it turns out to be a freshman noob mistake.

Tannoo, 2017-02-05:
Fix it and submit a pull request.

AnHardt, 2017-02-05:
That's funny.
Since some month im thinking about removing `MAKRPANEL` and renaming 'BOARD_MELZI_MAKR3D' to `BOARD_MELZI_128`, because they never have been in production (https://de.ulule.com/controller-board/  ).
An other hint for their not existence was to get no issues about `MAKRPANEL`.
'BOARD_MELZI_MAKR3D' devoloped a bit into the direction to be a general 1284P MELZI.

But apparently someone was clever enough to produce something similar from the available documentation. (http://makr3d.de/wordpress/wp-content/uploads/2013/05/lcd-schematic.png http://reprap.org/wiki/MaKrPanel )

I'd suggest to replace `MAKRPANEL` with the real name of your panel.

#2504 is from 24 Jul 2015. For me that's more ~17 month than 3 years.

thinkyhead, 2017-02-12:
Patching now. What is your `MOTHERBOARD` setting?

thinkyhead, 2017-02-12:
Please test #5813 when you have a chance. (i.e., https://raw.githubusercontent.com/thinkyhead/Marlin/746160b412094139bdb39416b4ce8805b17f2b20/Marlin/pins_SANGUINOLOLU_11.h)

thinkyhead, 2017-02-12:
> …renaming `BOARD_MELZI_MAKR3D` to `BOARD_MELZI_128`

That's not a bad idea. But for legacy and sanity-check we'd need to keep `BOARD_MELZI_MAKR3D` around (hidden, if we like), and add `BOARD_MELZI_128` with a different ID number.

thinkyhead, 2017-03-04:
Assuming you tested #5813 and it passed.

closed 2017-03-04 by thinkyhead