MarlinFirmware/Marlin #5774: Marlin (RC8) Very unpredictable for preprinting/ freezing mid print

Fickert07, 2017-02-06:
So I moved away from my Smoothieboard and am using an MKS gen 1.4 Ramps board with TMC 2100 Drivers. 

I also have a bltouch mounted and setup as my z endstop. I have a few things that are really messing with any chances of having it reliably print. 

As well I did a PID autotune for the hotend and bed. 

For the Marlin Setup, I watched a youtube video t setup marlin, and I remember setting up a setting that can give a more consistent/ accurate temp for the hotend and bed. Maybe that is the reason for all these issues?

- The bltouch will sometimes home just fine, otherwise it will detect the initial touch, but not the  second one. Then I have to restart the printer to reset it. 
- The Heat bed is next to impossible to get to temp if I just start a print. Say I set my temp to 70 C, it will go up to 65 C, then slow its temperature increase to a craw. on Repeter, it shows the temp goes up by 0.20 C, then down 0.10C over and over. Takes forever (about 15mins for the bed to get to temp). I can easily manually set a temp not in a print and the bed can go to 90+ C. 
- I can set the hotend temp to 255C and it will bounce around till it hits 255. All is good. Once I get up to 260 or 265C though, It will continue to bounce around the temps and never start (I let it run like this for 10 mins and it was bouncing around start temp for about 7 minutes) Very odd.
- Lastly I am using a tft 32 screen with sd card to print. I never had any issues with prints freezing with this on my smoothieboard but on the MKS gen 1.4 it is pretty prone to locking up. If I try to increase temp mid print the print WILL freeze and I have to abort the print. Never had this issue with the smoothieboard. I will say I updated the tft32 screen firmware though. I do no have the old firmware.

FHeilmann, 2017-02-06:
regarding your temperatures, depending on whether or not you are using PID, you have to run a PID autotune and set the correct K I and D values.

Fickert07, 2017-02-06:
I did run a pid tune and did set my K I and D values in the config and uploaded it to the board. I am going to try to find the setting I was told to change when setting up and see if that  helps it at all.

thinkyhead, 2017-02-06:
If using EEPROM don't forget to use `M502`, `M500` to set the new PID values in the EEPROM, otherwise last-saved values will still be in effect.

thinkyhead, 2017-02-06:
> The bltouch will sometimes home just fine

We're working on BLTouch reliability, including asking ANTCLABS to alter the code on the probe itself, because it has issues.

> I am using a tft 32 screen with sd card to print. I never had any issues with prints freezing with this on my smoothieboard but on the MKS gen 1.4 it is pretty prone to locking up.

It shouldn't merely lock up under normal use. Marlin is designed to reset the board if it gets into a locked-up state. Which version of Arduino IDE are you using to build / upload Marlin? You might try another version to see if it exhibits the same locking-up behavior, or if it does something differently.

Fickert07, 2017-02-06:
Good to hear. For PID savings I did an m500 as well as manually entered the PIDS into the code and reuploaded it to the board. My power supply just went out so I am hoping that was an issue with some of this. 

I am using 1.8.1 Arduino for uploading.  I can try another version. Not something that crossed my mind either.

thinkyhead, 2017-02-12:
Recommend you disable `PROBE_DOUBLE_TOUCH` for the BLTouch anyway.

closed 2017-10-31 by thinkyhead