MarlinFirmware/Marlin #5775: Marlin 1.1.0 RC8 resets on second layer, and stopping print

tbowmo, 2017-02-06:
Trying Marlin 1.1.0 rc8 out, I have printed a calibration cube without problems, but trying to print a fan shroud (http://www.thingiverse.com/thing:1496109) the printer resets about 1/3 into the second layer, causing the print to abort.

I'm using repetier-server, running on raspberry-pi, printing via USB. Going back to marlin 1.0.2, and I can print without problems.

troublesome gcode file attached

[bbNewFan.g.gz](https://github.com/MarlinFirmware/Marlin/files/755434/bbNewFan.g.gz)

Tannoo, 2017-02-08:
Reverting to older firmware doesn't find any issues. That only removes some possible protections that are in place on the later ones.

I am printing your file now. I changed the temps to PLA range because I don't have any ABS atm.
Where exactly does it stop? Mine is printing it just fine and I'm running RC8-RCBugFix with other (non-related) PR's included.

If you have an LCD, you should see the error that stopped it. You should also see the reason it stopped in a repetier log.

I believe you had a MINTEMP error. 
My printer is now on Layer #5 and still going...

Tannoo, 2017-02-08:
Layer 27 now.

tbowmo, 2017-02-08:
OK, I'll try rc8 again, and monitor the printer more closely.. (was working
on another project while it was printing..).

thinkyhead, 2017-02-12:
Keep us posted. It would be good to narrow down causes for spurious MINTEMP errors and give a more meaningful error message.

pcarew, 2017-09-27:
I think I'm seeing the same problem.
I'm on 1.1.0-RCBugFix and the firmware crashes/resets on the 2nd layer.

closed 2018-02-06 by thinkyhead