MarlinFirmware/Marlin #5776: after update to newest firmware, temeprature jumps around

TylonHH, 2017-02-06:
Hi,
In the past I used an old modified firmware from an user wich works fine. As I red all the new stuff you provide in the newest release, I gave it a try.
So I downloaded from here and modified it to my coreXY printer.
Then I noticed in S3D that the thermistor from heatbed jumps from -43 to over 700°C "randomly". On my 12864 full graph Display I dont notice that, except that the bed temperature raise some degrees higher then set. Nothin to worry about.

Here is a screenshot:
[image]

But what is realy strange, that I noticed **this** behavior while I was printing a cube:
[image]

So I checked the hardware and also replaced a new thermistor.

Here is my [Configuration.h](http://pastebin.com/URaRfwuf) (pastebin)

What else information do you need to solve this?

Blue-Marlin, 2017-02-06:
If you have activated `ADVANCE` or `LIN_ADVANCE` please deactivate both, retest and report back.

TylonHH, 2017-02-06:
I double checked
`//#define ADVANCE`
`//#define LIN_ADVANCE`
and reuploaded the firmware.

Problem still persist

thinkyhead, 2017-02-06:
Have you done a new `M303 U1` to re-tune PID for the new firmware (followed by `M500`)?

Blue-Marlin, 2017-02-06:
@thinkyhead 
Have you had a look on the graphs? This can not be reasoned by the PID-values. Jumps by 50 and more degrees can only come from the measurement. Temperature systems are much to inert for these jumps.

TylonHH, 2017-02-06:
No I didn't. Because the heatbed makes this trouble and the bed is not PID controlled.
But I run your commands and saved them.

TylonHH, 2017-02-06:
@thinkyhead so you guess its an hardware problem? Maybe some electrical inductive stuff?

3dGriff, 2017-02-08:
@TylonHH I will second @Blue-Marlin on this one.  Although I'm not sure what your heat bed is made of, given its (likely) thermal mass, drops like these are most likely a hardware issue.  Obviously check the connections, but more importantly check the thermistor connection closest to the resistor, they have a tendency to degrade based on our experience.  

On another note, in the past we have seen strange results like this in the past with RAMPS boards.  The issue occurred due to heat on the board caused by running both 12v power and USB.  If you can, let the machine sit (to cool down) and then attempt to heat the bed without the USB connected.  This is just speculation.

mauriciohvargas, 2017-02-10:
Hi, i have the same problem, so thinking it was a hardware problem i uploaded the firmware in other board (arduino Mega with RAMPS) and plugged just the other thermistors and the readings were the same. After some tries with the firmware i just commented the #define COREXY line and there were no weird readings. So i configured the printer firmware with commented #define COREXY line AND UNCOMMENTED #define COREYX and it worked. I am sure there is a bug with the COREXY code.

TylonHH, 2017-02-11:
@3dGriff 
> and then attempt to heat the bed without the USB connected.
But without the USB cable I can't see the graph vie S3D ;-)

3dGriff, 2017-02-11:
@TylonHH it seems that @mauriciohvargas has seen a connection the #define COREXY.  I tried to replicated this but wasn't able to reproduce the results.  I can try again when I get back to the shop, have you attempted what @mauriciohvargas suggested to verify the connection?

thinkyhead, 2017-02-12:
> I am sure there is a bug with the COREXY code.

Possibly. But don't be too sure. Remember the maxim: correlation does not necessarily imply causation. Note that `COREXY` produces precisely the same code as `COREYX`, with only one value negated.

billyd60, 2017-02-13:
I get this kind of behavior as well with rcbugfix but only when I cancel a print from the lcd panel and restart a print again without a power cycle to the printer board. If I don't cancel a print from the lcd panel my temps are rock solid. ps I always print from SD card never USB.

Roxy-3D, 2017-02-13:
> I am sure there is a bug with the COREXY code.

I'm sure there is a problem somewhere...  But I can tell you this:   When I brought up the UBL Bed Leveling system on a normal Prusa i3, absolutely nothing had to be done to make it work on Core-XY machines.   That is simply amazing.   That is only one data point, but I'm of the opinion the Core-XY stuff is 'Correct'.

mauriciohvargas, 2017-02-13:
It has to be a bug, because i switch to 1.1.0 RC6 with the same configuration and there is no problem. The only thing observed is that when the print starts and it's homing, the bed temperature reading drops to 0 and then it rices to normal.
When going again to RC7 or RC8 the problem reappears.

thinkyhead, 2017-02-14:
#5818 might address this.

Anxles, 2017-02-28:
I see there are some doubts about is it a software and hardware problem. By no meens I am a programmer, I even struggle a little with configuration.h to understand all options. BUT I have same problem with RC8, I didn't try M29 but I have problem sa soon as I start the print. I have CORE XY enabled, printer of my own construction on Megatronics V2 board. everything was OK before but I had to make dual Z stepper for torque on Z axis, and I had to update Marlin for that as the old one didn't support that option for my board, I changed all the relevant settings, uploaded firmware, tested all the axis lcd etc (everything looked fine) the problem started when I tried to make a test print. As soon as it started BED temperature became ERRATIC jumping bewteen 400+ C and some LOW (I didn't notice how much) the jumps were sudden a feh hunder degrees )
Long story short I checked the thermistor and connections, Changed to new thermistor and the same. Afterr looking here and found people withe the same problem, I configured RC6 to same parameters and so far for few hours its printing but still I can see some small and very short temp jumps (I have bed 40x40cm 10mm thick so thermal capacity is huge - by no meens possible any sudden temp, jumps)

closed 2017-10-31 by thinkyhead