MarlinFirmware/Marlin #5778: Change Filament Stop Endless Beep - Modification

gordo3di, 2017-02-06:
I modified the Change FIlament code so it doesn't beep nonstop. It will beep 5 times then do one fast double beep and stop.

It's in marlin_main.cpp line 7244 (from my build a few months ago). I'm not the best with git pull requests so i figured i would post this change here if anyone wants it.

 ```` c
#if HAS_BUZZER
      millis_t next_buzz = 0;
      static int runout_beep = 0;
    #endif

    // Wait for filament insert by user and press button
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_INSERT);

    // LCD click or M108 will clear this
    
    wait_for_user = true;
    while (wait_for_user) {
      #if HAS_BUZZER
        millis_t ms = millis();
        if (ms >= next_buzz) {
          if (runout_beep <= 3) { // Only beep 4 times
          BUZZ(300, 2000);
          next_buzz = ms + 2000; // Beep every 2s while waiting
          runout_beep++;
          }
          else if (runout_beep == 4 || runout_beep == 5) { // Two short beeps
          BUZZ(200, 2000);
          next_buzz = ms + 300; // Beep 
          runout_beep++;
          }
        }
      #endif
      idle(true);
    }
````

Roxy-3D, 2017-02-06:
Probably, we need to add some #define's to control this behavior.  Some people are going to want it to beep until the issue is resolved.  Other people are going to want to be alerted to the issue and will get around to changing the filament when ever it makes sense.

Perhaps the thing to do is have a #define that controls the total number of beeps before the printer goes quiet in configuration_adv.h:
```cpp
#define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS   250000L  // Number of alert beeps before printer goes quiet
```
and then change the M600 code to something similar to this:
```cpp
#if HAS_BUZZER
     millis_t next_buzz = 0;
     static unsigned long int runout_beep = 0;
   #endif

   // Wait for filament insert by user and press button
   lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_INSERT);

   // LCD click or M108 will clear this
   
   wait_for_user = true;
   while (wait_for_user) {
     #if HAS_BUZZER
       millis_t ms = millis();
       if (ms >= next_buzz) {
         if (runout_beep <= FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS ) { // Only beep as long as we are supposed to!
         BUZZ(300, 2000);
         next_buzz = ms + 2000; // Beep every 2s while waiting
         runout_beep++;
         }
         else if (runout_beep == FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS-1  ||
                     runout_beep == FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS) { // Two short beeps
         BUZZ(200, 2000);
         next_buzz = ms + 300; // Beep 
         runout_beep++;
         }
       }
     #endif
     idle(true);
   }
```
It might be good to merge these changes into https://github.com/MarlinFirmware/Marlin/pull/5737  That Pull Request allows the nozzle to cool down (and be re-heated) if the user does not change the filament quickly enough.

Tannoo, 2017-02-07:
I moved that "beep code" to it's own function because it's used twice.
```
#if ENABLED(FILAMENT_CHANGE_FEATURE)

  millis_t next_buzz = 0;
  unsigned long int runout_beep = 0;
      
  void filament_change_beep() {
    millis_t ms = millis(); 
    if (ms >= next_buzz) { 
      if (runout_beep <= FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS ) { // Only beep as long as we are supposed to! 
      BUZZ(300, 2000); 
      next_buzz = ms + 2500; // Beep every 2s while waiting 
      runout_beep++; 
      } 
      else if (runout_beep == FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS-1  || 
               runout_beep == FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS) { // Two short beeps 
        BUZZ(200, 2000); 
        next_buzz = ms + 500; // Beep  
        runout_beep++; 
      }
    }
  }

  bool busy_doing_M600 = false;
```
And changed this:
```
    wait_for_user = true;    // LCD click or M108 will clear this
    next_buzz = 0;
    runout_beep = 0;
    for( iii=0; iii<HOTENDS; iii++)      //Save nozzle temps
      temps[iii] = thermalManager.target_temperature[iii]; 

    while (wait_for_user) {
      millis_t ms = millis();
      if (nozzle_timed_out == true)
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_CLICK_TO_HEAT_NOZZLE);
      #if HAS_BUZZER 
        filament_change_beep();
      #endif //HAS_BUZZER

      if (ms >= nozzle_timeout) {
```
And this:
```
KEEP_CHECKING_TEMPS:
    idle();
    for( iii=0; iii<HOTENDS; iii++){
      if (abs(thermalManager.degHotend(iii)-temps[iii]) > 3 ) {
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_WAIT_FOR_NOZZLES_TO_HEAT);
        goto KEEP_CHECKING_TEMPS;
      }
    }

    wait_for_user = true;    // LCD click or M108 will clear this
    next_buzz = 0;
    runout_beep = 0;
    while (wait_for_user) {
      millis_t ms = millis();
      if (nozzle_timed_out == true)
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_INSERT);
      #if HAS_BUZZER
        filament_change_beep();
      #endif
      idle(true);
    }
```

Tannoo, 2017-02-07:
I changed this:
```
      else if (runout_beep > FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS  && 
               runout_beep <= (FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS + 2)) { // Two short beeps 
```
And now the double-beep works.

Roxy-3D, 2017-02-07:
> I changed this:
> ...
> And now the double-beep works.

OK!  Good!  That is the problem with not having any hardware to test things.   You never know for sure your changes are going to work!    I'm looking at what it takes to wire up the Filament Run Out sensor (switch).   It looks like for RAMPS boards, it assumes it is plugged into Servo-3.    Soon...   I should be able to test any changes!

Tannoo, 2017-02-07:
>`#define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS   250000L  // Number of alert beeps before printer goes quiet`

That's a LOT of beeps. lol

Tannoo, 2017-02-07:
BTW: I added this to `endstop.cpp` to help with checking the status of the `filament_runout_sensor`:
```
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    SERIAL_PROTOCOLPGM(MSG_FILAMENT_RUNOUT_SENSOR);
    SERIAL_PROTOCOLLN(((READ(FIL_RUNOUT_PIN)^FIL_RUNOUT_INVERTING) ? MSG_ENDSTOP_HIT : MSG_ENDSTOP_OPEN));
  #endif
```
And this to `language.h`:
```
#define MSG_FILAMENT_RUNOUT_SENSOR          "filament_Runout_Sensor: "
```

Roxy-3D, 2017-02-07:
Can you be talked into modifying the files at:   https://github.com/Roxy-3D/Marlin-Nozzle-cool-down-during-M600  ???

Tannoo, 2017-02-07:
Yeah. give me a few hours. Also have to find out why we have to click twice to start extruding if the nozzle didn't timeout.

Roxy-3D, 2017-02-07:
I didn't know that was happening.  Do you hear a 'Click' for each press?   My guess is the problem is here (in ultralcd.cpp) 
```cpp
void lcd_update() {

  #if ENABLED(ULTIPANEL)
    static millis_t return_to_status_ms = 0;
    manage_manual_move();

    lcd_buttons_update();

    // If the action button is pressed...
    if (LCD_CLICKED) {
      if (!wait_for_unclick) {           // If not waiting for a debounce release:
        wait_for_unclick = true;         //  Set debounce flag to ignore continous clicks
        lcd_clicked = !wait_for_user;    //  Keep the click if not waiting for a user-click
        wait_for_user = false;           //  Any click clears wait for user
        lcd_quick_feedback();            //  Always make a click sound
      }
    }
    else wait_for_unclick = false;
  #endif
```

Tannoo, 2017-02-07:
Well, yes. The clicks were there. But, I think before the nozzle shuts down, it is going straight to the "nozzle heated" screen (which is the same "change filament" screen) and the "nozzle heated" beep on the first click. The second click is to commence extruding.
I do like that code to guarantee a click is heard.

Tannoo, 2017-02-07:
Oh, and BTW, the filmanent runout sensor code DOES initiate the M600 when it's printing. It does nothing otherwise. That is why I added checking the status of it in the M119 code.

Tannoo, 2017-02-07:
Roxy, you made that quite painful. Can you redo the PR against RCBugFix? The RC is soooo out dated. I have all the commits already on your RCBugFix of you `Marlin-Nozzle-cool-down-during-M600` branch.

Tannoo, 2017-02-07:
I believe I got everything port back to the RC branch.

Roxy-3D, 2017-02-07:
Originally...  I made a mistake and put the code in the RC-8 side of the fork.   I later corrected it and moved everything to the RCBugFix side of the fork.    Is that what caused the problem?   It maybe you didn't know that was going on because I think I remember moving a few of your commits from RC over to RCBugFix.

Can you tell me what you know about things?   Is the RCBugFix branch 'Correct' ?  Do I need to do anything to clean up the mess?

Roxy-3D, 2017-02-07:
Ha!!!   We have a few more bugs to fix!   (I've got the new M600 code in the UBL branch I'm bringing up on RC8's RCBugFix.)   Check this out:
```cpp
    void lcd_filament_change_insert_message() {
      START_SCREEN();
      STATIC_ITEM(MSG_FILAMENT_CHANGE_HEADER, true, true);
      STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_1);
      #ifdef MSG_FILAMENT_CHANGE_INSERT_2
        STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_2);
      #endif
      #ifdef MSG_FILAMENT_CHANGE_INSERT_3
        STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_3);
      #endif
      STATIC_ITEM(MSG_FILAMENT_CHANGE_NOZZLE STRINGIFY(HOTENDS), false, true);
      lcd_implementation_hotend_status();
      END_SCREEN();
    ```
The line appears in all of the new LCD Panel routines that display the nozzle temperature.  But what it is displaying is the number of nozzles,  and not which nozzle the temperature data is being shown:
```cpp
   STATIC_ITEM(MSG_FILAMENT_CHANGE_NOZZLE STRINGIFY(HOTENDS), false, true);
```
I think we need to change the new routines to have a line that looks something like:
```cpp
   STATIC_ITEM(MSG_FILAMENT_CHANGE_NOZZLE STRINGIFY(active_extruder), false, true);
```

Roxy-3D, 2017-02-07:
I think everything I know about is resolved...    There were a lot of changes...   We need a few people to load it up and test it.    The good news is, it works correctly on my printer!!!!   :)

thinkyhead, 2017-02-10:
The `STRINGIFY` macro is used to display the literal value of a `#define`.

gsart84, 2017-03-12:
anything is not working here, after i changed filament it says "heating up nozzle... please wait..."
and the printer dont move anymore

closed 2018-02-06 by thinkyhead