MarlinFirmware/Marlin #5782: Startup script in gcode runs abnormally if prior print aborted from LCD panel

billyd60, 2017-02-08:
I am using rcbugfix from 12-6-16 on a Taz 5 printer.

If I stop a print from the lcd menu, and restart it (without rebooting the printer), my startup priming code will not run correctly. Here is the code I use to prime my Taz 5:

G28 ; home all axes
G92 E0 ; zero extruder
G1 Z0.4 ; position nozzle
G1 E25 F300 ; purge nozzle
G1 X5 Y5 Z0.15 F1200 ; slow wipe
G1 X10 Y10 Z0.5 F1200 ; lift
G92 E0 ; zero extruder

Then the main gcode prints.

My end code is nothing special just:

M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G28 X0 ; home X axis
M84 ; disable motors

The first time I run the gcode, the startup code executes correctly. However If I abort a print during the main gcode section, and then restart the print,, the purge nozzle command is ignored... specifically this command: 

G1 E25 F300 ; purge nozzle

The stepper never activates. All other commands in the startup code seem to execute. And then the print prints normally (except with the first few cm's of extrusion missing due to a lack of prime).

Any idea why this happens?

Blue-Marlin, 2017-02-08:
Cold extrusion protection.

billyd60, 2017-02-08:
But the extruder is at temperature... I assume this doesn't matter for some reason? As I said the main gcode prints fine after the prime fails. How do I get the startup code to respond without a reboot? Or is that the only way?

wookie666, 2017-02-08:
Stopping the print from the LCD should cancel everything. And when you start a new print, it should start again from the gcode beginning, copy your entire gcode or atleast the first 100 lines or so

billyd60, 2017-02-09:
Hi I have listed my startup code in the first post which is identical in every print I make. Again you can see it is trying to execute the startup code after a restart because it does everything, homes the extruder, lifts the extruder .4mm pauses (this is where it is supposed to be extruding) then does the wipe move and then proceeds to do the print normally without issue.

In a nutshell, after a print is cancelled from the lcd and restarted, the first extrusion command sent is ignored. Again specifically this line: G1 E25 F300 ; purge nozzle

This happens with every model I print if I cancel it.

From a fresh boot, after the extruder and bed are at temperature, the exact same code works perfectly. It's only if I cancel the print from the LCD screen and restart (bed and extruder stay heated when I stop a print) that it doesn't work correctly. If I power cycle the printer it will print normally.

billyd60, 2017-02-09:
Perhaps the problem lies with the LCD stop print command not turning off heat to the extruder and bed and then the cold extrusion thinks they aren't heated initially until after the first extrusion command refreshes the routine somehow?

Blue-Marlin, 2017-02-09:
Your startup-code should ensure the nozzle is at a sufficient temperature to extrude.

billyd60, 2017-02-09:
Ok that is added by S3D automatically. This is added before my priming sequence automatically by S3D

e.g. for a PLA print:

G90
M82
M106 S0
M140 S60
M190 S60
M104 S215 T0
M109 S215 T0

And then the startup code I listed earlier. Sorry since this is stock from the S3D slicer I didn't think this would be the cause...

Blue-Marlin, 2017-02-09:
Is this sequence really added before before you start your sequence or only if you connect freshly?

billyd60, 2017-02-09:
Yes I only print from SD card so it's reading the same code every time, both the S3D intial sequence and my own priming code.

Blue-Marlin, 2017-02-09:
Please send a 
```
M111 S1
```
to your printer and observe the serial log to ensure the sequence is as you expect it.

billyd60, 2017-02-09:
Ok will try it but remember two things. 

1. Everything works perfectly unless I cancel a print from the lcd screen and restart it. So I don't see how it can be the Gcode
2. This behavior does not happen with the same gcode files on prior versions of Marlin

closed 2017-10-31 by thinkyhead