MarlinFirmware/Marlin #5783: How to add custom code to Marlin?

ronan635, 2017-02-08:
Hello, I'm currently developing my own simple GUI with python to control the basic movements of my 3D printer. I am using an Arduino Mega 2560 with a RAMPs 1.4 board. I am wondering how and where would I add custom code for me to be able to control the stepper motors when a specific button is created within Python. Or could someone point me in the direction of where the functions are for the buttons/commands within marlin that I could use, instead of adding more code, to control the printer with my GUI developed with Python? Any help would be greatly appreciated!

Blue-Marlin, 2017-02-08:
I assume, if you are writing in python, your program is running on a host - not on the Arduino. In that case you have to communicate via the serial line with g-code.

ronan635, 2017-02-08:
Yes I know that I have to use the serial line but I'm having trouble of how to actually send the command to the Arduino for it to process the command.

stigjoergensen, 2017-02-08:
Just googleed this - so it might help you

import timeimport serial

ser = serial.Serial(
    port='\\\\.\\COM4',
    baudrate=115200,
    parity=serial.PARITY_ODD,
    stopbits=serial.STOPBITS_ONE,
    bytesize=serial.EIGHTBITS)if ser.isOpen():
    ser.close()
ser.open()
ser.isOpen()

ser.write("M119\r")
out = ''# let's wait one second before reading output (let's give
device time to answer)
time.sleep(1)while ser.inWaiting() > 0:
    out += ser.read(40)
if out != '':
    print(">>" + out)

ser.close()

Here im seding the M119 command (show status of endstops)

For a nearly complete list of G-Codes see this link:

http://reprap.org/wiki/G-code

Roxy-3D, 2017-02-11:
If you look in Marlin_main.cpp, you will find the 'M'-Code handler.   If you just grab an unused MCode, you can splice in your code.  You would just duplicate the syntax already in place but put your code in the body of the switch() statement:
```cpp
    case 'M': switch (codenum) {
      #if ENABLED(ULTIPANEL) || ENABLED(EMERGENCY_PARSER)
        case 0: // M0: Unconditional stop - Wait for user button press on LCD
        case 1: // M1: Conditional stop - Wait for user button press on LCD
          gcode_M0_M1();
          break;
      #endif // ULTIPANEL

      case 17: // M17: Enable all stepper motors
        gcode_M17();
        break;
```

closed 2017-10-31 by thinkyhead

amrithmmh, 2019-02-15:
@thinkyhead  i tried the above python code with baud 250000  but i get only rubbish/gibberish values printed in output
any suggestion?

revilor, 2019-02-15:
@amrithmmh I guess your python code and Marlin are not set to the same baud rate.

amrithmmh, 2019-02-18:
@revilor thanks for the reply  i just pressed the reset of arduino and then next next i connected with the same code slight modification and it worked
now i am trying to use pronterface(procore class) it is also working but i noticed that sending a G0 command return no OK or done as feedback

amrithmmh, 2019-02-20:
seems that there was an OK after every command but pronterface is not showing it but when i used arduino ide serial monitor it is giving Ok feed back also @thinkyhead  code is not formatted one

`import time
import serial

ser = serial.Serial( port='/dev/ttyUSB0', baudrate=250000)
if ser.isOpen():
  ser.close()
ser.open()
ser.isOpen()
ser.write(("M114\r").encode())
out = ''# let's wait one second before reading output (let's give device time to answer)
time.sleep(1)
while ser.inWaiting() > 0:
  out += str(ser.read(40))
  if out != '':
      print(">>" + out)
ser.close()
`