MarlinFirmware/Marlin #5786: 'DIOA11_DDR' was not declared in this scope

spilz87, 2017-02-08:
hello,

when I try to compile the last RC8 version RCBugFix, I got : 

> 'DIOA11_DDR' was not declared in this scope

does someone less get this too ?

spilz87, 2017-02-08:
the full message is : 

In file included from sketch/MarlinConfig.h:26:0,
                 from sketch/Marlin.h:36,
                 from sketch/temperature.cpp:27:
sketch/temperature.cpp: In member function 'void Temperature::init()':
fastio.h:74: error: 'DIOA11_DDR' was not declared in this scope
 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)
                             ^
sketch/fastio.h:99:25: note: in expansion of macro '_SET_OUTPUT'
 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)
                         ^
sketch/temperature.cpp:1052:7: note: in expansion of macro 'SET_OUTPUT'
       SET_OUTPUT(E0_AUTO_FAN_PIN);
       ^
In file included from sketch/MarlinConfig.h:26:0,
                 from sketch/Marlin.h:36,
                 from sketch/temperature.cpp:27:
fastio.h:74: error: 'DIOA11_PIN' was not declared in this scope
 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)
                                                         ^
sketch/fastio.h:38:28: note: in definition of macro 'MASK'
   #define MASK(PIN)  (1 << PIN)
                            ^
sketch/fastio.h:99:25: note: in expansion of macro '_SET_OUTPUT'
 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)
                         ^
sketch/temperature.cpp:1052:7: note: in expansion of macro 'SET_OUTPUT'
       SET_OUTPUT(E0_AUTO_FAN_PIN);
       ^
exit status 1
'DIOA11_DDR' was not declared in this scope

Bob-the-Kuhn, 2017-02-09:
It's not happy with **E0_AUTO_FAN_PIN**.  

Please attach your **configuration.h**, **configuration_adv.h** & **pins_YOUR_BOARD.h** files & I'll see if I can re-produce it.

Have you tried the latest version of RCBugFix?  A lot of the time this will clear up the problem.

spilz87, 2017-02-09:
I download the last RCBugFix 30min before sending the issue :(

my files are : 

[Archive Spilz.zip](https://github.com/MarlinFirmware/Marlin/files/762759/Archive.Spilz.zip)

thanks for your help

regards

spilz87, 2017-02-09:
it looks like the issue comes from affecting A11 to E0_AUTO_FAN_PIN
it works well on RC7

Bob-the-Kuhn, 2017-02-09:
Agreed - assigning pin A11 to E0_AUTO_FAN_PIN is wrong.  A11 is not a valid pin number.

Where did you get the A11 value?

Pins are numbered 0 - 69.  Some documentation shows D0-D69 but what you put in the define statement is just the number.

Analog pins have two names.  54-69 can be either analog or digital.  If  you need the analog function then you specify 0-15 and use a name that the system recognizes as an analog function (like THERM_0).

I expect that the correct pin number for your fan is 11 or 65 or maybe 9.

spilz87, 2017-02-09:
That was the issue
Solved by changing by 65

Thanks for your help

closed 2017-02-09 by spilz87

Bob-the-Kuhn, 2017-02-09:
It compiled but is that the correct pin?

If you any documentation like a schematic or a pin diagram I'd like to see it.

spilz87, 2017-02-09:
It compiled and it works 
I don't have specific schematic, juste it's close to Ramps 1.3 and there are names on additional pins/connectors

Bob-the-Kuhn, 2017-02-09:
Good news!

I was expecting that the heater fan would be assigned to pin 9.  To do that you need to have the following in your configuration_adv.h file.
```
#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
```

Please send me what docs you have or provide the link.  I've accumulated the schematics & pin lists for almost all the boards supported by Marlin & I'd like to add this board to it..

spilz87, 2017-02-18:
[MKX.zip](https://github.com/MarlinFirmware/Marlin/files/784983/MKX.zip)
It's all I have ...