MarlinFirmware/Marlin #5788: Need help on Z_DUAL_ENDSTOPS with positive offset on the probe

token47, 2017-02-09:
This is similar to #5733 but different.

I want a similar setup but with a twist: my probe is the nozzle itself (with a micro-switch hidden inside the carriage that gets pressed when the nozzle touches the bed -- therefore a fixed probe, this is important). With my setup the nozzle goes up a little bit until it clicks, so the "probe" is in reality _above_ the nozzle (positive values on the probe z offset). This feels strange at first but it works.

Now, the problem: I can't figure a way to at the same time align the bed using dual endstops for Z and also perform the BAL using the nozzle. If I try to put the switches on the MIN side the nozzle will stay in the way all the time... If I put the switches above, the Z axis will crash on them. I don't mind separating the Z_MIN from the probe, but still I can't figure how to arrange the three of them. Do you see a way to use all three on the MIN side?

Can I use two MAX Z endstops for homing and one MIN probe for BAL? Do the BAL work the same if the bed was homed to MAX? (Take into account that homing Z to max must bring a whole lot of other problems if you cannot use BAL to fix it later).

Another idea is to use optic sensors, of the kind that you can go through without bumping on it and put them in the middle of the way, and use it only to align the two sides and then finish the home itself with the nozzle (and I'm talking of only G28 alone here). It would be like: 1) touch probe, 2) go back to the sensors and align, 3) touch the probe again to really home. But I'm not sure if this is an allowed scenario.

Thanks!

token47, 2017-02-09:
Another doubt is: even if the probe was a deploying one, it seems that if I separate the endstop Z_MIN from the probe, the endstop will be used for homing and not the probe. Then, I could use the Z_MAX & Z2_MAX for aligning the bed and the probe for homing, but that does not seem to be the case (still talking only about G28)..

Bob-the-Kuhn, 2017-02-10:
I'm running a DUAL_Z setup.  I have it setup so that it homes to Z_MAX.  Here's my config files.
[Configuration_dual_Z.zip](https://github.com/MarlinFirmware/Marlin/files/765367/Configuration_dual_Z.zip)

The trick is to be sure to set your Z_MAX after homing to a value 10-20 mm less than the actual z max.  That way the G29 will move quickly to just above the false Z 0 and then move slower until the Z_PROBE triggers.

After doing a G28 then a G29 then the offset determined by G29 is added to both the Z_MAX and the Z_0 settings.  A G0 Z0 puts my nozzle right on the glass.

One caveat on the config files.  I have some custom code that uses Z_DUAL_ENDSTOPS_ADJUSTMENT.  You'll have to use M666 to set every time you connect.  See PR #5762 for details.

There may be other custom code items in there.  After a while I tend to forget what I customized.

token47, 2017-02-10:
Thank you @Bob-the-Kuhn!

I suspected about that solution. Please confirm my understanding that if homing will leave the nozzle a little bit above the bed, that would make a requirement to run a G29 every time after a G28, right? I mean, today I can run only a G28 and print, and G29 is an option to improve things a little but not a requirement.

Not that I'm against it, as in practice I always do both on every print, it's just to confirm. Also, I imagine that the correction matrix will look a bit funny with big negative numbers on all positions.

Thank you for posting your configuration, I will play with it as soon as possible.

Bob-the-Kuhn, 2017-02-10:
Big negative numbers are OK.  Mine are in the -27 range.

I have homing in the +Z direction.  That'll leave the carriage at the top after doing a G28.

If you home using the Z probe then you won't get the benefits of using Z_DUAL_ENDSTOPS.

Right now Z_DUAL_ENDSTOPS could use some improvements.  I've already mentioned Z_DUAL_ENDSTOPS_ADJUSTMENT.  The other thing is that G28 turns off auto bed leveling.  You'll need to use the M420 S1 V command to turn it on or run G29.  PR #5781 takes care of that annoyance.

token47, 2017-02-10:
Ok, thank you again. I will try it.

My feeling is that homing and bed aligning should be two completely separate things. This is how I think it should work:

1) Bed aligns left/right based on two switches positioned anywhere (mostly on the MAX side if they are microswitches, but I can see scenarios with they positioned near the MIN side if they are optic go-though type).
2) Do a normal homing on the MIN side using the probe. This step is the same as always and will not have any relation with the above.
3) Do a bed leveling using the probe, also, this also is the same as always.

The problem I see is that today 1+2 are mixed, if you position your endstops at MAX to align the bed you end up having to home there too. This should be independent.

OBS: I feel that the optic sensors are a very good alternative because they could be positioned near the bed, and not take a long time to go to the MAX and back while homing. But, as they would be in the middle of the way, initially there would be no way to know where the current Z is relative to them, so maybe a touch on the bed with the probe could be done first (just to make sure you are below them), then align (step nr. 1 above).

Is this plausible? How does it sound?

Bob-the-Kuhn, 2017-02-10:
Yes, it is definitely annoying to have G28 leave the head at Zmax.

Right now the endstops are used to determine the head's position & to keep the head from going outside the machine's bounds.  Sounds like you want to separate those functions.  That would require a lot of code changes or running without endstops enabled.

You are correct about putting the position sensor in the middle.  You'd have to determine which way to home.  In your solution you use the Z probe to get you to a known position.  That means the sensor in the middle would never be used.

I'm using A1125 hall effect sensors on my system.  Here's how I mounted [them.](http://www.openbuilds.com/threads/hall-effect-limit-switches-mounting.7832/#post-44580)

token47, 2017-02-10:
They would get used later. First you touch bed with probe to know the min position, then you go up a little to the optic/hall sensor to align and then you touch the bed again to really home (it seems that you are already homing on the first step but the alignment will probably invalidate it so do it again).

About the code changes, it did not seem too much at first sight, The two portions are already implemented, you would just have to only align (and not consider it "homing" yet) with the align code already done, and then simply home (making sure you use the probe and not the endstops).

strips, 2017-02-10:
Following this one as I am trying to get two Z max endstops and a probe to work. I actually have two optical slit switches for Z max. I chose them because then I can easily use them for Z min as well.

thinkyhead, 2017-10-31:
> Following this one as I am trying to get two Z max endstops and a probe to work.

This has finally been implemented, and will be included in Marlin 1.1.7, coming soon.

closed 2017-10-31 by thinkyhead