MarlinFirmware/Marlin #5789: Auto-leveling - repeated probing

Gad, 2017-02-09:
Hello,
I just move to the latest RC8 bug fix to take advantage of the bilinear algorithm (I was in RC4 where apparently it wasn't yet available and my bed is very ), did my configuration again and everything is working as expected apart from one detail : there is no second probing on each point as it used to be the case. 
I did uncomment "#define Z_MIN_PROBE_REPEATABILITY_TEST" as I thought it was the option to use.
I'm using Auto-leveling with a BLtouch, 4x4 grid,  bilinear "mode" on Prusa I3/RAMPS
Did I miss something ?
Thanks !

FHeilmann, 2017-02-09:
what's your 

#define Z_CLEARANCE_BETWEEN_PROBES ? if it is too low the "fast touch" is omitted and only the "slow touch" is carried out.

Roxy-3D, 2017-02-09:
And just to be clear:   If you have a BL-Touch, you can't go super slow or you will lose position and often times cause the probe to go into an error state.       There are two fixes.  I am using both fixes on my printer with a BL-Touch probe and have not had a nozzle crash since I did that.

First, if you can turn on 
```cpp
// Enable this feature if all enabled endstop pins are interrupt-capable.
// This will remove the need to poll the interrupt pins, saving many CPU cycles.
#define ENDSTOP_INTERRUPTS_FEATURE
``
in your Configuration.h file.   That way you should not miss the 5ms pulse the BL-Touch puts out.   But then add this Pull Request to your firmware: https://github.com/MarlinFirmware/Marlin/pull/5729

It checks the end stops more often (which shouldn't be active with the above Configuration.h setting).  But it also is looking for the BL-Touch to be in an error state and takes steps to clear it.

Bob-the-Kuhn, 2017-02-10:
Is **#define PROBE_DOUBLE_TOUCH** commented out?

Roxy-3D, 2017-02-10:
@Blue-Marlin   Why the down vote?   What is incorrect in what I said?

Blue-Marlin, 2017-02-10:
@Roxy-3D 
Not incorrect, but no relation to the question.

Roxy-3D, 2017-02-10:
I disagree...  People with BL-Touch probes are vulnerable to nozzle crashes at low speeds.   If he is looking to have Double Touch probing that implies he is trying to make the probes happen at a slow speed.   

I was trying to help him protect his printer from damage.

Blue-Marlin, 2017-02-10:
Q: What time is it?
A: February

Not wrong - but not the answer.

Q: there is no second probing on each point as it used to be the case. ... Did I miss something ?
A: #define PROBE_DOUBLE_TOUCH 

100%

Gad, 2017-02-10:
Hi, thanks to all for the suggestions.

- Yes indeed  #define PROBE_DOUBLE_TOUCH in commented out. I'll leave it this way. Moreover, I've set the Z axis acceleration very low already
- as a side note, I'm pretty sure it does the double probing on G28 (I don't have access to the printer until next week)
- I experience BL Touch alarm every once in a while, quite seldom to tell the truth so I'd rather wait for the new RC to be posted.
Thanks again!

closed 2017-02-10 by Gad

Bob-the-Kuhn, 2017-02-10:
> I experience BL Touch alarm every once in a while

When triggered, BLTouch retracts the probe, waits and then drops it again.  If it triggers on the second drop then it goes into alarm.

Either you're running at extremely slow speeds or you have your Z_CLEARANCE_DEPLOY_PROBE or Z_CLEARANCE_BETWEEN_PROBES set too low.