MarlinFirmware/Marlin #5791: Heated Bed heating up then cooling down

wmhoneycutt, 2017-02-10:
I recently received a Monoprice Maker Select V2 and changed the Melzi board over to ramps 1.4. I've installed the latest version of Marlin, and although I'm relatively new to the 3D printing world, I've worked through most of my bugs except for one. 

Whenever I try to Preheat PLA Bed, the temperature goes up to 49 deg Celsius and then starts to go back down. The thermistor is reading room temp just fine. I'm sure I'm doing something wrong, so I apologize in advance for my ignorance.

Bob-the-Kuhn, 2017-02-10:
By chance could BED_MAXTEMP be set to 49C?

wmhoneycutt, 2017-02-10:
On any given day, very valid question. However, No sir.

// When temperature exceeds max temp, your heater will be switched off.
// This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
// You should use MINTEMP for thermistor short/failure protection.
#define HEATER_0_MAXTEMP 260
#define HEATER_1_MAXTEMP 260
#define HEATER_2_MAXTEMP 260
#define HEATER_3_MAXTEMP 260
#define BED_MAXTEMP 120

3dGriff, 2017-02-11:
What happens if you heat the bed via the control menu?  Also, does the set point stay at 49c or does it go to zero?

wmhoneycutt, 2017-02-11:
Mr. Griff... Thank you!!
 The MOFSET was getting really hot. Then I remembered something I read in a thread somewhere else, about how a kid put a fan on his board a. 't want to enclose everything until I had it working. But once I held the fan over the circuit board, the heated bed finally heated up. With the Control Procedure AND the Pre-Heat procedure. I tested them both to verify that it was indeed a fix. 
[image]

3dGriff, 2017-02-11:
@wmhoneycutt Great!  Happy I could help, do keep an eye on it and make sure you keep a fan on it.  I've had mixed results with the RAMPS and various issues related to heat.  About a year ago I did a pretty thorough dive into the points of failure on the board, in the end I could only resolve that sometimes they fail and generally you get what you pay for. 

One thing to keep in mind with the MOSFET situation is this:  MOSFET's short on, meaning if they go bad and the board is powered the bed will continuously heat.  I see that you have a power switch so just be cognizant of the possibility of failure on that MOSFET and try to keep it powered off when not in use.  

Feel free to contact me directly if you run into any similar issues with the retrofit.  

[image]

Bob-the-Kuhn, 2017-02-11:
I'm wondering if your hotbed is drawing more current than your RAMPS board is designed to handle.  Were any of the hot bed connections or wires also hot?

Any chance that you're running a 12V heating plate at 24V?

You may also want to buy a heatsink for the FET.

closed 2017-02-12 by thinkyhead