MarlinFirmware/Marlin #5795: Is this  a bug about stop printing?

Jolly2015, 2017-02-10:
Printing a big square model on delta or prusa i3, when click "Stop print" on LCD, sometimes(with high probability) the machine won't be stop immediately and move to a unknown position and then stop.
Only appeared on Square model(I think it's bearing on commond buffer).

Tannoo, 2017-02-10:
Repetier has the option to move to X, Y after job killed or completed.

Check you host software for a simular feature.

Jolly2015, 2017-02-10:
@Tannoo   It's print from SD card(slicing software is cura),Test on marlin RC8,and the older version all had the same issue.

Tannoo, 2017-02-10:
Try RCBugFix?

Tannoo, 2017-02-10:
If you are hitting to kill button to stop, it should stop right then.
If you are selecting "Stop Print", then it will stop reading new commands and finish the buffer out.

Jolly2015, 2017-02-10:
@Tannoo  I havn't try RCBugFix,but I saw the code "lcd_sdcard_stop()" is same.
Yeah,click kill should stop right then, but program will dead in while(1) and need to restart.
when click "stop print", can I  stop  current move immediately ? cause if "stop print" was selected,means there was some issue happen,so needn't wait to finish the buffer out.

Tannoo, 2017-02-10:
Your LCD should have a separate button. Click that one and the printer will stop everything.

Tannoo, 2017-02-10:
Selecting "stop print" from the lcd will stop after the buffer is emptied.

Tannoo, 2017-02-10:
That separate button is treated as an Emergency stop. It will stop and kill everything. Steppers, heaters, everything. The controller will have to be reset after that.

Tannoo, 2017-02-10:
That "E-Stop" is good for issues like a bad endstop / endstop mount, loose belt, magic smoke, etc, etc..

Jolly2015, 2017-02-10:
@Tannoo Yeah, I know the button and it's function, but the problem is: If I stop from LCD(like the situation is not very emergency )when print a square,then the nozzle running to a unknown pos(not stop when endstop was triggered ), what about this?

Tannoo, 2017-02-10:
Yeah, I'm not sure about that yet. I had that happen with filament change back on RC7 when it would return the print position, it just seemed to drift at random for a few seconds while extruding. Made mess of the parts.
I don't know what fixed it. But RC8-RCBugFix doesn't have that issue.

So, try RCBugFix and see if the issue is still there.
Most of the help with coding issues here will happen on RCBugFix.

Jolly2015, 2017-02-10:
@Tannoo, Thanks, I will try it and I think I can solve it.

billyd60, 2017-02-11:
In my opinion there are all kinds of problems with the latest rcbugfix when you stop a print from the LCD panel. In fact I think it ultimately causes the firmware to crash if lin_advance is enabled and you stop and restart a print without a cold boot. Also it seems to cause problems with the thermistor on both the extruder and heat bed in that you get weird temperature drops that occur randomly. I think all of these issues are related to stopping a print prematurely with the LCD panel and then restarting. If I start up the printer and print without stopping from the panel I have no issues, perfect prints every time. I have also seen the cold extrusion prevention incorrectly prevent the first extrusion in a print job after stopping a print from the lcd screen and then restarting.

Also yes when you stop a print from the LCD panel the printer will sometimes continue to print or get stuck in a command and will drive the printer right into the endstops.

So avoid stopping a print from the lcd screen with RCBUGFIX imo.

closed 2017-02-11 by Jolly2015