MarlinFirmware/Marlin #5797: Remember Last Toolchange Used?

sarf2k4, 2017-02-10:


Bob-the-Kuhn, 2017-02-10:
What you propose is doable but ...   Automatically writing to the EEPROM is discouraged because you can literally wear it out.

I think the better approach is to put the tool change command in your slicer's start script.

sarf2k4, 2017-02-10:
simplify3d does that everytime, but on power loss/reset, it will go back to extruder 1, but the one loaded is actually extruder 2, unless putting several micro switches to detect the current extruder that is loaded on the teflon tube of each extruder, but having the micro switch on them will make it even more complex.

Unless there are some other methods to retain the last used tool/extruder

Bob-the-Kuhn, 2017-02-10:
I didn't answer one of your original questions.  The only thing that changes the active extruder is the tool change command or going through reset.   Reset always selects extruder 0.  Reset occurs on power up and whenever you disconnect/connect to the printer.  There is no timeout on which extruder is selected.

Since "simplify3d does that everytime" then the correct extruder will be selected before the print starts.

I guess I'm not understanding why you want to save the last used extruder.  How would that change/improve your print setup process?

sarf2k4, 2017-02-10:
Sorry, I don't want to raise unhealthy argument, I'm trying my best to clarify what I want to address.

If for example the loaded extruder is extruder 2, and we don't know the current selected extruder, doing the toolchange script to swap the extruder filament would caught someone off guard if the owner of the printer doesn't even know if the printer had a reset or power failure before they started to use the printer. 

Another would be that, the user had power outage during printing/crash/thermal runaway and had to reset while the one loaded is extruder 2, got to manually retract the loaded filament same way as the toolchange script would do.

Bob-the-Kuhn, 2017-02-10:
I'm running a single nozzle so I have not experienced what you're trying to improve.

I'm going to just leave it as "I don't understand".

token47, 2017-02-10:
The problem is that the filament from extruder 2 is going through the Y splitter. If the printer resets and on the next print simply loads filamento from extruder 1 they will crash inside the splitter. It should remember that extruder 2 was the last one used so when trying to use 1 next, it would unload the filamento from 2 and load from 1. Unless he puts a microswitch on each filament to see which one is being used, there is no way to know which one is loaded unless it remembers from last use.

Bob-the-Kuhn, 2017-02-10:
I take it that you can't see which filament is loaded within the splitter.

How about automatically unloading both on power up / reset?  If you change **card.checkautostart(false);**  to **card.checkautostart(true);** in Marlin_main.cpp then the g-code file auto0.g will automatically be executed on reset and power up.

thinkyhead, 2017-02-12:
I understand your proposal makes some sense for your setup, and maybe some commercial printer will want to use this kind of scheme. Of course just because the last tool was T1 doesn't necessarily mean the filament has also been loaded. It would be better if there was a sensor that could detect this.

sarf2k4, 2017-02-12:
What if, by using a microswitch that is driven by a servo arm? I dont know
how marlin handles the servo everytime it resets, does it do like z probe
movement?

Lets say i have a z probe arm at position 90 for retract and position 12
for deploy, on my end it goes to somewhere in around position 40 then goes
to retract position afterwards

thinkyhead, 2017-04-02:
An optical switch in the filament path would be my recommendation.

closed 2017-04-02 by thinkyhead