MarlinFirmware/Marlin #5799: Second Z Min Endstop

FloydPerfect, 2017-02-11:
I was trying to get a second Z_MIN endstop running. But it is not working the way I would like to have it.

I have a quickout system for the head. 3D print head actually has a BLTouch mounted. Another head is just a camera head that has not level sensor attached. I have mounted a second sensor on the z axis. If a head is mounted with a bed level sensor it should have priority. Means it comes first due to its mechanical position. The second sensor comes when no level sensor is installed about 5mm later. In this case I can do homing without level sensor. Practically this is ORing of two Z_MIN_PINs. 

I have tried Z_MIN_PROBE_ENDSTOP with the settings written in configuration.h but this takes just the sensor input.

Is there any way to configure an OR of two inputs for Z_MIN ?

Tannoo, 2017-02-11:
Is it possible to just wire the two enstops in parallel and use just one endstop input?

FloydPerfect, 2017-02-11:
yes it is possible under the following conditions:
- mechanical switches
- electronic switches with open collector  output and both npn or pnp 

I am not shure but I think BLTouch is using a push pull output.

Bob-the-Kuhn, 2017-02-11:
Yes, BLTouch has a push/pull output.

Below are the config settings I think will allow you to do exactly what you want.  This is from a previous issue with the same concern.

Let me know either way if his works for you.  I have a PR open that makes this recommendation.

=====================================================

The settings needed to use the Z_MIN_PROBE_PIN are:
1. select the type of probe you're using
2. define Z_MIN_PROBE_PIN in your pins_YOUR_BOARD.h file
3. disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
4. enable Z_MIN_PROBE_ENDSTOP

NOTE – if Z_MIN_PIN is defined then it’ll be checked during all moves in the negative Z direction.

FloydPerfect, 2017-02-11:
Thanks for jumping in. I have tried this before. 
I have a MKS GEN 1.4 board. This uses pins_RAMPS.h. As written in the configuration.h I tried to use the
X_MAX_Pin for the BLTouchAnd I just tried again. I am using the boards Z_MIN_PIN for the endstop.
This works. I  #define Z_MIN_PROBE_PIN 2 which should result to the X_MAX_PIN. The BLTouch is then plugged on this pin. But it does not deploy its pin. So this is not working by now.
The servo connector of the BLTouch is still on the same place.

This is the config:

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false //  set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.

#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.

and 

// The Z_MIN_PROBE_PIN sets the Arduino pin to use. (See your board's pins file.)
// Since the RAMPS Aux4->D32 pin maps directly to the Arduino D32 pin, D32 is the
// default pin for all RAMPS-based boards. Most boards use the X_MAX_PIN by default.
// To use a different pin you can override it here.
//
// WARNING:
// Setting the wrong pin may have unexpected and potentially disastrous consequences.
// Use with caution and do your homework.
//

#define Z_MIN_PROBE_PIN	2                                                      // v.t. Use this pin X_MAX_PIN
#define ENDSTOPPULLUP_XMAX							// v.t. pullups for this pin

//
// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//

#define Z_MIN_PROBE_ENDSTOP                                  // v.t Eingeschaltet für Endschalter und Probe

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.

//  #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN                                               // v.t Ausgeschaltet 

I would expect that the probe pin is deployed and the first that gets hit triggers z home.

Any idea why the pin of the BLTouch is not deployed ?

Bob-the-Kuhn, 2017-02-11:
Can you manually deploy the BLTouch via the M280 P0 S10 command?

By chance do you have any documentation on your board?  I have the docs for almost all the Marlin supported controllers.  I'd love to add this one to it.

Please ZIP & attach the following files.  I'll see if I can reproduce the problem.  
- Configuration.h
- Configuration_adv.h
- pins_YOUR_BOARD.h (only if you've modified it.

Also, which version of Marlin are you running?

====================================================

The "quickout system" interests me.  Please point me to where I can get more info.

FloydPerfect, 2017-02-11:
the m280 is working and deploys the pin of the BLTouch.

Unfortunatly I also have no information on the board. I have done a lot of searches on the web. As far as I have found by now it is just a combination of Ramps 1.4 and the Arduino Mega board.

I have done the quickout for a Tevo Black Widow. You will find a picture of the mounted printerhead and
the camera head aside. There is a IDC 37 connector on top that gives enough pins for using a dual head. There is also an I2C interface on the connector because I am thinking of retreaving some informations for Marlin at startup or change. But this is a future melody.
If you are interested I can send you the files. 

Attached is also a zip file with the config files.

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/768800/conf.zip)
[image]

Bob-the-Kuhn, 2017-02-11:
What version of Marlin are you running?   It's definitely not the latest.

Bob-the-Kuhn, 2017-02-11:
Also, try the following
M120
M119
M280 P0 S10
M280 P0 S60
push the probe up with your finger
M119

The probe state reported by M119 should have changed from OPEN to TRIGGERED.

FloydPerfect, 2017-02-11:
The version of Marlin is RC8

>>> m120
SENDING:M120
>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: open
>>> m280 p0 s10
SENDING:M280 P0 S10
>>> m280 p0 s60
SENDING:M280 P0 S60
>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: TRIGGERED

that all looks ok. But maybe  I have a wrong understanding. When I issue a G28, I would expect that the pin is deployed and that it also works as endstop.

Bob-the-Kuhn, 2017-02-11:
Try disabling  **Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN**.    Right now you have it enabled.  That may be confusing the software.

FloydPerfect, 2017-02-11:
Sorry,
but this one is disabled. VT_USE_Z_PROBE_OR_ENDSTOP is defined and  Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is in the #else

[image]

Bob-the-Kuhn, 2017-02-11:
Also, turn off EEPROM.  If that's on it ignores a lot of the stuff in the config files and gets it out of EEPROM.

Can cause lots of confusion & frustration when trouble shooting a problem.

Bob-the-Kuhn, 2017-02-11:
You are correct on the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.

I'm having a devil of a time trying to get my config back to where it uses the BLTouch for Z homing.  I think I'll take a break & see if things clear up.

FloydPerfect, 2017-02-11:
Well I have tried a M502 without success and I did remove EEProm Support also behaving the same.

The pin is not deloyed.
Anyhow I am a little concerned about this comment from the configuration.h

//
// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//
 
That is how it behaves and it is not the way I am looking for.

Also will have a break ( it is about 22:30 )

Bob-the-Kuhn, 2017-02-11:
I think I have the solution.

In the file **Conditionals_post.h** replace 
` #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))`

with

` #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0)`

I never was able to modify your config to run on my machine so I modified my config.

On my machine it resulted in the BLTouch deploying as soon as it started the -Z travel on G28 and deploying on G29 after getting down to the Z_HOMING_HEIGHT (or is it Z_CLEARANCE_DEPLOY_PROBE).

I don't have a Z_MIN sensor on my system so I can't verify that works as you want.

======================================================

I saw that your Z_PROBE_OFFSET_FROM_EXTRUDER was set to zero.  That's unusual.  Usually it's a small negative number (mine is -2.7).  At zero that means the BLTouch is triggering exactly when the nozzle touches the bed.

FloydPerfect, 2017-02-12:
You made it. Great job.
The Z_PROBE_OFFSET_FROM_EXTRUDER was set by the M851 command and stored in EEProm.
I have changed this. 

For anyone else who wants to do the same, here are the steps:
- in configuration.h:
  -   #define Z_MIN_PROBE_PIN	2           // set the pin to use for the probe (X_MAX_PIN)
  -   #define ENDSTOPPULLUP_XMAX	     // pullups for this pin
  -   #define Z_MIN_PROBE_ENDSTOP 
  -   // #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN      comment this out
- in Conditionals_post.h
  replace
      #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 &&
                                       ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
  with
      #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0)

thinkyhead, 2017-04-02:
> replace … with …

I haven't found a 100% reliable way to know whether homing is to be done with the probe, but possibly `!HAS_Z_MIN` can help…?
```cpp
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 \
  && (ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) || !HAS_Z_MIN))
```

closed 2017-04-02 by thinkyhead

Beaving, 2018-04-04:
This caused me so many hours of headache and I thought I have got yet another faulty BLTouch...

The Trigorilla board on the Anycubic i3 Mega has some additional daughter boards that prevent you from easily connecting probes or a BLTouch to Zmin, so I'm using some other PINs. It would do the self test successfully, I could control it via M280 but it just would not home with the probe like I'm used to.

@thinkyhead Shoudn't homing always be done with the probe? We should at least add a better reachable define somewhere, maybe in the probe/BLTouch settings, in case you don't come up with another solution.

For my specific usecase, I want to use the existing dual Z endstops for synchronizing both Z motors (the height of the axis) and then the BLTouch for probing/homing/bed leveling. I think in that usecase, !HAS_Z_MIN wouldn't help, not sure though.

Bob-the-Kuhn, 2018-04-05:
On my dual Z I home to the Z_MAX endstops so Marlin will automatically adjust the offset.

I use my BLTouch only for bed leveling.

thinkyhead, 2018-04-10:
> Shouldn't homing always be done with the probe?

If there's a probe, and if the home direction is down, and if the probe is connected to the Z min pin, homing will be done with the probe.

```cpp
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
```

KalessinRoke, 2018-04-19:
@Bob-the-Kuhn this thread I believe is what I am looking for.  I have followed what is instructed here but it does not appear to be working.  Running 1.1.8 and dual Z endstop works fine.  Once the configuration to enable BL Touch is activated, homing will only use the BL Touch.  The functionality that I seek is as you stated above 

> On my dual Z I home to the Z_MAX endstops so Marlin will automatically adjust the offset.

> I use my BLTouch only for bed leveling.

Any thoughts on what could potentially still be off?  I expect that homing will use the dual Z endstops and I can call on BL Touch for bed leveling only.

thinkyhead, 2018-04-19:
- In `Configuration.h`, define `Z_MIN_PROBE_PIN` as the pin you want to use for the probe. It's ok to use the same value as `Z_MIN_PIN`.
- Disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` (which auto-assigns the pin).
- Enable `Z_MIN_PROBE_ENDSTOP` so Marlin knows to use `Z_MIN_PROBE_PIN` for the probe.

thinkyhead, 2018-04-19:
- If homing to MAX, define `Z_HOME_DIR` as 1. That should do the trick.

If not homing to MAX, then…
- In `Configuration.h`, define `Z_MIN_PROBE_PIN` as the pin you want to use for the probe. It's ok to use the same value as `Z_MIN_PIN`.
- Disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` (which auto-assigns the pin).
- Enable `Z_MIN_PROBE_ENDSTOP` so Marlin knows to use `Z_MIN_PROBE_PIN` for the probe.

thinkyhead, 2018-04-19:
Strange. And your `Z_HOME_DIR` is set to positive 1?

KalessinRoke, 2018-04-19:
I just added to configuration.h

#define Z_HOME_DIR 1 

and the same result, BL Touch is initiated for homing

thinkyhead, 2018-04-20:
For now just change this in `Conditionals_post.h`:
```diff
- #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
+ #define HOMING_Z_WITH_PROBE false
```
Later we'll add an option to override this usual behavior.

KalessinRoke, 2018-04-20:
@thinkyhead I have to thank you for the responses.  I believe we have found the answer.
I updated the entry in `Conditionals_post.h`:

`#define HOMING_Z_WITH_PROBE false`

= **SUCCESS**

This got me to "thinky" though.  The logic defined in the original statement parameter is exactly correct for the functionality I desire:

`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
`

The problem was really understanding the 3 conditions to set = true.  

In my setup I have defined that Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN as DISABLED.  This should have made the #define statement be false as && ENABLED is DISABLED.

However, looking at what Bob-the-Kuhn was stating he must have been forcing the enabling of the HOMING_Z_WITH_PROBE by removing the && ENABLED statement.  

With my case, both of the conditions stated HAS_BED_PROBE and Z_HOME_DIR < 0 were true therefore HOMING_Z_WITH_PROBE was active!

So, it seems there is not a bug here but an understanding of the intended functionality and the statements to apply.

Dual Z homing and Probe for Bed Leveling only.  This is only due to the fact that the dual Z axis over time "slip" so need to enforce re-zeroing the on the two endstops.

Thank you again for hammering this out with me.

thinkyhead, 2018-04-21:
Good news! Happy printing!!