MarlinFirmware/Marlin #5801: endless Wnarrowing while compiling but no error

TylonHH, 2017-02-11:
When I compiling I get tousand warnings, like:
``sketch\thermistortables.h:677:1: warning: narrowing conversion of '1.50323203e+4' from 'double' to 'const short int' inside { } [-Wnarrowing]``

I can still upload and use the firmware, so no functional error so far.
But how can I get rid of it?

Bob-the-Kuhn, 2017-02-11:
The stock "take two aspirin & call me in the morning" suggestion is to try the latest RCBugFix.  You'd be surprised how often this fixes things.

I don't know if you're seeing a real issue or if it's a compiler flag issue.  Since you're functional I expect it's a compiler issue.

The compiler is telling you that it's converting a double number into a short.  The compiler does the number format changes all the time but seeing a warning is unusual.

Are you using the Arduino IDE?  If yes, which version?  1.8.0 had some issues.  1.8.1, 1.6.12 & 1.6.13 seem to be working well.  Try a different version of the Arduino IDE & see if that eliminates the warnings..

Are you downloading an extension to the Arduino IDE?

TylonHH, 2017-02-11:
I use the latest arduiono 1.8.1 and
``#include <U8glib.h>``

Maybe I'll try the RCBugFix, if I know how to save my changes....

Bob-the-Kuhn, 2017-02-12:
If you have a config that works in one of the RC releases then chances are that you'll be able to drop them into RCBugFix without a lot of work.  Some of the names have changed but the compiler will let you know what to fix.  

Options/features have been added to RCBugFix.  You'll need to move your machine specific items into the config files to take advantage of them.

I use notepad++ as my editor.  It has a really nice plugin that will do a side by side display of the differences in a file.  Every month or so I'll do that.

TylonHH, 2017-02-12:
That's an good alternative. I use N++ allot.
My first try was to fork this in my own branch so I can merge the updates by myself. But I'm really new into that.

Let me give feedback when I'm done.

Bob-the-Kuhn, 2017-02-12:
Take a look at PREFERENCES under FILE in the Arduino.  I see selections for verbosity and for compiler warnings.  Maybe one of them is what you're looking for.

[image]

closed 2017-10-31 by thinkyhead