MarlinFirmware/Marlin #5803: RC8 mintemp bed and extruder error

rektbink, 2017-02-11:
My printer works fine on RC7 other than the auto level z adjust, which it why I wanted to try RC8. Everything works well that when ever there is a retraction the bed or extruder min-temp error pops up and the printer shut down. New sensors on the bed and extruder, and proper settings in Config.

Bob-the-Kuhn, 2017-02-11:
Try the latest RCBugFix.  It has a lot of bug fixes in it.  

If that doesn't fix it then the next questions that come to mind are:
- Is the temperature actually dropping? 
- Is it dropping as usual at the end of a print or is going from normal to a low value instantaneously?
- Can you reproduce this manually?

If the latest RCBugFix doesn't fix it then please ZIP up the following files & attach.  I'll try to reproduce the issue.
- Configuration.h
- Configuration_adv.h
- pins_YOUR_BOARD.h

======================================

How do you know that these were the errors?  I'm using Repetier host and all I get is a big popup that basically says you're SOL.

rektbink, 2017-02-12:
The latest RCBugFix does not correct the issue. It seems that I am getting temp spikes on the extruder and heatbed. I was able to replicate the issue by manually extruding 10mm at a time until the temp spikes were present, then the printer shut down. This is all with no fan running. The temp drops as one would expect during extrusion, this is PTEG so the base temp is 250c. To test I had extruder 10mm repeatedly with one second spacing at 155mms. Spacing was to prevent a jam. The tempurature is not spiking physically like the plot represents and this is based on my thermal imager. I have encountered two errors while printing, neither present simultaneously, heat bed, and heater 0. So hear bed and extruder 1. During manual extrusion I have the bed turned off so I cannot say whether this error is repeatable. Imediately after this test I relaoded RC7 and ran the same test with 10mm extrusion at 155mms and I had no spikes or temp errors. 

However, I will not rule out any hardware faults for the time being. I have a new block, heater and temp sensor on the way and a new ramps board in the closet I can try if need be. 

[img]http://i.imgur.com/RRf1ULZ.png[/img]

Here are the files you asked for. 
https://drive.google.com/open?id=0B53t_ZvnBllXelZjYWo1ZnpIcDQ

Bob-the-Kuhn, 2017-02-13:
> The tempurature is not spiking physically like the plot represents and this is based on my thermal imager. 

The plot shows spikes but your thermal imager says the temperature isn't changing??!!!?

That implies the problem is in the reporting system.  That's a big step forward.

Has the printer thrown an error & gone into shut down because of this?

Try enabling the log in your host software.  It would be good to know what the host is seeing.

I'm trying to see if there is a way to log the USB traffic to the SD card.  So far I've only found logging the incoming command stream.

===========================================================-

Here's my musings before I noticed/understood the thermal imager thing.

===========

I tried to reproduce your issue but didn't see anything unusual before my extruder broke.

I copied your files into today's RCBugFix & modified them for my machine.  I then manually set the extruder temperature to 250C & the bed to 50C.  After they got to temperature I started issuing G1 E10 commands.  I'm using Repetier Host to interface to the printer.

I got to 680mm before my extruder broke.

How long did you have to issue G1 E10 commands before you saw the spikes?

From the graph provided by Repetier Host I can tell you that your extruder negative spikes were very similar to what I saw when I turned off the heater for a minute.

Maybe your negative spikes are being caused by the power FETs getting too hot.  But ... if it works with RC7 that casts doubt on this theory.  Maybe put a fan on your board to see if that helps

Try issuing a M111 S2 command.  Sometimes it'll report something unusual.

You might also try enabling in Configuration.h the following.  It might show something.
```
#define PID_DEBUG // Sends debug data to the serial port.
#define PID_BED_DEBUG // Sends debug data to the serial port.
```

I'm really scratching my head over the hot bed spikes.  My hot bed has a heat spreader in it so I can't respond that fast.

Positive spikes & spikes to -15C.  Very strange.

The -15C readings happen when the analog input is near 5V.  Since there's a 4.7K pullup resistor on the analog inputs for the extruder & the bed,  an open sensor or a broken cable connection would cause a -15C reading.

Usually an open would result in the temperature reading dropping immediately but your graph shows a 1 minute duration drop from normal to -15C.  

I checked your pin definitions and there's nothing else assigned to TEMP_0_PIN & TEMP_BED_PIN.  I saw one configuration where a digital out was assigned to the same pin as an analog in.  Definitely made the analog in unusable.

=======================================

Maybe you could try reproducing the spikes by turning the heaters off for a minute and then back on.  If the on/off graph looks close to the one you posted you can probably conclude that something physical is happening.

You'd should do this for both the hot bed and the extruder.

If something physical is happening then we should start wondering if there is something interfering with the turning the heaters on & off.

====================================

rektbink, 2017-02-13:
"Usually an open would result in the temperature reading dropping immediately but your graph shows a 1 minute duration drop from normal to -15C."

The scale of the plot is in **seconds**.

The new block and heater will be here today but before I install them I am going to test the thermistor using an external heater. It being a gas heater, I should suspect that the results are similar using RC8. 

I would reach appx 250-300mm before the spikes became present in the plot, but it would take another 100+mm before the system would produce a MINTEMP error and shutdown. 

I will try out your other suggestions too this afternoon.

Bob-the-Kuhn, 2017-02-13:
The plots are in seconds - that eliminates heater control as the problem.  The hot end and the bed just can't change temperature that fast.

There's a proposed change to the temperature software that might fix your issues.  Please copy the attached files into your RCBugFix, compile & see if the problems are gone.

[temperature_fix.zip](https://github.com/MarlinFirmware/Marlin/files/771782/temperature_fix.zip)

rektbink, 2017-02-15:
The temp fix you posted will not compile.

`sketch\stepper.cpp: In static member function 'static void Stepper::isr()':

stepper.cpp:375: error: 'ENDSTOPS_ENABLED' was not declared in this scope

     if (step_remaining && ENDSTOPS_ENABLED) {   // Just check endstops - not yet time for a step

                           ^

stepper.cpp:383: error: '_NEXT_ISR' was not declared in this scope

       _NEXT_ISR(ocr_val);  //

                        ^

stepper.cpp:399: error: '_NEXT_ISR' was not declared in this scope

     _NEXT_ISR(200); // Run at max speed - 10 KHz

                  ^

stepper.cpp:440: error: '_NEXT_ISR' was not declared in this scope

       _NEXT_ISR(2000); // Run at slow speed - 1 KHz

                     ^

stepper.cpp:455: error: 'ENDSTOPS_ENABLED' was not declared in this scope

     if (ENDSTOPS_ENABLED) endstops.update();

         ^

stepper.cpp:368: error: 'ENDSTOPS_ENABLED' was not declared in this scope

       if (ENDSTOPS_ENABLED && L > ENDSTOP_NOMINAL_OCR_VAL) { \

           ^

sketch\stepper.cpp:636:5: note: in expansion of macro 'SPLIT'

     SPLIT(timer);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL

     ^

stepper.cpp:637: error: '_NEXT_ISR' was not declared in this scope

     _NEXT_ISR(ocr_val);

                      ^

stepper.cpp:368: error: 'ENDSTOPS_ENABLED' was not declared in this scope

       if (ENDSTOPS_ENABLED && L > ENDSTOP_NOMINAL_OCR_VAL) { \

           ^

sketch\stepper.cpp:692:5: note: in expansion of macro 'SPLIT'

     SPLIT(timer);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL

     ^

stepper.cpp:693: error: '_NEXT_ISR' was not declared in this scope

     _NEXT_ISR(ocr_val);

                      ^

stepper.cpp:368: error: 'ENDSTOPS_ENABLED' was not declared in this scope

       if (ENDSTOPS_ENABLED && L > ENDSTOP_NOMINAL_OCR_VAL) { \

           ^

sketch\stepper.cpp:743:5: note: in expansion of macro 'SPLIT'

     SPLIT(OCR1A_nominal);  // split step into multiple ISRs if larger than  ENDSTOP_NOMINAL_OCR_VAL

     ^

stepper.cpp:744: error: '_NEXT_ISR' was not declared in this scope

     _NEXT_ISR(ocr_val);

                      ^

exit status 1
'ENDSTOPS_ENABLED' was not declared in this scope`

Bob-the-Kuhn, 2017-02-15:
Try using RCBugFix again.  They've just released fixed temperature code (the code you tried to test).

closed 2017-10-31 by thinkyhead