MarlinFirmware/Marlin #5804: Leveling work from G29 but not adjusting Z while printing

pcsentinel, 2017-02-11:
I have set up a probe that use the Z Minstop connector

Z Homing working fine, Z Safe homing working fine, G29 bed levelling working fine, BUT when you actually try and print then there is no modification of the Z position during the print. So I have a bed that is high in the front right. run G28 XY, G28 Z, G29 then print, but the print is fat in the back left and the head is squashed against the plate front right

Gcode and configuration.h in attached file. 
Board is K8200

[failed levelling.zip](https://github.com/MarlinFirmware/Marlin/files/768723/failed.levelling.zip)

pcsentinel, 2017-02-13:
Pain Over, loos like I discovered the issue, I moved G29 further down the start G code and its now working. the only remaining issue is that I have a weird setting for the z Probe offset, when measured its 2.2 mm but I have to use M851 Z-1.2 when I want it to print at the right height.

pcsentinel, 2017-02-13:
Not entirely Pain Over, It is now adjusting as I can see it turn the z motors as it moves across the bed on each layer, but it doesn't seem to be adjusting enough. Back Left to thick, Fornt right to squished.

Iakouben, 2017-02-13:
I'm having a similar issue at the moment as well. Using an inductive sensor as a probe against a cast aluminium bed and RCbugfix from 27/1/17.

I have found however that the inductive probe gives me a fair amount of error in the reading. I can get a variance of about 0.15mm in the z value when probing the same point.
Repeatability test result:

    READ: Mean: 0.305855 Min: 0.201 Max: 0.355 Range: 0.154
    READ: Standard Deviation: 0.053675

So really I think your pains could be due the the inaccuracy of your probe, as is the case of mine.

Roxy-3D, 2017-02-13:
> I can get a variance of about 0.15mm in the z value when probing the same point.

That is way too much for reliable operation!

Iakouben, 2017-02-13:
Yeah, I know, it's essentially a layer heigh of error that can occur randomly during probing. I'm waiting on a mini differential IR sensor. Friend of mine managed to get this for repeatability with the IR sensor:

    Recv: Mean: 1.971175 Min: 1.967 Max: 1.976 Range: 0.009
    Recv: Standard Deviation: 0.003182

Which is an excellent result in my opinion.

pcsentinel, 2017-02-13:
I don't have a sensor problem, the repeatability test is returning a 0.002 deviation which is brilliant, it just seems to not calculate the z corrections properly.

Iakouben, 2017-02-17:
Got my IR sensor last night and can confirm that I'm also having an issue where tramming isn't quite compensating fully. Printed 5 single layer squares on each corner and center and one was perfect, while the rest had a very loose infill.

Repeatability of the IR after probing same spot 20 times:
    READ: Mean: 0.095729 Min: 0.095 Max: 0.098 Range: 0.003
    READ: Standard Deviation: 0.000946

pcsentinel, 2017-02-19:
I also tried bilinear but that had a similar affect. Given up for now, will retry when have my patience back.

closed 2017-10-31 by thinkyhead