MarlinFirmware/Marlin #5806: SD_CHECK_AND_RETRY not retrying

StrangelyTyped, 2017-02-11:
I've been experiencing SD card read errors multiple times per print, and while digging into the code I think I've found an issue preventing the code from retrying.

On [this line](https://github.com/MarlinFirmware/Marlin/blob/cb4829233822897cf564f9e40e69e0e0310fae1f/Marlin/Sd2Card.cpp#L399) the code decrements the retry counter, and if non-zero exits out of the loop. I believe this check is supposed to be if the retry counter is zero then exit. This appears to affect both command errors and CRC mismatches. Note that this is just from a reading of the code, I haven't attempted to debug to verify the behaviour, but I will be trying out a modified version of the firmware later today to see if the printer still reports read errors.

StrangelyTyped, 2017-02-11:
I'd also like to note that the behaviour of the code when a read error is encountered appears (at least to me) to be somewhat undesirable - when an SD read error is encountered, [this block](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L1133-L1146) commits as much of the command as it's read thus far, which can lead to interesting effects - depending on where the command is sliced I've seen it do anything from move halfway across the build plate (G1 X10|04.2 Y112...., where | is where the command was sliced), to dropping to a rediculously low movement rate (G1 F36|00 X...), or rediculous amounts of reverse extrusion which in my case were at least caught by the PREVENT_LENGTHY_EXTRUDE setting (G1 X123 Y123 E1|3456). 

It may be better to drop the bad command entirely, though I appreciate it may depend on what the command is.

StrangelyTyped, 2017-02-12:
6 Hours of printing later, not one report of an SD read error in the log after inverting the logic on the if statement linked in the initial post.

Bob-the-Kuhn, 2017-02-12:
Since I don't see complaints about bad SD card reads & everyone has the same bad logic then I suspect either your card or your controller has problems.

===========================================================

I agree that the line **if (--retryCnt) break;** is wrong.  Fixing that is definitely a step in the right direction.

If you trace that back up to Marlin_main.cpp you run into this code:
```
    while (commands_in_queue < BUFSIZE && !card_eof && !stop_buffering) {
      int16_t n = card.get();
      char sd_char = (char)n;
      card_eof = card.eof();
      if (card_eof || n == -1
          || sd_char == '\n' || sd_char == '\r'
          || ((sd_char == '#' || sd_char == ':') && !sd_comment_mode)
      ) {
        if (card_eof) {
          SERIAL_PROTOCOLLNPGM(MSG_FILE_PRINTED);
          card.printingHasFinished();
          card.checkautostart(true);
        }
        else if (n == -1) {
          SERIAL_ERROR_START;
          SERIAL_ECHOLNPGM(MSG_SD_ERR_READ);
        }
        if (sd_char == '#') stop_buffering = true;

        sd_comment_mode = false; //for new command

        if (!sd_count) continue; //skip empty lines

        command_queue[cmd_queue_index_w][sd_count] = '\0'; //terminate string
        sd_count = 0; //clear buffer

        _commit_command(false);
      }
```

It fetches a character from the SD card buffer, notices it's the bad read flag (-1), prints a message, throws out the g-code command that it's working on and then merrily grabs the next byte.   

As best I can tell that next byte is in the 512 byte block after the read error.  That means multiple commands will be missing.  It looks like my slicer generates commands that are about 30 characters in length so on my system that would result in 15-20 commands being skipped.  If that gap happens to be in a fill area then it's probably OK to continue the print.  If it's on a surface then there'll be a noticeable defect.

If you want to always abort immediately then all that's needed is to add **stop_buffering = true;** right after the error print statements.

If you want to continue the print then it looks like we'll have to fool around with the command buffer.  From what you're saying it looks like the code would have to throw away three commands:  
- The one where the error occurred (already being done)
- The last command put into the queue may have been incomplete
- The next one coming off the SD card is also probably incomplete.

StrangelyTyped, 2017-02-12:
Actually that's not quite right, when it encounters a read error, it prints out the message but it doesn't throw the partial command away it queues it up for execution, hence my second comment - I've seen it running these partial commands and doing unexpected things. The next command is also generally bad too but usually invalid so less of a danger. I'm not sure I follow your logic with regard to how the previous command could be invalid however, if we didn't encounter a read error during or before the end of that line then the command should be good.

Regardless, I agree there's likely something wrong with my hardware causing these "transient" read errors I'm getting but with the retry logic fixed they seem to now be recovered from without issue, so I can live with that at least for the time being.

thinkyhead, 2017-02-12:
Good catch! Patched at https://github.com/MarlinFirmware/Marlin/commit/e1702816f6fc2c74e20c246faf900f1805c3fa7f

closed 2017-02-12 by thinkyhead

Bob-the-Kuhn, 2017-02-12:
I was wrong about there being an attempt to throw away the data that hasn't yet made it into the command buffer.  It just sits there in the local buffer until data starts coming in again from the SD card.  that explains the beginning of the g-code command mashup.

I expect that the data from the next 512 byte block will not start on a command boundary so I can see where the trailing portion of the mashup is coming from.

thinkyhead, 2017-02-12:
So at the low level, SD always reads in a whole 512 byte sector at a time?

StrangelyTyped, 2017-02-12:
@Bob-the-Kuhn just for reference, here's a snippet from a log where a read error occurred, showing what it tried to run, taken with M111 S7, as you can see when the read error occurred both parts of the command in progress get queued up separately, and if PREVENT_LENGTHY_EXTRUSION hadn't been turned on the extruder would likely have spit the filament out mid print, and this is just one example of what can happen.

```
echo:G1 F3600 X99.14 Y110.454 E1
echo: too long extrusion prevented
echo: too long extrusion prevented
echo:025.37114
echo:Unknown command: "025.37114"
echo:G1 F3600 X97.865 Y108.048 E1025.40111
echo: too long extrusion prevented
echo: too long extrusion prevented
```