MarlinFirmware/Marlin #5807: RC8 and RCBugFix: Shrinked Layers (Michelin Man Results)

bhorn, 2017-02-11:
I recently switched from 1.0.2-2 to 1.1.0-RC8 and confirmed it on RCBugfix 1e4d4e5.
My print results on 1.0.2-2 where ok I guess. Needed some minor Hardware-Tweaking.
But when I try to print something on RC8 or RCBugfix 1e4d4e5 the hull looks like the Michelin Man:

[image]
From left to right:

1. **1.0.2-2**
2. **RC8** You see the rings?
3. **1.0.2-2** Back at 1.0.2 but with a faster acceleration
4. **RC8** Again with same acceleration of 3th
5. **RCBugFix 1e4d4e5** Still got the rings

These where 5 subsequent prints with the same PLA, exact same G-Code, same configuration (from 2nd to 3th the acceleration changed), same extruder, same hotend, same day and same everything.

Any Ideas? Since the problem is gone if I go back to 1.0.2-2 I consider it a Marlin-Bug rather then a hardware issue.

epatel, 2017-02-11:
Do you print from USB or SD?

epatel, 2017-02-11:
This? Heatbed behaviour? https://3dprinting4u.wordpress.com/printing/

emartinez167, 2017-02-12:
I can confirm that ensuring the heated bed is at the right temperature will get rid of those patterns. Worked for me.

Regards,

Ernesto

bhorn, 2017-02-12:
@epatel @emartinez167 I figured out the Problem. Looks like its linked to my other issue (5805) in coop with a falsely pid-tuned bed (90c instead of 65°)

**Feature Request**: Support for multiple PID's and automaticaly choose the nearest tuned one (i.e. if Tuned values are 60°c, 70°c and 80°c and you want to print something at 64°c bed, marlin should select the 60°c PID)

closed 2017-02-12 by bhorn

thinkyhead, 2017-02-12:
> Feature Request: Support for multiple PID's and automaticaly choose the nearest-tuned one

We may consider this after the 1.1 release.