MarlinFirmware/Marlin #5809: Request: checkExtruderAutoFans during long operations (G28, G29)

chrissv, 2017-02-12:
Using Marlin 1.1.0-RC8 with correctly configured extruder cooling fan (via E0_AUTO_FAN_PIN). While idle, when the hot end heats up and goes above the EXTRUDER_AUTO_FAN_TEMPERATURE threshold, the fan comes on just fine.

However, my slicer G-code starts the hot end heating up (via M104 S220) and then executes a G28 followed by a G29. Those operations take multiple seconds, depending on the starting Z height and the type of bed leveling.

If the hot end reaches EXTRUDER_AUTO_FAN_TEMPERATURE during the G28 or G29, the fan does not come on until those operations are concluded. I am worried at a possible clogging because the hot end isn't cooled quickly enough.

This request is to execute the checkExtruderAutoFans function within the code for any functions which may take a "long" time (like anything over a second), such as G28 and G29. That way the fan will come on as soon as the threshold is met.

Roxy-3D, 2017-02-12:
> This request is to execute the checkExtruderAutoFans function within the code for any functions which may take a "long" time (like anything over a second), such as G28 and G29. That way the fan will come on as soon as the threshold is met.

Actually...   I was thinking something similar to what you are requesting.  Except the motivation was a little bit different.  My fan shroud on my new printer is very close to the nozzle.   I'm worried that if the nozzle is hot, and especially if it isn't moving much, I'm worried it will start to melt the plastic duct work.

thinkyhead, 2017-02-12:
Checking the code in `RCBugFix`, it looks like `Temperature::checkExtruderAutoFans` is being run whenever `idle()` or `manage_heater()` is called. And `idle()` is called frequently even when `G28` and `G29` are underway. So I'm not sure what's preventing the fan from starting up as soon as the required temperature is reached.

chrissv, 2017-02-12:
I cold have been mistaken in my observation - it's hard to judge when the temperature crosses without feedback. I'll do some experimenting and post any contradictory results.

thinkyhead, 2017-02-15:
Recent patches to the temperature code might have a beneficial effect. Give the latest `RCBugFix` another spin when you have the time.

chrissv, 2017-02-15:
With no firmware changes, I did a test of moving my Z axis very high (so a G28 takes a long time). Then I turned on the hot end and did a G28. Indeed the fan did start running midway through the G28. So it looks like my initial observations were not accurate. This appears to be a non-issue. Thank you everyone for your comments and help!

closed 2017-10-31 by thinkyhead