MarlinFirmware/Marlin #5810: FILAMENT_RUNOUT_SENSOR and M119

Roxy-3D, 2017-02-12:
I just wired up a filament run out sensor...   And with the filament in position and everything as it needs to be to print, M119 reports the Filament Run Out Sensor is TRIGGERED.   That seems backwards to me.   It reports 'Open' when the filament is missing and needs to be changed.

Should we flip the nomenclature?

Blue-Marlin, 2017-02-12:
`FIL_RUNOUT_INVERTING`?

Roxy-3D, 2017-02-12:
Yes that control is available.   But when everything is normal and the extruder is ready to print, M119 reports the 
```cpp
filament_Runout_Sensor: TRIGGERED
```
That seems backwards to me.  If we are going to have that report 'Triggered', the name of the sensor should be something that conveys 'Filament is loaded and available'.

Do you have a filament runout sensor?   If so, give it an M119 and see what it reports.

Tannoo, 2017-02-13:
Would this be better?

In `language.h`:
```
#define MSG_FILAMENT_RUNOUT_SENSOR          "Filament: "
#define MSG_FILAMENT_RUNOUT_LOADED          "Loaded"
#define MSG_FILAMENT_RUNOUT_UNLOADED        "Unloaded"
```
In `endstops.cpp`:
```
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    SERIAL_PROTOCOLPGM(MSG_FILAMENT_RUNOUT_SENSOR);
    SERIAL_PROTOCOLLN(((READ(FIL_RUNOUT_PIN)^FIL_RUNOUT_INVERTING) ? MSG_FILAMENT_RUNOUT_LOADED : MSG_FILAMENT_RUNOUT_UNLOADED));
  #endif
```

Tannoo, 2017-02-13:
Reports: `Filament: Loaded` when ready to go and `Filament: Unloaded` when it's not.

Roxy-3D, 2017-02-13:
> Reports: Filament: Loaded when ready to go and Filament: Unloaded when it's not.
> Maybe Filament: Out would be a better report?
Yes...   This seems more logical to me.   Let's see if the idea gets any adverse reactions.

Tannoo, 2017-02-13:
This addition hasn't been merged yet. It was part of the M600 Nozzle shutdown PR.

Roxy-3D, 2017-02-13:
> I don't believe this addition has been merged yet. It was part of the M600 Nozzle shutdown PR.

Are you suggesting there is an outstanding Pull Request we can piggy back on?   If so...   I'm 'on board' !

Tannoo, 2017-02-13:
Look at the files changed in #5794, I added it when you about to setup your runout sensor. It was to help with checking it. I didn't have that option when I setup mine (more than a year ago).

Tannoo, 2017-02-13:
I can't PR against this yet. Still getting the hang of the new way to do git without special privileges.

Roxy-3D, 2017-02-13:
> If you can PR the changes, be my guest.

I'm worse at Git than you are!    Please keep trying!

Tannoo, 2017-02-13:
I used to be good at the command line stuff back in the DOS days and even dabbled in Linux.
Learning a new command line structure is like trying to learn to make a perfect print by typing out the gcode in notepad without the help of a slicer.

I'm working on it, though.

Roxy-3D, 2017-02-13:
Supposedly, Git is helpful and is indispensable.    But I'm starting to be of the opinion that people that say that are part of a cult.   It isn't intuitive and simple things I want to do are impossible.

Tannoo, 2017-02-13:
I agree. But, nothing is impossible. It's only a matter of understanding.

thinkyhead, 2017-02-15:
Filament runout sensor is perhaps named poorly. Filament presence sensor is what it really is.

thinkyhead, 2017-02-15:
> It isn't intuitive and simple things I want to do are impossible.

Just ask for guidance. I know it well.

closed 2017-10-31 by thinkyhead