MarlinFirmware/Marlin #5812: BLTouch unresponsive after update to RC8

DynamiteHack, 2017-02-12:
**@Roxy-3D** I was reading the description on the UBL-Dev Branch and noticed you require an LCD panel. I was using the Discount-LCD before I installed the BLTouch. I found I couldn't run them both. I ead somewhere that it was a power issue. Didn't seem like a big deal so I put it off. What did you do to resolve the issue. Oh and I just received the LCD 12864 Graphic Smart Display Controller that I ordered so I'd probably want to use that one.
Thanks!!

Roxy-3D, 2017-02-12:
> I was using the Discount-LCD before I installed the BLTouch. I found I couldn't run them both. I read somewhere that it was a power issue. 

So...   I'm not sure what to tell you.   I don't see any power issues with the BL-Touch probe.  It doesn't take much power.    But...   I have a RAMPS board on all of my printers.  What I do is run +5 volts from an external source to the +5 pin of the power connector next to the servo connectors.   I do NOT jumper the VCC pin to the +5v pin of the power connector next to the servos.    

One thing you might want to look at doing is this:   On the RAMPS board you have the big green +12 volt power connectors coming into the board (to drive the stepper motors and extruders).  On the bottom side of the board you can solder a 3 legged LM7805 to get solid clean power for the servos.   Two of the legs get soldered to the +12v and ground of the big green connector.   And the last leg of the regulator can be wired to what ever needs +5 like the +5 pin of the servo power connector.   It is a little bit of a hack, but it can be done very cleanly and with no chance of a short circuit.   The LM7805 just ends up between the RAMPS board and the processor board.

Bob-the-Kuhn, 2017-02-12:
There's a chance that enabling the LCD resulted in one or both of the BLTouch signals being re-assigned to pin(s) that particular LCD isn't using.

If you attach the config files where the BLTouch worked and the config files with the LCD added I can see if there is conflicting pin assignments.

What do you mean by "didn't work"?  
- wouldn't even do the power up deploy/retract self test (that's definitely power)
- wouldn't deploy automatically
- deployed but the head crashed into the bed

DynamiteHack, 2017-02-12:
**@Roxy-3D** - I will try that hack. I do have the jumper in place to power the BLTouch. 

**@Bob-the-Kuhn** - Actually, the LCD was installed before I installed the BLTouch. After I installed the touch, the display only displayed garbled text. I just pulled the display off leaving it enabled in the config file. I enabled it in the new config file figuring I'd figure out how to use it sometime.

> What do you mean by "didn't work"?

With the new config file that I posted earlier my BLTouch powers up and runs its self test.That is all. When homing, the pin never drops and the LED remains lit. The BLTouch does not respond to M280 commands at all.

The configuration that worked in the past is attached. To be clear, I removed my display _after_ installing the BLTouch. The BLTouch always worked, it was the display that gave me problems.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/769491/Configuration.zip)

DynamiteHack, 2017-02-13:
**@Roxy-3D** - Okay. So I tried the hack, extending the +5v to the pin next to the servo plugs. Pictures are attached to confirm I understood what you explained. After making the change I uploaded the UBL sketch you suggested. I attached the discount LCD I have and powered up. Everything looked good. I had a LCD display working properly and the BLTouch glowed!!

I did a G28 to home the nozzle. X homed correctly. Y homed correctly. As soon as the nozzle attempted to start the Z homing, the LCD went blank and a puff of "magic smoke" escaped from the area of my hack. I did notice when I disconnected the printer from the main power, the LCD turned back on, still getting power from the USB. Was there anything I should have changed in the settings before  I tried to run it? Something pin related maybe??

I took everything apart. Arduino seems okay. Hopefully the Ramps is okay and I just lost the regulator. I don't want to power back up until I know if there is some setting I should have changed.

Sorry. My first time doing this. Exciting stuff!!

[image]
[image]

Roxy-3D, 2017-02-13:
That looks correct.   The middle leg is connected to ground and the input is on the +12 volt connection. Did you measure the voltage on the blue wire before plugging it in?  I would have done that, but it sure looks like that should have worked.    Did you visually inspect there was no short circuit when you plugged the two boards together?   There aren't any metal parts on the processor board where you soldered the LM7805 so it should have been OK.   I even pulled out a MEGA-2560 / RAMPS board and looked at the mounting holes.  Even if you were using metal screws to mount the boards, they are clear of where you did the work.  

Knowing what fried would be helpful to figuring out what went wrong.   I've done this 3 or 4 times and never had a problem.  Did you see where the smoke came from?   

> Was there anything I should have changed in the settings before I tried to run it? 

No.  The processor board is setup so that it doesn't care where the power it gets is coming from.   It can come from the power pin where you soldered the LM7805 or it can come from the 12 volts coming in from the big green connector, or it can come from the USB.

Bob-the-Kuhn, 2017-02-13:
I believe that the 7805 was defective.   It has thermal overload and short circuit protection so it shouldn't have lost it's magic smoke.

DynamiteHack, 2017-02-13:
Thanks for the response. I'll put things back together and see if I can figure out what went wrong. I did not measure the voltage coming out of the wire. I'll try that next...I'm going to put a new regulator on and see what happens. I'll let you know...

Roxy-3D, 2017-02-13:
I would take the regulator off....   and see if everything still works as it did before.    And if that is the case, I would measure the voltage on the pin I soldering the input of the LM7805.   It should be a rock solid 12 volts.   I don't think this is happening, but if it was 20 volts, that could fry the LM7805.

The thing is, LM7805's are very tough parts.   And they can with stand a lot of abuse.   It is very strange if that part died as soon as power was applied.    It might also make sense to disconnect any and all servos plugged into the servo connectors the first time you power up.   After you measure the voltage coming out of the LM7805 you can verify it doesn't change when you connect the blue wire to +5 Volt line of the servo power connector.   

Other than you had a bad experience and saw smoke...   None of these extra measurements are really necessary.   When I've soldered LM7805's there, I've just connected them up and plugged in the servo's.  Everything has always gone as expected.

DynamiteHack, 2017-02-14:
**@Roxy-3D** - So I replaced the 7805 and most everything is working as expected. It was either a bad 7805 or my connection to the leg was too loose causing a spark? I crimped the connector to the leg and it is working.  I have both the display and BLTouch powered now.

I ran a G28 and XY homed as expected. Z however made the coordinated move to the centerish of the bed, raised about 5mm paused and continued up 13 and stopped. I know it's 13 because that is what ends up displayed in Repetier. The pin never dropped in the Touch. Using Repetier to make manual moves, polarity is correct and the head mover along the Z axis.

I have not touched any settings in the UBL branch you gave me. I will start investigating those now.

Thanks for your patience!!

DynamiteHack, 2017-02-14:
As a sanity check, I went back to my original firmware package. The Homing worked fine and the BLTouch deployed. I still have the problem with the z-offset not bring reflected in the final position to begin printing the first layer. At least I have demonstrated that the BLTouch remains functional.

I'd love to be on the latest firmware so let me know where I might start looking...I have attached the firmware that I have installed now.

Thanks!
[FT-2020-DISCOUNT-LCD_BLTouchAdded.zip](https://github.com/MarlinFirmware/Marlin/files/772911/FT-2020-DISCOUNT-LCD_BLTouchAdded.zip)

Roxy-3D, 2017-02-14:
> It was either a bad 7805 or my connection to the leg was too loose causing a spark? I crimped the connector to the leg and it is working. I have both the display and BLTouch powered now.
> I ran a G28 [...]

I just got home from playing darts and I always win a few beers.   So I'll comment more fully in the morning.   But if you fired up the default UBL Configuration.h file I have it set up to do 'Safe Homing' in the center of the bed.   With a mesh based leveling system, it just makes sense to zero the center of the bed.   If you go back to the UBL code, the new BL-Touch code will be there soon enough.

DynamiteHack, 2017-02-14:
Cool!! Darts rock!!

Mesh editing, huh? I've gotta get myself up to speed quick!! It's pretty cool what you guys are coming up with. Don't know how you do it but, keep it up!!

I'll pull the bugfix down in the morning and play around! I just realized that my display only works with the usb connected to my laptop. Not sure if I hurt something or not but, not a huge deal as I'll be moving to octoprint once I get all the issues ironed out.

Thanks again!!

thinkyhead, 2017-02-14:
Ref: #5705

DynamiteHack, 2017-02-18:
@thinkyhead Feel free to close this ticket if you'd like. My Arduino stopped accepting uploads during all of this. I have a new one and hope to figure out the other one so I have a spare. 

I am going to stick to my old version of firmware since it works. Went down so many rabbit holes trying to update my firmware I think I'll take a break for now and follow the progress others make.

Thanks for the assistance **@Roxy-3D** and **@Bob-the-Kuhn**! I did get another 5V power source out of this process and for that I'm grateful!!

Cheers!
'hack

Bob-the-Kuhn, 2017-02-18:
I was able to resurrect one of my controllers by buying a USBtinyISP and then using the Arduino IDE to burn a new bootloader.

DynamiteHack, 2017-02-18:
Thanks. I have an UNO that I may try to use. I'll look into the USBtinyISP, though.

Roxy-3D, 2017-02-18:
@DynamiteHack    Really...   For $15 (including shipping) you can get a RAMPS /Mega-2560 on eBay!   If you want an easy way to keep your printer running...  That is the way to go!   At least half of the people here are using that setup.

DynamiteHack, 2017-02-18:
I happened to have a Saint-Tech controller kit. Just wasn't expecting to use it so early. :)
Is there any benefit to switching to an MKS board?

Roxy-3D, 2017-02-18:
Well...   The big advantage is how standard it is and how many people have it.   You will be in the main stream if you plug that in.   That is pretty much what I was suggesting you consider doing!

DynamiteHack, 2017-02-18:
Gotcha!! I'm a little slow on the uptake at times!! 
I'll give one of those a shot!
Thanks!

Augermsiester, 2017-03-07:
I am having the same printer and exact same problem as the OP. I have uploaded RC8 firmware, but not the RCBUGFIX. Would that be the next logical step?

Bob-the-Kuhn, 2017-03-08:
You can run a BLTouch and an LCD separately but not at the same time?

This thread is running in the direction of a power issue.  The 7805 method has the advantage of being compact & neat.  The alternative is to get a 5V power supply to power the BLTouch and the LCD.  That's a more robust method but requires more cabling.

Independent of the power issue, the latest RCBugFix is highly recommended when using a BLTouch.  It has several BLTouch bug fixes and enhancements.

Roxy-3D, 2017-03-08:
@Bob-the-Kuhn     This is a little bit off topic...   But THANK YOU for helping with so many BL-Touch issues and other RC-8 RCBugFix topics!

Tannoo, 2017-03-08:
No problem. I try to help where I can.

Bob-the-Kuhn, 2017-03-08:
I'm retired & I like to trouble shoot so it's a natural fit.

Augermeister, 2017-03-08:
@Bob-the-Kuhn  I worked that out by providing separate 5v power to the BLTouch before I updated to RC8. BLTouch was working before upgrade, now no reaction to pin up or pin down commands; just the self test at power up. I'll try the RCBUGFIX! Thanks.

Bob-the-Kuhn, 2017-03-08:
Worked before updating to RC8.  Interesting

When you update to RCBugFix please copy your machine specific items over to the new config files.  You can get into trouble if you just drop your old config files into the RCBugFix directory.

If you still have the problem after updating then I'll want to see your config files and any other files that you changed.  Just ZIP them up and drop onto your reply.  I'll also want to know what pins you're using for the BLTouch.

Augermeister, 2017-03-08:
@Bob-the-Kuhn thanks for your help. I did copy my machine specific items over to RC8, as I saw how different the configuration.h file is now. I will do the same with RCBUGFIX and see how that goes. My pins for BLTOUCH are as suggested by the manufacturer. Z-min is pin 18 on RAMPS 1.3 board and the first servo signal pin. 5v is from my power supply and is verified at 5v with voltmeter.

Roxy-3D, 2017-03-08:
> 5v is from my power supply and is verified at 5v with voltmeter.

The 5 volt supply should be verified while the servo (or in this case, the BL-Touch) is moving.   That is when the current required is the highest.

Augermeister, 2017-03-08:
@Roxy-3D I will try to get a reading when BLTouch does it's self-test at initial power-up. Thanks!

Roxy-3D, 2017-03-09:
Unfortunately...  You might need an oscilloscope to do it.  The problem (especially with RC Servos) is the motor takes a huge amount of current when it is sitting still.   That is because it isn't moving and is not generating any back EMF.   But the same principle applies to the inductive pin on the BL-Touch.    With a normal volt meter, you may not see the voltage drop when it tries to deploy.

Augermeister, 2017-03-11:
I got my issue figured out. I had been using the old firmware that came with my Folgertech and had RAMPS 1.3 setup in configuration.h 
Although board version isn't labeled, I tried RAMPS 1.4 boards and wala, it worked. Different pin assignments for the servo!
Thanks for your help. Now to research z-offset issues...

Bob-the-Kuhn, 2017-03-11:
z-offset issues?

Is this with bilinear leveling or with the probe's Z offset?

Augermeister, 2017-03-12:
I'm having z-offset issues. can anyone direct me to a step by step z-offset setup process?

Roxy-3D, 2017-03-12:
> I'm having z-offset issues. can anyone direct me to a step by step z-offset setup process?

Personally...   I think it is best to have the Z-Offset set at 0.000mm    And then adjust the Z_PROBE_OFFSET_FROM_EXTRUDER number to be exactly correct.    That is what I do.

Incidentally...    In the UBL code I'm thinking about eliminating the Z-Offset.   It will save some floating point additions.   But in truth, there is no need for it.   The Mesh is already adding and subtracting offsets from the Z position.   It would be better to just adjust the entire mesh up or down instead of having a separate Z-Offset that needs to be added into each position calculation.

Augermeister, 2017-03-12:
I thought Z-Offset and Z_PROBE_OFFSET_FROM_EXTRUDER were the same thing. I looked and couldn't find Z-Offset in the configuration.h file (RCBUGFIX), so thought it had been removed already.

closed 2017-10-31 by thinkyhead