MarlinFirmware/Marlin #5820: RAMPS 1.4plus with RRD fan extender

heimerdinger91, 2017-02-12:
I have been scouring the Internets for a week, attempting to figure out a solution to this. I've tried multiple setups and code tweaks but nothing gives me control of my part cooling fan. Here's my problem:

1. D9 mosfet is fried and cannot be used for PWM part fan control

2. I purchased an RRD fan extender board as a workaround.

I cannot for the life of me determine the following:

A. Correct placement of the fan extender on my RAMPS 1.4plus board (which is of course a different setup entirely than the standard RAMPS board that everyone else in the world has.)

B. The proper way to assign a fan control pin in marlin once I DO get the board placed correctly.

C. A semiscientific method of methodically testing setups one-by-one

--------

Currently I have the fan extender board (FEB) powered by the servo pins next to wifi. The FEB is on pins D4 D6 A5 and A10 in the AUX2 grouping next to the MOSFETs.

Can anyone please provide some sort of guidance here as I'm getting rather frustrated.

Many thanks

Bob-the-Kuhn, 2017-02-13:
If you can point me to the schematics for your particular flavor of fan extender card and the RAMPS 1.4plus then I'll get back to within a day on how to set it up.

Tannoo, 2017-02-13:
Like the one mentioned here? #5332

Bob-the-Kuhn, 2017-02-13:
Tannoo - thanks for the pointer.

@heimerdinger91 - what I need from you are:
- Pointer to the schematics for your particular flavor of fan extender card or at least the URL to where you bought it
- Your pins_your_board.h file
- Your Configuration.h file
- Your Configuration_adv.h file

heimerdinger91, 2017-02-13:
Yes Bob-the-Kuhn they're different, which is what's causing me the headache. Seems like the fan extender is pretty much plug and play on a regular RAMPS 1.4, but clearly not so for RAMPS 1.4 plus.

Last night I looked at pictures of a regular RAMPS 1.4, and it actually has 12 pins on the AUX1 group to match the 12 on the fan extender. My 1.4plus has only 4 pins there, which I was suspicious of already. 

If you think there's still a possibility of making it work, I can post my pin and configuration files but honestly at this point I think it will be easier to buy a regular RAMPS 1.4 shield and mega 2560 board. Then I'll have my d9 back AND somewhere to put the fan extender.

heimerdinger91, 2017-02-13:
Edit: sorry there are 8 pins on my AUX1 group, I only have 4 of them connected at the moment

Bob-the-Kuhn, 2017-02-13:
I won't know for sure if your fan extender can be adapted unless I see it's schematic.  

I've read a couple of threads where supposedly the same extenders were in fact different.  That's why we need to go to wherever you got you unit from & get the schematic from them.

Have you considered replacing D9?   Just get the numbers off your bad D9 and call Digikey.  I expect you'll pay less than $1 for the part plus $3 for postage.

heimerdinger91, 2017-02-13:
https://m.aliexpress.com/item/32392297400.html#autostay

Above is the aliexpress page for the extender I bought.

My original thought way back when I first blew the d9 was to replace it, but I'm not very experienced with PCB soldering and would like to avoid doing anything risky on what is currently my only printer board.

Someone on facebook has suggested an external mosfet module and pin reassignment, which I would have jumped on right away had it been suggested earlier. 

http://m.ebay.ca/itm?itemId=321485262345

I will take a few minutes to write out the AUX pin setup on my RAMPS 1.4 plus

Roxy-3D, 2017-02-13:
> Have you considered replacing D9? Just get the numbers off your bad D9 and call Digikey. I expect you'll pay less than $1 for the part plus $3 for postage.

My experience with Digikey isn't very good.  They have a minimum $25 or $50 order size.    But what I did was pull a MOSFET off of an old RAMPS 1.4 board to replace a blown FET on another board.   My guess is the RAMPS Plus board would use the same FET's  (but of course...  that should be verified!)

heimerdinger91, 2017-02-13:
@Roxy-3D do you have any spare FETs from that board? :D

I think I'm going to go with the external MOSFET module for now and if we can figure out how to get the fan extender working I'll have bonus PWM channels.

Bob-the-Kuhn, 2017-02-13:
Maybe I've ordered enough from DigiKey that I no longer have a minimum.  I just go online & place my order.

@heimerdinger91 - the easiest way to determine what's on the header is to use the M43 W1 command and then use a jumper (with a resistor about 1k) to short the pins to ground.  You'll need to enable **pinsdebugging** in configuration_adv.h.

The M43 W1 command kicks off a background task that monitors all the pins for changes.  When a change is detected then info on that pin (pin number, name if assigned) is sent out the USB port.

Roxy-3D, 2017-02-13:
>  do you have any spare FETs from that board? :D

Sorry...  I burned up a MOSFET with some sloppy wiring.   And I broke the leg off of the first MOSFET I tried to remove.   And I succeeded on the last MOSFET.   But at this point...  There are no MOSFET's left on the board.    Otherwise, I would have offered to take the last one off and mail it to you.    

And unfortunately...  eBay isn't a great deal.  There are no single STP60NF06L chips cheap.   The best listing will let you get 5 of them for $4.50.    That is why the board with the blown MOSFET became a donor!

http://www.ebay.com/itm/5pcs-STP60NF06-P60NF06-MOSFET-TO-220-/170865604369?hash=item27c8623711:g:ZtoAAOxyLchRt82Q

What I actually did was take one MOSFET off the board to put in parallel with a MOSFET on another board to double the current capability of that pin.   It turns out...  The original single MOSFET had plenty of current capacity.

heimerdinger91, 2017-03-08:
Hi all I'm back. Sorry for the long delay. Work, illness, lame excuses, etc etc. 

I got my hands on an external MOSFET module for my new blower fans, exactly as pictured on eBay.

http://m.ebay.ca/itm/5Pcs-IRF520-MOS-FET-Driver-Module-for-Arduino-Raspberry-pi-New-/321485262345?txnId=1713815488011

Before I wire everything up and cause problems by trying to reassign fan pins, is replacing my D9 as simple as snipping the VCC pin, soldering SIG and GND to corresponding pads on my fried D9 (with or without removing the original?), and then wiring my fan to V+/V- and my power source to Vin and GND?

Or would it be safer to just plug the MOS module into my D11 servo pins and try to switch the fan control to D11?

Roxy-3D, 2017-03-08:
Or...   If it is a 'compatible' MOS-FET (and almost for sure it is!), you could unsolder the bad one from your board, and put that one in its place.

heimerdinger91, 2017-03-08:
But then I would feel silly for not thinking of that and over-complicating it

Bob-the-Kuhn, 2017-03-09:
Digital pin 11 can't be used as a hardware PWM because the timer used by this PWM is in use by Marlin for the stepper ISR.

Digital pin 11 can be used as a software PWM.  The software PWM updates at least 8 times a second so you won't notice variations in the fan speed.

To switch to D11 you'll need to edit your pins_YOUR_BOARD.h file.

To guarantee functionality the unmodified driver module is the better choice.  Unfortunately it also means a lot more cabling.  

Your proposed wiring sounds right. I suggest doing a few sanity checks to be sure how to hook it up.  Please verify the following:
- Both GNDs should be tied together.  
- Vin should be tied to V+.

I like the idea of soldering to the pads on the fried D9.  It's not a good idea to solder the pins of the module directly to the RAMPS PLUS board.  The mechanical stresses will probably eventually rip the solder pads off the board.  Better to use some flexible wires.

I think the best way to attach the wires would be to clip the pins at the body of the bad FET and then solder the wires to the pins.

I don't see any functional issues with clipping the Vcc pin.  I think it only powers the LED on the module.

----

On the other hand there's very little to loose by taking the FET off one of the 5 modules you have.  

First go to YouTube and look at a few soldering videos.

Clipping the leads is the easiest way to get them off the boards.  Before you do that on the module look carefully at where the leads are soldered to the board.  A lot of these FETs have a wider pin at the body and then steps down to a narrower section that gets soldered into the board.  If the pin at the solder connection is wider than the pin on the bottom side then the narrow part is soldered in & you can't just clip it.  If that's the case you'll need to put the module into a vice, use a pliers on the heatsink to apply a rocking motion/force and then apply a 40W (or smaller) soldering iron to the bottom side of the module on one of the edge pins until it comes loose.  You'll need  to apply enough force with the pliers to bend the other edge pin in order to get any movement.  After that it's just a matter of going up & down the pins with the soldering iron and wiggling it out.  Don't worry if you destroy the module's board.  You'll need to straighten the pins on the FET and remove any globs of solder before attaching it to the RAMPS PLUS board.

On the RAMPS board clip the leads at the body of the FET.  Clamp the board, use a pliers to pull on a pin, apply a little solder to the soldering iron and  apply the soldering iron to the bottom side of the board for 5-10 seconds.  Don't do it more than 15 seconds as you can damage the board.

Roxy-3D, 2017-03-09:
> Don't do it more than 15 seconds as you can damage the board.
Or...  Cook the part!!!

heimerdinger91, 2017-03-09:
Looks like I f**ked up my board, woohoo! 

I took off the dead mosfet, put on the new one, wired everything back together...
And my computer won't recognize the printer and the LCD screen doesn't even turn on. 

*takes printer to a field*
*smashes printer with baseball bat*

Tannoo, 2017-03-09:
First, remove that FET and try again without it.

closed 2017-10-31 by thinkyhead