MarlinFirmware/Marlin #5824: G28 z Forbidden

PeachLie, 2017-02-13:
I just wanted to install new motor driver. Aber that i updated im Marlin just the steps/mm and now i get the message "G28 z Forbidden". I can move all axis and the endstops are working fine. The z endstop is a capazitive sensor. When i am holding an object under the sensor and send M119 i get a "triggered" as receive. But when i manually drive the bed in negative z direction it doesn´t stop although the sensor is triggered ... I hope somebody can help me :(

PeachLie, 2017-02-13:
Ok i deactivated safe homing, now it works ..

psavva, 2017-02-13:
@PeachLie , Please can you close this incident.

thinkyhead, 2017-02-14:
Sorry about that, @PeachLie. See http://marlinfw.org/docs/configuration/configuration.html#z-safe-homing

closed 2017-02-14 by thinkyhead