MarlinFirmware/Marlin #5825: Where is the Gcode Parser?

MrJulz, 2017-02-13:
If one needs to add their own custom MCodes to Marlin, where is this done?  Aka in which source file is the parser?

Apologies I could only find code for the emergency parser.

(Is there any document that gives an overview of the code structure?)

Blue-Marlin, 2017-02-13:
The parser is in `Marlin_main.cpp`.

MrJulz, 2017-02-13:
Thankyou.  Note: for others searching, look for:  process_next_command()

(am slightly surprised the g/m code parsing is not in its own unit)

closed 2017-02-13 by MrJulz

thinkyhead, 2017-02-14:
> am slightly surprised the g/m code parsing is not in its own unit

Marlin_main.cpp is basically the GCode parser writ large, along with every GCode command handler.

It will be broken out into its own unit (well, broken up into several .h files but still within a single compilation unit) after the 1.1 release, as we begin to work on 1.2.