MarlinFirmware/Marlin #5826: LCD displays menu options on one line

pearson, 2017-02-13:
Here's an odd glitch I ran into today using the RCBugFix firmware. When switching to the "Prepare" menu, the menu options all appeared on one line. I've only seen happen the one time, although it's persistent until the board is reset. (As the video below shows.) It happened after choosing the pre-heat option from the "Prepare" menu, but I've chosen that option in the past without issue.

I've put [a video of what it looks like onto YouTube](https://youtu.be/XYzhs8GQn0g).

The printer is a HICTOP 3DP-17, a Prusa i3 clone.  The display itself is a 20x4 2004a LCD. The LCD board is marked as a MLCD2004.2

My configuration has the following set:
```
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define LCD_LANGUAGE en
#define DISPLAY_CHARSET_HD44780 JAPANESE
```

Please let me know if any other information would be useful! As I'm new to this firmware, it'll be a while before I'm at the point of being able to debug this myself. :-)

Tannoo, 2017-02-14:
If you are running the RCBugFix version, this should have been addressed here: #5647

Tannoo, 2017-02-14:
Maybe you found a new bug though.

Roxy-3D, 2017-02-14:
> Maybe you found a new bug though.

Agreed.   Does it behave normally some of the time?   And then switch into this mode?

pearson, 2017-02-14:
Yes, it behaves normally most of the time. I've now seen the problem a couple of times AFTER printing an object. What seems to happen is this:

1. An object is printed (in PLA)
2. The print isn't removed right away, so it is left to cool.
3. The LCD "Prepare" menu is used to reheat the machine for PLA (to help remove the object and to remove the filament)
4. The "Prepare" menu is used in hopes of choosing "cool down", but the "Prepare" menu becomes one line, and not all options are appearing on that line. (Including "cool down")

A couple notes:
1. The main menu always displays correctly, even after leaving the one line "Prepare" menu.
2. I have NOT checked if other menus display this problem.
3. Later today I'll do a clean clone of the bug fix branch to confirm that I have all the current patches, and to see if this is still repeatable.

I should also throw in the obligatory "good job" on this project. 🙂 This firmware is really good stuff! I'm looking forward to contributing to it in some way.

Roxy-3D, 2017-02-14:
The reason for the questions "Does it power up and work correctly?  What did you do to make it start displaying incorrectly?"   is this:   Marlin knows from the configuration.h file how many lines are on the display.   If all of a sudden it is only allowing one line, I'm wondering if we have a data corruption problem and that information is getting zapped.

thinkyhead, 2017-02-15:
@pearson So far I'm unable to reproduce the issue. It does look like the sort of issue that might be caused by a buffer overflow.

pearson, 2017-02-15:
Currently *I* am unable to reproduce this issue. I've since changed my ending g-code script, so perhaps that is a factor in this since the issue happened after printing. I'll try to get back to a state where I can reproduce the issue.

pearson, 2017-02-16:
My apologies. I believe this may have been my own mistake with the firmware versions. I simply cannot reproduce this bug after pulling the latest bugfix firmware.

closed 2017-02-16 by pearson