MarlinFirmware/Marlin #5827: Cant save to EEPROM

elfag, 2017-02-14:
When i use M500 in Marlin RC8 to change Z offset the saved offset is reset when i power off and reboot

thinkyhead, 2017-02-14:
Are you referring to the Z offset set by `M851`? Do you have `EEPROM_SETTINGS` enabled in your `Configuration.h` file?

elfag, 2017-02-14:
Yes i am. Yes its enabled. I can use M851 to change offset. But when i tur off the printer the offset is back to zero

elfag, 2017-02-14:
Oh no. EEPROM was not enabled. I am so sorry. I was thinking it was enabled on default and it must be enabled of me to use m852 in the first place. Sorry for wasting your time

elfag, 2017-02-14:
I tested it now. It works. I miss the the save function in my LCD though.

Tannoo, 2017-02-14:
Should listed as 'Store Config' in the 'Control' menu.

closed 2017-02-15 by thinkyhead