MarlinFirmware/Marlin #5828: pb with bed thermal runaway using part fan

jfpion, 2017-02-14:
When using a part fan (bridges or small parts  the part fan is so efficent that my heated bed (not so efficient) cool to much and the printer halt.
I can ask the slicer to blow less but this time it is not enough for the bridges.
I can disable the safety, but a safety is supposed to be on 
is there a way to add a trigger to disable the thermal runaway protection for the bed (like xx% for the m106 command and yy sec after to let it the time to reheat.
as the bed temp is not as critical than extruder temp, it would keep the safety and make use  of the fan easier.

calibration of the trigger could be simple move the head over the bed sensor got to z1, try the different speed for the fan and you easily get the minmum speed which allow the bed to stay warm enough

the pb is mostly with the use of high temp bed (ABS for exemple) when there isn't a lot power in reserve

thank you

fiveangle, 2017-02-14:
That's an interesting, albeit clumsy solution for poor hardware design that is probably better addressed by correctly adjusting the Marlin thermal control parameters to better match your printer needs, but I'll let others comment on that.

How I solved the poor HW design on my PB anemic heat bed was to change the power supply from 12v ATX to a 15v meanwell 200w one. It made a huge difference. This is RevF Printrboard. I took the time to add a small aluminum heat sink to the FET that controls the heat bed but it probably wasn't necessary.

Good luck !

-=dave

jfpion, 2017-02-14:
my heat bed is the standard prusa one for 12V
the pb is it is not easy to get a 15V power switch powerful enough for the printer (bed/stepper/hotend)
for the command I  use the bang bang control with a standard automotive relay (nice cli clic when the temp is ok ;-) to avoid to much current through the ramps (already burnt one fet)

Blue-Marlin, 2017-02-14:
Make a better fan-shroud - less aiming on the bed - more pointing to the part..

Bob-the-Kuhn, 2017-02-14:
I agree with @Blue-Marlin - a better fan shroud is the best long term solution.

In the short term you might consider adjusting some of the protection parameters in Configuration_adv,h. Search for THERMAL_PROTECTION_BED_HYSTERESIS to find them.

jfpion, 2017-02-14:
even with a good one when you print small part <1cm high the aix flux cool the bed a lot and if you bed is  not powerful enough, the print stop.
I will work with the hysteresis but it not the ideal solution because i'll need a lot of try and fail to find the good value . 
thank you

Blue-Marlin, 2017-02-14:
`THERMAL_PROTECTION_BED`s main purpose is to prevent the bed from igniting something.
If you beds heater can not provide enough energy to maintain your wanted temperature, when the part-fan is on, it's likely self limiting. Probably it will not go hotter than ~130° when the bed heater is full on and the fan off. 
Since materials what could ignite at such a low temperature are usually not near the printer, i see no problem with switching off `THERMAL_PROTECTION_BED` for this beds.
(But be sure to have a suitable fuse in your heater circuit. Shorting the heater wires at the heater can burn down the wires. With the usual boards this is normally not a problem.) (With my 230V 500W 20x20cm heaters `THERMAL_PROTECTION_BED` is a must! At full throttle they destroy themselves within a few seconds.)

jfpion, 2017-02-14:
ok thank you
it will be easier

closed 2017-02-15 by thinkyhead