MarlinFirmware/Marlin #5830: z probe incomprehension

jfpion, 2017-02-14:
i just tried to configure rc8 , 
i've got a simple hand moved switch configured à z min switch for the mesh leveling.
the explaination is not really clear how to handle this simple configuration

the procedure is:
I turn by hand the probe with the switch
i launch the print (g28 z0/g29/pause)
i turn the probe  back 
i click on resume

when i try to compile marlin i get  "Auto Bed Leveling requires a probe! Define a Z Servo, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or FIX_MOUNTED_PROBE." but my system is not in the list.

must i define a virtual Z servo and doing the mouvement by hand ?

thank you

Bob-the-Kuhn, 2017-02-14:
FIX_MOUNTED_PROBE is what you need to enable.

jfpion, 2017-02-14:
OK thank you
 in the configuration.h 
// A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
// For example an inductive probe, or a setup that uses the nozzle to probe.
// An inductive probe must be deactivated to go below
// its trigger-point if hardware endstops are active.
//#define FIX_MOUNTED_PROBE

the explaination is confusing , my probe need manual deployement but is fix anyway .
I  suggest to delete "or need manual deployement"

anyway thank you again

manianac, 2017-02-15:
I'd say the documentation is correct, you are in fact "manually deploying" the probe when you 
>I turn by hand the probe with the switch

closed 2017-02-15 by thinkyhead

jfpion, 2017-02-15:
ok I just think a small add in the writing, it's a bit confusion for no very fluent english reader
i understood doesn't ....need manual deployement.
may be write "either needs manual deployement or doesn't deploy " will be less confusing
my 2 cents and not a big change
thank you all again

thinkyhead, 2017-02-15:
We'll keep working on the language as time allows. There are no doubt many places where ambiguity arises.

jfpion, 2017-02-15:
i just stumble on one ;-)
thank you