MarlinFirmware/Marlin #5832: Does Marlin have "5 second" GCode timeout protection?

MrJulz, 2017-02-14:
(I hope this is the correct place for this question, please accept apologies and give pointers if it is not)

Does Marlin have some form of timer check - where if any GCode takes longer than 5 seconds to execute, the printer resets?  Perhaps as some form of safety/sanity check?  This is the behavior I am seeing.

Example, if I add the following simple code for "M99": 
```
inline void gcode_M99() {
    for(int i=0;i<10;i++){
    SERIAL_ECHOLN(String("Iteration: ")+String(i));
    delay(1000);
}

```

Then I will see just 5 iterations, until i==4, (exactly 5 seconds), and then the printer will reset itself and boot afresh. 
 
Is there some protection function that has logic akin to: 
_if (GCode_Execution_Time>5000ms) then ResetPrinter()?_  
Or is there perhaps another explanation for the behavior I am seeing?

Thankyou

Roxy-3D, 2017-02-14:
You need to call idle() periodically in that loop.   idle() keeps the heaters doing the right thing.   If idle() is not called periodically, Marlin tries to protect the printer and shuts down the heaters.

MrJulz, 2017-02-14:
Smart. Thanks.

closed 2017-02-14 by MrJulz

thinkyhead, 2017-02-15:
The feature that resets the board is called `USE_WATCHDOG` and it's actually a 4 second timer.