MarlinFirmware/Marlin #5833: automatic bed levering failed : G29 matrix seems wrong with LJC18A3-H-Z/BX

moriamoria, 2017-02-15:
Hello,

First, many thanks for your help.

I have an issue with a brand new anet A8. I have install Merlin/Skynet 3.2.1 firmware.

When I use auto levering, the result is wrong. I got bad result with the matrix 3*3.

I have set the bed with a milimeter.
I have install a capacitif sensor LJC18A3-H-Z/BX
I have set M851 Z offset.

But when I run G28 and G29, I got following results :

 G28
< X:136.00 Y:150.00 Z:10.50 E:0.00 Count X: 13600 Y:15000 Z:4200
> G29
< G29 Auto Bed Leveling
< Eqn coefficients: a: 0.00120040 b: 0.00009445 d: 0.41823997
< Bed Level Correction Matrix:
< +0.999999 +0.000000 +0.001200
< -0.000000 +1.000000 +0.000094
< -0.001200 -0.000094 +0.999999
< X:219.01 Y:210.00 Z:10.27 E:0.00 Count X: 21900 Y:21000 Z:4200

when I execute G0 X100 Y100 Z0 the noze is 1mm above the bed.

to test my sensor, i have check Z position with following commands :

G28
G0 X20 Y20 Z0.1 F8000 ; found offset =-0.09
G0 X20 Y200 Z0.1 F8000 ; found offset =-0.09 not bad but the Bed Level Correction Matrix gives +0.99999

I think the problem is here

Can you help me ?

Regards

Blue-Marlin, 2017-02-15:
Please look up what https://en.wikipedia.org/wiki/Rotation_matrix is.

moriamoria, 2017-02-15:
Thanks for your answer. So my rotation matrix shows that my bed is quiet good align.

So my first question remains. Why after G29 my bed is 1mm above the bed. 

I have removed command G29 after g28.

I will execute commande G29 v4 and gives result soon.

Regards

Bob-the-Kuhn, 2017-02-15:
I think you're comparing apples & oranges.

Bed Level Correction Matrix are the numbers created by G29 that are used to make corrections to the Z height.

If you issue G0 X20 Y20 Z**0.1** , the resulting Z position has been compensated so that, in theory, the distance between the nozzle and the bed will be **0.1** mm. This assumes the Z offset is set perfectly.

The actual offsets of  +1.0, -0.09 & -0.09 have a wider spread than what I see on my machine.  Mine are usually in the +-0.1mm which is as good as I can measure using the paper method.

For improved accuracy you may want to go to a 5x5 matrix.

You may also want to run the M48 repeatability test near several of the probe points.  That'll give you some idea of how much error your probe introduces into the Bed Level Correction Matrix.

You can compare corrected vs. uncorrected Z positions by using the M420 S1 & the M420 S0 commands to turn on/off bed leveling.

moriamoria, 2017-02-15:
Thanks,

I will proceed the test and give you results

Regards

b3niup, 2017-02-15:
@moriamoria your issue may be the same as I've got with this exact same sensor. I've described it in issue #5687.

Here you can see failed test print using correctly set up linear auto-leveling:
[image]

moriamoria, 2017-02-20:
Hello,

I just made some modification on my bed 

I glue on the bed magnetic folder

The I have cut a 200*200 steel with zinc plate.

On the steel zinc plate, i have glue lokbuild plate.

When I run G28 and G29, I got following results 👍 

G29
< G29 Auto Bed Leveling
< Eqn coefficients: a: 0.00002480 b: 0.00083333 d: 0.62757425
< Bed Level Correction Matrix:
< +1.000000 +0.000000 +0.000025
< -0.000000 +1.000000 +0.000833
< -0.000025 -0.000833 +1.000000
< X:219.00 Y:210.01 Z:10.31 E:0.00 Count X: 21900 Y:21000 Z:4236

then I check with paper following positions and all seems good

> G0 X100 Y100 Z0.1 F9000
> G0 X10 Y200 Z0.1 F9000
> G0 X20 Y200 Z0.1 F9000
> G0 X200 Y20 Z0.1 F9000
> G0 X200 Y200 Z0.1 F9000
 
So I don't understand why I got issue before. 

Please find attached some picture and test I have made.

What look odd, is that printing is not homoginious . More ABS on (20,20) and less on (20,150)
[image]
[image]
[image]
[image]
[image]
[image]

Regards and thanks

benlye, 2017-03-10:
@moriamoria and @b3niup Not sure if you're still having this problem, but I and others have had similar problems.  A few of us looked quite closely at how bilinear auto-levelling works and were satisfied that the code is doing the right thing (#5898).

In my case my Z-probe Y offset was out by 2mm, and setting that correctly has solved my auto-levelling problem.  It might be worth you checking that you X and Y offsets are correct.

closed 2017-10-31 by thinkyhead