MarlinFirmware/Marlin #5834: How to create custom gcode/pin references? marlin.main.cpp

kcheeeung, 2017-02-15:
I have an Azteeg X3 Pro board (ATMEGA2560 and RAMPS 1.4 base). How would I be able to create such a custom gcommand in the Marlin software? In marlin.main.cpp, I don't see anything referring to a specific pin, just outputs like fan or extruder.

I am trying to generate a pulse like this with a gcommand.
[image]

I want to do something akin to this: (Would this even work? If not, where/how should I structure it so I can make a gcommand generate a pulse?)
```

const int kPinPulse = 8; // You can select any pin of arduino
const int kPinSW = 9;    //This switch will work as command

void setup()
{
   pinMode(kPinPulse, OUTPUT);
   pinMode(kPinSW, INPUT);      
}

void loop()
{
   if(digitalRead(kPinSW) == LOW) //When Switch pressed Command Sent
   {
      digitalWrite(kPinPulse, HIGH); // start your pulse
      delayMicroseconds(200);        // It's provide your time interval of 200 microseconds
      digitalWrite(kPinPulse, LOW);  // end your pulse
      while(digitalRead(kPinSW) == LOW); 
   }     
}
```
http://electronics.stackexchange.com/questions/283950/limitations-of-arduino-voltage-pulse-ttl

Blue-Marlin, 2017-02-15:
Use, copy or alter M240 (PHOTOGRAPH_PIN).

Bob-the-Kuhn, 2017-02-15:
Why do you want to generate the pulse?   Maybe there's an easier solution to your problem.

Bob-the-Kuhn, 2017-02-15:
Since all your reads and writes have fixed pins then replace digitalRead with READ and digitalWrite with WRITE.  The macros run faster and use less program space.

----

The preferred location for pin definitions for your board is in the pins_AZTEEG_X3_PRO.h file.

The preferred naming convention would result in the following being added to pins_AZTEEG_X3_PRO.h file:
```
#define KPINPULSE_PIN     8
#define KPINSW_PIN          9
```

----

Pin definitions can be "interesting".   The naming conventions have changed over the years and there have been lots of different authors.

Almost all pin definitions originate in a file with this naming convention: pins_YOURBOARD.h.  You'll find a few in Configuration.h & Configuration_adv.h (discouraged).

Because of history you can find the same function given different names.  FAN_PIN, FAN0_PIN, EXTRUDER0_FAN_PIN and the currently preferred E0_AUTO_FAN_PIN all refer to the fan for extruder 0.  FAN_PIN & FAN0_PIN get converted to E0_AUTO_FAN_PIN by the pre-processor.  Users of EXTRUDER0_FAN_PIN are told to convert to the new naming convention.

All the RAMPS based cards & derivatives pull in a one of the pins_RAMPSxx.h files.  In your case it's pins_RAMPS.h.  Most of the AzteegX3pro pin definitions are actually in pins_RAMPS.h. 

----

You'll also find that some pins have multiple functions assigned to them.  Usually that's not a problem because only one of the functions has been enabled in the Configuration.h & Configuration_adv.h files.  This is especially common with the pins for the LCD controllers.

Bob-the-Kuhn, 2017-02-15:
Are you wanting this function to always be running in the background?  If yes then we'll need to do something special.

Marlin commands are almost all "run to end & terminate" style.  Movement commands (G0, G1, G29, ...) are queued up and executed consecutively.  Almost all other commands are executed as they are received.

The command sequence G0 X10, Y10 Z10, G0 X100, Y100 Z100, M240 will result in the head moving to 10,0,10, then the head will move to 100,100,100 but the pin for the M240 command will be pulsed well before the head gets to 10,10,10.

kcheeeung, 2017-02-16:
@Bob-the-Kuhn 
We have a device that actually needs a pulse to become activated and requires a very fast time in microseconds. It's a modification of a 3D printer that extrudes a special material in droplets rather than a continuous flow.
In our gcode commands, we have the printer move to a location then pulse once. So I guess we only need to terminate it after one pulse each time.

It'd be like this. We make our own G/M commmand
G0 XYZ
M# pulse here
G0 XYZ
M# pulse here
...

To understand correctly, is the behavior this?
Move to XYZ (stop), pulse, Move to XYZ (stop), pulse, ...

What do you think of this code?
This is what we have so far:
```
  inline void gcode_M700() {
    static int pulse_usec = 500; //This just defaults to 500 usecs
    bool set = code_seen('S'); //attempting to figure out this S business
    if (set) {
      pulse_usec = code_value_int();
    }
    digitalWrite(FAN_PIN, HIGH);
    delayMicroseconds(pulse_usec);
    digitalWrite(FAN_PIN, LOW);
  }
```
Can I replace the FAN_PIN is something like HEATER_2_PIN instead of the actual number?
Also ideally we'd want to say M700 S###, pulse for ###usecs. We're still figuring out how to vary the microseconds with this S.

So I'd replace digitalWrite > WRITE in this one, yes?

Bob-the-Kuhn, 2017-02-16:
You're headed in the right direction.

The big thing is you need a means of delaying the pulse until the head gets to the desired position.  The M400 command will do that.

Your command sequence would be:
G0 XYZ
M400
M# pulse here
G0 XYZ
M400
M# pulse here

The M400 code could be included in your code if desired.

Please pick your gcode name from this [list](http://reprap.org/wiki/G-code).  If nothing fits then use one that isn't in it.

----

I've taken the liberty of putting your code in the format used by Marlin.  Hope that's OK

All code goes into Marlin_main.cpp

Insert the main code around line 7000

 ```
 // M400 Sxx  pulse pin high for xx uSeconds
  //           defaults to 500
    
  inline void gcode_MXXX() {
    #define DEFAULT_PULS_WIDTH  500
    #define INIT_DELAY          500
    uint16_t pulse_usec = code_seen('S') ? code_value_int() : DEFAULT_PULS_WIDTH;
    stepper.synchronize();             //wait for head to reach destination (replaces M400)
    OUT_WRITE(PULSE_PIN, LOW);           // init pin in case this is the first time
    delayMicroseconds(INIT_DELAY);     //  Is this needed?
    WRITE(PULSE_PIN, HIGH);
    delayMicroseconds(pulse_usec);
    WRITE(PULSE_PIN, LOW);
  }
```

insert command link around line 8700

```
  case XXX: // MXXX: Pulse high
    gcode_MXXX();
    break;
```

insert init code around line 10400 ( if you want it executed when Marlin starts up)

`  OUT_WRITE(PULSE_PIN, LOW);  //init pulse pin`

kcheeeung, 2017-02-18:
@Bob-the-Kuhn Okay thank you so much!

kcheeeung, 2017-02-19:
@Bob-the-Kuhn 
Sorry one more followup. Apparently we need an analog signal instead of digital.

On the Azteeg X3 Pro board, which outputs do you think we should use? (We've been using the heater/fan ports. Should be do an digital to analog conversion?)

Or should I try using the analog numbers available on the board? Is there anything special I need to do in order to use those pins?

We decided to keep the functions separate and chain them with the M400.
```
inline void gcode_M430() {
  #define DEFAULT_PULS_WIDTH  200
  #define CHECK_DELAY          50
  uint16_t pulse_usec = code_seen('S') ? code_value_int() : DEFAULT_PULS_WIDTH;
  delayMicroseconds(CHECK_DELAY);     //  Is this needed?
  WRITE(FAN_PIN, HIGH); //Fan pin doesn't work since it's not analog
  delayMicroseconds(pulse_usec);
  WRITE(FAN_PIN, LOW); //Fan pin doesn't work since it's not analog
  delayMicroseconds(CHECK_DELAY);
}
```

closed 2017-02-19 by kcheeeung

Bob-the-Kuhn, 2017-02-19:
Sounds like you need to use a hardware PWM port so that you can have an analog output.  A hardware PWM will produce a 1k-15kHz signal that varies in duty cycle.  To a slow device this looks like an analog signal.  The frequency of the signal varies with duty cycle.

 

If that’s the case then just replace the “WRITE(PULSE_PIN, HIGH);” statement with “analogWrite(PULSE_PIN, value);” where value can range from 0 through 255.

 

If you’re using one of the on-board FETS then a value of 0 will give you maximum output.

 

Here’s the useable hardware PWMs on your board:

 

*      Timer    Digital  Normally    

 *      & Port      Pin    assigned    

 *      TIMER3B   2     X_MAX                       

 *      TIMER3C   3     X_MIN                       

 *      TIMER0B   4     HEATER_4

*      TIMER3A   5     HEATER_5

*      TIMER4A   6     HEATER_6

*      TIMER4B   7     LCD

*      TIMER4C   8     HOTBED

*      TIMER2B   9     HEATER_1

*      TIMER2A   10    HEATER_0

*      TIMER5C   44    LCD                         

 *      TIMER5B   45    LCD                         

 

Because of the way pins are defined for the Azteeg X3 Pro it’s hard to tell what is assigned to a particular pin.  There may be multiple functions assigned to one pin.  To determine what’s assigned in your system do the following:

1.       In Configuration_adv.h enable PINS_DEBUGGING

2.       Issue M43 I1 command 

3.       Note the name of the signal assigned to the pin you want to use

4.       Add #undef signal name to the end of your pins_AZTEEG_X3_PRO.h file

5.       Repeat steps 2-4 until unassigned shows up

6.       Add #define PULSE_PIN  xx to the end of your pins_AZTEEG_X3_PRO.h file

thinkyhead, 2017-02-21:
Timer & Port|Digital Pin|Normally assigned
------------|-----------|-----------------
TIMER3B|2|X_MAX
TIMER3C|3|X_MIN
TIMER0B|4|HEATER_4
TIMER3A|5|HEATER_5
TIMER4A|6|HEATER_6
TIMER4B|7|LCD
TIMER4C|8|HOTBED
TIMER2B|9|HEATER_1
TIMER2A|10|HEATER_0
TIMER5C|44|LCD
TIMER5B|45|LCD

LabFnP, 2018-04-19:
Hello, may I ask which version of Marlin @kcheeeung used?  
I have the same situation for my project. 
Thanks

Bob-the-Kuhn, 2018-04-19:
You should be able to do this with any version of Marlin.

If you need more help then please provide details on your board and what you're trying to do.

LabFnP, 2018-04-20:
@Bob-the-Kuhn Thank you for your reply.
My board is RAMPS V1.4

Below is my custom G code:
`inline void gcode_M430() { 
  analogWrite(9, 1); //Fan pin doesn't work since it's not analog
  delay(20000);
  analogWrite(9, 0);
  delay(20000);
}`

Currently, I can use this "M430" in the serial port monitor, but I found the delay time (20000ms) is incorrect when this command executed.  It seems the delay time have the maximum (only about 8000ms). When the delay time is above maximum delay time, the PWM output will be suspended (at 8000ms).  Furthermore, I found the serial port monitor will be reloaded and show the config echo. But if I enter the delay time which is below to the maximum time, serial port monitor still can reply 'ok'. Due to we need to stay at same PWM much longer, I still try to find out the solution.

Do I need to revise any default setting of Marlin?

thinkyhead, 2018-04-20:
You've rebooted the card with your code. You need to use `safe_delay` so that the machine can still operate and reset the watchdog timer.

LabFnP, 2018-04-20:
Wow, @thinkyhead you are right! Thank you so much.