MarlinFirmware/Marlin #5835: Request: Smart (part) fan management for dual X carriage/dual extrusion.

oysteinkrog, 2017-02-15:
When changing tools, we can disable fans on the inactive tool.
E.g. Switching to T1 we can turn off fan0 and turn on fan1, and then when switching back to T0 we can turn on fan0 and turn off fan1.
The advantage here is that it will automatically work with existing slicers (which don't understand multiple fans).
Will make these kinds of machines more silent :)

Bob-the-Kuhn, 2017-02-15:
Currently Marlin software assumes that all extruders can be switched in & out multiple times within a print job so the hot end temperatures are maintained on the inactive extruder(s).  Fan control is part of maintaining the hot end temperature.  Turning the fan off while keeping the hot end at temperature could result in the filament melting further up inside the hot end than intended.

oysteinkrog, 2017-02-15:
I am not referring to extruder fans, but to _part_ cooling fans :)

Bob-the-Kuhn, 2017-02-15:
Never mind ...

Never been in the dual carriage world

oysteinkrog, 2017-02-15:
I think this would also be good for some "normal" dual extrusion setups (two extruders on one carriage), where the part cooling fan is not shared between them.

thinkyhead, 2017-04-02:
As a current solution, you can ask the slicer to insert `M106`/`M107` fan commands before/after the `T#` tool-switch commands. Some slicers have an option to set code that runs on each tool-change, and some slicers provide substitution tokens. So, for example:
```gcode
M107 P0 ; turn off all fans
M107 P1
M106 S200 P<extruder> ; turn on active extruder fan
```

oysteinkrog, 2017-04-04:
Aha!
That is a nice workaround, I was not even aware I could define multiple fans:P
After fiddling with the pins a bit, and hooking up some external mosfet driver boards I got it working.
I changed the tool change gcode to use slicer variables for the current fan speed, e.g. for simplify 3d it would be like this:

```M107 P0 ; turn off all fans
M107 P1
M106 S[fan_speed_pwm] P[new_tool] ; turn on active extruder fan
```

I guess it fails in the cases where the slicer changes the fan speed within the same layer, but that is not that often I guess :)
Thank you @thinkyhead :)

Roxy-3D, 2017-04-04:
> I guess it fails in the cases where the slicer changes the fan speed within the same layer, but that is not that often I guess :)

I don't know how often that happens...  But I've been seeing it on a lot of prints lately.  If my printer is bridging a gap, it cranks up the fan speed prior to doing that.   That does seem to help the bridge be pulled tight from side to side without much sag.

oysteinkrog, 2017-07-12:
I went ahead and did some quick hacks for this myself: 
https://github.com/oysteinkrog/Marlin/tree/feature/DualXAutoFanControl

hardin162, 2017-08-22:
I am running a dual carriage on ramps. I can't find where to define the pins for the part cooling fans in marlin since i am using a fan extender . It still believes that the pin is D9 and if I engage the part cooling fan the second extruder heats up. If you knew what pin in marlin that i have to change to not have the hotend heat up and to change it to the pin I have the fan extender attached to, that would be awesome!! 

I originaly thought that it was but 
ndef FAN_PIN
  #define FAN_PIN -1
#endif
#ifndef FAN1_PIN
  #define FAN1_PIN -1
#endif
#ifndef FAN2_PIN
  #define FAN2_PIN -1

that did not seem to fix my issue.

RollingGecko, 2017-10-28:
Have you already set a pull request?

closed 2018-02-06 by thinkyhead