MarlinFirmware/Marlin #5836: Auto bed Leveling starting Script

pinheirofox, 2017-02-15:
Hi 
I'm setup a BLTouch in my homemade printer, but Im having problems when I add  G0/1 Z0 to the Gcode Script. When I start printing it runs the G28 and G29 but after, this it was supposed run the Go Z0 But it starts to print without running G0 Z0. So the printer does not recognize Z offset.
Does anyone know if I forgot to set something on marlin? My marling version is RC7

thinkyhead, 2017-02-15:
Try the current `RCBugFix` and see if it's any better. RC7 has many bugs that were fixed with `RC8`, and `RC8` has bugs that have been fixed in `RCBugFix`.

Bob-the-Kuhn, 2017-02-15:
What makes you think it didn't run G0 Z0?

If it's because the nozzle is above the bed then you need to adjust z_offset via M851 and re-run G29 until the desired clearance is achieved.  Don't forget to put the new z_offset back into your config file.

Is the machine already at Z0?  Try doing the commands manually and issue a M114 command instead of the G0 Z0 command.

And yes, RCBugFix takes care of a lot of problems in RC7.  I think it's a good idea to switch even if M851 solves your problem.  Especially since you're running a BLTouch.  Lots of recent BLTouch fixes/improvements.

pinheirofox, 2017-02-15:
My problem as been solverd @thinkyhead. I update the firmware to RCBugFix. Thank you, now my printer works! @Bob-the-Kuhn Thanks for the opinion, but I had already adjusted the Z_offset, even stranger this is happening. Now everything is fine and thanks again.

psavva, 2017-02-18:
Please close this incident if your issue is resolved

closed 2017-02-20 by pinheirofox