MarlinFirmware/Marlin #5838: G29 calibration off on delta printer

ahito89, 2017-02-15:
Hello,
I have Kossel XL printer and I've recently added an dc42 ir probe to it and I've flashed RCBugFix to my Ramps 1.4 board. I have a 250 mm bed so I've set:
**DELTA_PRINTABLE_RADIUS 125.0**
and disabled **DELTA_HOME_TO_SAFE_ZONE**
for the probe:
**FIX_MOUNTED_PROBE
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
X_PROBE_OFFSET_FROM_EXTRUDER 0
Y_PROBE_OFFSET_FROM_EXTRUDER +10;
Z_PROBE_OFFSET_FROM_EXTRUDER -1.3
AUTO_BED_LEVELING_BILINEAR
ABL_GRID_POINTS_X 5
ABL_GRID_POINTS_Y ABL_GRID_POINTS_X**

After this i sent a G28, followed by M502 and M500(i have eeprom enabled) and after that G29
The problem is that it starts probing in the middle of the build plate and then after a few points I have to stop it or else it will crash outside the build plate. I thing the matrix points generated are offset and I don't know why it keeps probing from the middle and the goes in YZ direction.
Any ideas? Here's a picture to ilustrate the problem with the probing points
[image]

JohnOCFII, 2017-02-15:
@ahito89 I had the exact same issue.

What version of Marlin are you using?  

When using RC8, I had made a number of changes, but I think the one that allowed the G29 tp probe the correct area of the bed was to define DELTA_HOME_TO_SAFE_ZONE.

When Moved to RCBugFix, I was able to get the bed to probe the correct areas without having to define DELTA_HOME_TO_SAFE_ZONE.

John

Blue-Marlin, 2017-02-15:
Ether let on `DELTA_HOME_TO_SAFE_ZONE` or leave the 'danger zone' after G28 and before G29.

ahito89, 2017-02-15:
Hello,
I am using an RCBugFix branch cloned yesterday. I'll try DELTA_HOME_TO_SAFE_ZONE tomorrow evening and post the results here, but why wouldn't it probe correctly after normal homing?

ahito89, 2017-02-20:
Hello,
Could not get it to probe reliably. DELTA_HOME_TO_SAFE_ZONE helped but still goes off bed on some points. I've reverted back to version 1.0 and with that version the probing works without any issues, just need to fix calibration issues after probing.

ahito89, 2017-02-22:
I've got the most recent version of RCBugFix and the issue was fixed.

closed 2017-02-22 by ahito89