MarlinFirmware/Marlin #5843: Freeze when homing

theodroid, 2017-02-16:
Hi,
I have a problem with the latest RCBugFix. This bug is present for a long time now (alrerady here in the RC7).
After homed the X axis, the printer homes the Y axis, go backwards few mm, and bumps again the Y endstop but slower. At this moment the printer freezes...
I have a SainSmart 2-in-1 board. I send you my configuration.h and configuration_adv.h files.
Do you think it's a bug ?

Yours sincerely,

[config and config_adv files.zip](https://github.com/MarlinFirmware/Marlin/files/779943/config.and.config_adv.files.zip)

Tannoo, 2017-02-16:
Download the latest RCBugFix again. There have been recent changes that may help you. Those config files are at least a week old.

theodroid, 2017-02-16:
Thanks for your answer.
The RCBugFix I was using was the version of 14th february.
I downloaded the latest RCBugFix again, and uploaded it to my printer, but the problem is always here: the X axis always homes perfectly, but when the Y axis bumps the Y endstop, sometimes Marlin freezes instead of moving the print head backwards to bump the Y endstop a second time.
So I can no longer send commands to the printer since it's freezed... The screen is also freezed.

Tannoo, 2017-02-16:
Have verified that the Y endstop works correctly? Is your 5v still good when the system freezes?

theodroid, 2017-02-17:
I probably found the problem... The Y endstop works correctly, but my 2 endstops (X and Y) were wired incorrectly to the motherboard. What surprises me is that the controller board never burn out because of a short-circuit... It worked for few months...
And the X axis always homed OK, the problem concerned only the Y axis altough the two endstops were wired the same.
Anyway, I re-wired my endstops and now it works!
So sorry for the inconvenience, it was my fault!

Tannoo, 2017-02-17:
Glad you found it. You can now close this issue.

closed 2017-02-17 by theodroid