MarlinFirmware/Marlin #5846: Is fastio.h TOGGLE macro destructive?

MrJulz, 2017-02-16:
In fast.io there is a toggle macro:
`#define _TOGGLE(IO)  do {DIO ##  IO ## _RPORT = MASK(DIO ## IO ## _PIN); } while (0)
`

Question: is this not destructive? There is no bitwise OR operator - it is just a straight assignment - so whichever pin you "Toggle", the remaining pins will all actually be Zero'd?

Practical example: if TOGGLE is used is enable an pull-up resistor on PIN7, wouldn't this macro also (inadvertently) disable pullup resistors on pins 0..6 at the same time?

Should the code perhaps read?
`#define _TOGGLE(IO)  do {DIO ##  IO ## _RPORT |= MASK(DIO ## IO ## _PIN); } while (0)
`

Apologies if I am misunderstanding something, am new to Marlin and Microprocessors.

MrJulz, 2017-02-16:
Ah, now I understand a little more: IF DDRx is set so pin(x) is OUTPUT, then writing a 1 to a PINx will toggle the corresponding PORTx value. (i.e will indeed toggle the "output" of that pin)

However, IF **any** pins are set to INPUT in the DDR, (and have pull up-resistor enabled), then wouldn't the TOGGLE macro still inadvertently turn off those pullup resistors?

Roxy-3D, 2017-02-16:
That is a good question.   If you unplug your RAMPS board from the Mega-2560 (assuming that is the hardware you have) you get easy access to all of the I/O pins.   It would be easy to do a simple test and see what the behavior is.    I think there is a fair chance toggling an I/O pin will eliminate the pull up resistor from being active.   But I don't know that.

Blue-Marlin, 2017-02-16:
Maybe.
The TOGGLE() macro is defined in fastio.h but it's never used in Marlin. So likely it's untested.

MrJulz, 2017-02-16:
Thanks.  Upon improving my understanding:  (assuming PINn is in Input Mode)

1) My primary mistake: to enable a pull up resister, you write 1 to PORTxn, (**NOT** PINxn). 
2) Writing a 0 to PORTxn does indeeed disable a pullup resister.

3) Writing a 1 to PINxn toggles (bitflips) the corrosponding data bit in PORTxn.
4) Writing a 0 to PINxn does **nothing**, ever.

5) The code structure is such that the macro is hardcoded to only write to the PINx register.  While a bytewise TOGGLE on a PORTx would indeed disable all other pull up resistors as suggested- that is impossible to do with the given code structure.

6) Specific relevant code (see "**DIO ##  IO ## _RPORT**" below):
```
#define DIO87_RPORT             PINE
#define _TOGGLE(IO)  do {DIO ##  IO ## _RPORT = MASK(DIO ## IO ## _PIN); } while (0)
```

Summary: There is no problem with this TOGGLE macro. Appreciated the great responses. Sorry for wasting your time.

closed 2017-02-16 by MrJulz